AltBeacon/android-beacon-library #1010: App Crashed in Android version 10

Gagandeep1408, 2020-11-09:
Caused by java.lang.IllegalStateException
Not allowed to start service Intent { cmp=com.***.***/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{63bd9fc u0a165 RCVR idle change:uncached procs:1 seq(0,0,0)}
org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer.bindService

android.app.ContextImpl.startServiceCommon (ContextImpl.java:1617)
android.app.ContextImpl.startService (ContextImpl.java:1572)
android.content.ContextWrapper.startService (ContextWrapper.java:669)
org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer.bindService (RegionBootstrap.java:219)
org.altbeacon.beacon.BeaconManager.bind (BeaconManager.java:438)
org.altbeacon.beacon.startup.RegionBootstrap.<init> (RegionBootstrap.java:105)
com.****.****.****.AppController.onCreate (AppController.kt:137)
android.app.Instrumentation.callApplicationOnCreate (Instrumentation.java:1189)
android.app.ActivityThread.handleBindApplication (ActivityThread.java:6488)
android.app.ActivityThread.access$1400 (ActivityThread.java:226)
android.app.ActivityThread$H.handleMessage (ActivityThread.java:1884)
android.os.Handler.dispatchMessage (Handler.java:107)
android.os.Looper.loop (Looper.java:214)
android.app.ActivityThread.main (ActivityThread.java:7386)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:492)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:980)

matdev, 2021-01-25:
Same exception using the Reference app with lib version 2.17.1. It occured after the app went to background :

```
 java.lang.RuntimeException: Unable to create application org.altbeacon.beaconreference.BeaconReferenceApplication: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{b0f93f0 u0a571 SVC  idle change:uncached procs:1 seq(0,0,0)}
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6652)
        at android.app.ActivityThread.access$1600(ActivityThread.java:231)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1952)
        at android.os.Handler.dispatchMessage(Handler.java:107)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7682)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:516)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)
     Caused by: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{b0f93f0 u0a571 SVC  idle change:uncached procs:1 seq(0,0,0)}
        at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1650)
        at android.app.ContextImpl.startService(ContextImpl.java:1605)
        at android.content.ContextWrapper.startService(ContextWrapper.java:669)
        at org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer.bindService(RegionBootstrap.java:228)
        at org.altbeacon.beacon.BeaconManager.bind(BeaconManager.java:438)
        at org.altbeacon.beacon.startup.RegionBootstrap.<init>(RegionBootstrap.java:114)
        at org.altbeacon.beaconreference.BeaconReferenceApplication.onCreate(BeaconReferenceApplication.java:104)
        at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1197)
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6647)
        at android.app.ActivityThread.access$1600(ActivityThread.java:231) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1952) 
        at android.os.Handler.dispatchMessage(Handler.java:107) 
        at android.os.Looper.loop(Looper.java:214) 
        at android.app.ActivityThread.main(ActivityThread.java:7682) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:516) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950) 

```

davidgyoung, 2021-01-25:
What phone model?
Was the reference app configured to use a foreground service?
Were beacons detectable when this happened?

matdev, 2021-01-25:
> What phone model?
OnePlus 6
> Was the reference app configured to use a foreground service?
no it wasn't
> Were beacons detectable when this happened?
Yes

davidgyoung, 2021-08-13:
The problem here is that Android 10 won't allow you to start a service from the background and you have configured the library to do so.  Android 8+ generally does not allow you to keep a service running in the background, and Andorid 10+ will give you this exception if you even try to start a non-foreground service from the background.

In order to fix this, you must do one of two things to get around Android restrictions:

1. Accept the use of the job scheduler to schedule your scans (this is the library default -- just don't turn it off by calling the same method with `false`.)`beaconManager.setEnableScheduledScanJobs(true)`

2. Configure a foreground service notification to go with your scanning service.    See instructions [here](https://altbeacon.github.io/android-beacon-library/foreground-service.html)

closed 2021-08-13 by davidgyoung