AltBeacon/android-beacon-library #1015: Unable to get nearby BLE devices, when requesting location updates using FusedLocationProviderClient

kpraveen455, 2020-12-11:
### Expected behavior

### Actual behavior

### Steps to reproduce this behavior

### Mobile device model and OS version

### Android Beacon Library version

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

kpraveen455, 2020-12-11:
When in my application using the following code:

        LocationRequest mLocationRequest = new LocationRequest();
        mLocationRequest.setInterval(1 * 1000); // 1 sec interval
        mLocationRequest.setFastestInterval(1* 1000); // 1 sec interval
        mLocationRequest.setPriority(LocationRequest.PRIORITY_HIGH_ACCURACY);

        fusedLocationProviderClient.requestLocationUpdates(mLocationRequest, locationCallback, Looper.myLooper());

the nearby BLE beacon detection is stopping. When I commented the above line, the Alt Beacon library, is able to get the nearby BLE devices, (every single second).

davidgyoung, 2020-12-11:
@kpraveen455 for assistance on using this library with a custom application (including troubleshooting combining it with other APIs like the fusedLocationProvider), please post a question on StackOverflow.com.  This forum is reserved for feature requests and bug reports with the library itself.  

If you post your question on StackOverflow.com, please include in the context your beacon scanning setup code, and the definition of Looper.myLooper().  it may be that you have a threading issue where the thread is blocked sending becon updates.

closed 2020-12-11 by davidgyoung