AltBeacon/android-beacon-library #1016: Fix BeaconManager in master

lihop, 2020-12-18:
Two problems:
1. Listeners not unregistered. When terminating the app 'regionDidEnter' listeners are not removed so beacons keep ranging. Then when app is started again listeners are added again so mutiple listeners get fired. The number increases every time the app is terminated and started.
2. Beacons not ranging when screen is off. See: https://github.com/AltBeacon/android-beacon-library/issues/769

Fixes implemented for master-rental, but should be applied to master.

davidgyoung, 2020-12-18:
What are the specific symptoms of the library API?  What is a “listener”?  The library has no such name in its API.

lihop, 2020-12-18:
So sorry, this issue was meant for a private repo. Opened here by accident :frowning:

closed 2020-12-18 by lihop