AltBeacon/android-beacon-library #1018: API 30/android 11 library won't add anymore.

JimSeker, 2021-01-28:
### Expected behavior
    compileSdkVersion 30  
 implementation 'org.altbeacon:android-beacon-library:2+' 
  should run, but fails.
### Actual behavior
     java.lang.NoClassDefFoundError: Failed resolution of: Landroidx/localbroadcastmanager/content/LocalBroadcastManager;
### Steps to reproduce this behavior
   You only need the implementation line and    beaconManager = BeaconManager.getInstanceForApplication(this); to cause the failure.
### Mobile device model and OS version
  Android 11.
### Android Beacon Library version
   implementation 'org.altbeacon:android-beacon-library:2+'  //2.17.1 I think is current version.

note this is fixable by adding implementation 'androidx.localbroadcastmanager:localbroadcastmanager:1.0.0'    to module build.gradle file. 
See 
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

Kaszmir, 2021-01-28:
@JimSeker did you add 'androidx.localbroadcastmanager:localbroadcastmanager:1.0.0' to your app module or to fetched beacon lib module?

davidgyoung, 2021-01-28:
Yes, it is necessary to add `implementation 'androidx.localbroadcastmanager:localbroadcastmanager:1.0.0' `  You can see an example of this here:

https://github.com/davidgyoung/android-beacon-library-reference-kotlin/blob/master/app/build.gradle#L35

I have made a full kotlin version of the reference app here which includes the above:

https://github.com/davidgyoung/android-beacon-library-reference-kotlin/

I still need to update the Library docs to show this proper setup.  A future release will make this unnecessary, but for now it is required.

JimSeker, 2021-01-28:
thanks.   I look forward to the new release.

davidgyoung, 2021-04-05:
Resolved via #1022.  If using 2.18-beta or higher (now available on Maven Central as JCenter is going away) you no longer need to include the localbroadcastmanager dependency.

closed 2021-04-05 by davidgyoung