AltBeacon/android-beacon-library #1019: Host artifact elsewhere - JCenter shuts down on May

alvindizon, 2021-02-04:
According to this [article](https://jfrog.com/blog/into-the-sunset-bintray-jcenter-gocenter-and-chartcenter/), JCenter will be shut down on May. Now's the time to host the artifact/library elsewhere.

davidgyoung, 2021-04-05:
Resolved via #1022.  2.18-beta1 is the first release in two years again available on MavenCentral

closed 2021-04-05 by davidgyoung

ostrya, 2021-04-12:
Hi, I noticed there is a new beta release of this library available on Maven Central. However, the URL and license description is wrong: https://search.maven.org/artifact/org.altbeacon/android-beacon-library

The artifact now refers to https://github.com/getstream/stream-chat-android as its source location and proclaims the "Stream License". Is this intentional? The content indeed still seems to be this library.

davidgyoung, 2021-04-12:
Thank you for alerting me to this, @ostya.   This is indeed an oversight.  Those incorrect metadata artifacts come from the tutorial I used for the Maven publishing here:  https://getstream.io/blog/publishing-libraries-to-mavencentral-2021/. I will fix this in a new beta release.

davidgyoung, 2021-04-12:
The metadata is fixed in 2.18-beta3 here:  https://search.maven.org/artifact/org.altbeacon/android-beacon-library/2.18-beta3/aar