AltBeacon/android-beacon-library #1020: BeaconParser Error with Eddystone UID

jagsus, 2021-02-05:
### Expected behavior
Eddystone UID Layout should parse without Error?!

### Actual behavior
```
D/BeaconParser(13992): Parsing beacon layout: s:0-1=feaa,m:2-2=00,p:3-3:-41,i:4-13,i:14-19
D/BluetoothAdapter(13992): STATE_ON
E/MethodChannel#pl.pszklarska.beaconbroadcast/beacon_state(13992): Failed to handle method call
E/MethodChannel#pl.pszklarska.beaconbroadcast/beacon_state(13992): java.lang.IllegalArgumentException: Beacon has 3 identifiers but format requires 2
```

### Steps to reproduce this behavior
I'm currently developing a Flutter App. I'm using the beacon_broadcast plugin, which uses this library. When trying to find out where the Error message comes from, I could only find it in the altbeacon code.

### Mobile device model and OS version
Samsung Galaxy A10s / Android 10

### Android Beacon Library version
2.16.4 (as set in the build.gradle from beacon_broadcast library)

davidgyoung, 2021-02-05:
This exception happen when you try to start transmitting with the `BeaconTransmitter` class, and the beacon you use for transmission is set up with more identifiers than the layout.  It looks like your code is trying to transmit a three identifier beacon with Eddystone UID, which only has two identifiers.  That isn't allowed.

Here is the root cause of the exception:  https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/java/org/altbeacon/beacon/BeaconParser.java#L668

closed 2021-02-05 by davidgyoung

jagsus, 2021-02-05:
Thanks for the quick reply, it helped a lot! Turns out the beacon that's used in the beacon_broadcast plugin is always built with 3 IDs, no matter the format...