AltBeacon/android-beacon-library #1021: Measuring 1-Meter Reference RSSI value

beemavishnu, 2021-02-17:
Hello,

I have basic query of measuring 1-Meter Reference RSSI value.

Whether 1-meter Reference RSSI value is to be measured in a chamber without any interference or need to measure once Beacon/Tag installed in the actual location.

I am bit confused, some websites will mention about pre-programming the 1-Meter reference RSSI value in factory and some websites will mention about calibrating the 1-Meter reference RSSI value once Beacon/Tag installed at actual location.

Please let me know the best approach.

Thanks & Regards
Vishnu Beema

davidgyoung, 2021-02-17:
The answer really depends on your use case. 

If you are looking for the most generic measurement that will be best applicable to a variety of possible placements of the beacon vs. the phone, use a chamber.  If your app is designed to work in one specific location with the user typically in one general direction relative to the beacon (or a number of similar placements), then measuring in-place will usually provide better results.

closed 2021-02-17 by davidgyoung

beemavishnu, 2021-02-22:
Thank you for your inputs