AltBeacon/android-beacon-library #1026: App not starting after phone reboot before Android ask for PIN

rakathal, 2021-05-05:
### Expected behavior
My App should run in background even after restarting the Phone before Android ask for PIN from the device protected storage area.

### Actual behavior
My program services are not able to run in the device protected storage which means after the boot before android ask for PIN, Debugger is getting lost some where in the alt beacon library and pausing the activity.

### Steps to reproduce this behavior
1. Create a BootBroadcast Receiver.
public class BootBroadcastReceiver extends BroadcastReceiver {

    private static final String TAG = "BootBroadcastReceiver";

    @Override
    public void onReceive(Context context, Intent intent) {
        boolean bootCompleted;
        Intent intent1;
        String action = intent.getAction();
        Log.i(TAG, "Received action: " + action + ", user unlocked: " + UserManagerCompat
                .isUserUnlocked(context));
        if (BuildCompat.isAtLeastN()) {
            bootCompleted = Intent.ACTION_LOCKED_BOOT_COMPLETED.equals(action);
        } else {
            bootCompleted = Intent.ACTION_BOOT_COMPLETED.equals(action);
        }
        if (!bootCompleted) {
            return;
        }
        Toast.makeText(context.getApplicationContext(), "Boot Complete event Received !",Toast.LENGTH_SHORT).show();
        Intent serviceIntent = new Intent(context, MyService.class);
        context.startService(serviceIntent);

    }
}

2. MyService.java
   public class MyService extends Service implements BeaconConsumer, RangeNotifier {
    private BeaconManager mBeaconManager;    

    @Override
    public void onCreate() {
        super.onCreate();
        //String input = intent.getStringExtra("inputExtra");
        Intent notificationIntent = new Intent(this, MainActivity.class);
        PendingIntent pendingIntent = PendingIntent.getActivity(this,
                0, notificationIntent, 0);
        Notification notification = new NotificationCompat.Builder(this, App.CHANNEL_ID)
                .setContentTitle("MyServices Service is running")
                .setContentText("")
                .setSmallIcon(R.drawable.ic_baseline_directions_walk)
                //.setSmallIcon(R.drawable.ic_myservices_48)
                .setContentIntent(pendingIntent)
                .build();
        startForeground(1, notification);
    }

    /**
     * Default onStartCommand
     *
     * @param intent
     * @param flags
     * @param startId
     * @return
     */
    @Override
    public int onStartCommand(Intent intent, int flags, int startId) {
        PowerManager mgr = (PowerManager) getSystemService(Context.POWER_SERVICE);
        wl = mgr.newWakeLock(PowerManager.PARTIAL_WAKE_LOCK, "MyService:MyWakeLock");
        wl.acquire();  
        createBluetoothInstance();
        return START_NOT_STICKY;
    }

    /**
     * On Stop button close the service
     */
    @Override
    public void onDestroy() {
        disconnectBluetooth(); // Disconnect bluetooth
        disconnectMQTT(); // Disconnect MQTT server
        wl.release(); // Release sleep lock
        super.onDestroy();
    }        

    /**
     * Inform Mobile Device to detect only EddyStone BLE signals
     */
    public void createBluetoothInstance() {
        mBeaconManager = BeaconManager.getInstanceForApplication(this);
        // We use Eddystone protocol, so we have to define here
        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_UID_LAYOUT));
        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_TLM_LAYOUT));
        // Binds this activity to the BeaconService
        mBeaconManager.bind(this);
    }

    @Nullable
    @Override
    public IBinder onBind(Intent intent) {
        return null;
    }

    /***
     * Start scanning the Beacons
     */
    @Override
    public void onBeaconServiceConnect() {
        // Encapsulates a beacon identifier of arbitrary byte length
        identifiers = new ArrayList<>();
        // Set null to indicate that we want to match beacons with any value
        identifiers.add(null);
        // Represents a criteria of fields used to match beacon
        region = new Region("AllBeaconsRegion", identifiers);
        try {
            // Tells the BeaconService to start looking for beacons that match the passed Region object
            //Log.i("","Inside Service Connect");
            mBeaconManager.startRangingBeaconsInRegion(region);
        } catch (RemoteException e) {
            e.printStackTrace();
        }
        // Specifies a class that should be called each time the BeaconService gets ranging data, once per second by default
        mBeaconManager.addRangeNotifier(this);
    }

    /**
     * Collect connection from all the beacons around within the defined region
     *
     * @param beacons
     * @param region
     */
    @Override
    public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
        try {
            if (beacons.size() > 0) {                
                for (Beacon beacon : beacons) {
                    if (beacon.getServiceUuid() == 0xfeaa && beacon.getBeaconTypeCode() == 0x00) {
                        // This is a Eddystone-UID frame
                        //Identifier namespaceId = beacon.getId1();
                        //Identifier instanceId = beacon.getId2();
                        int rssi = beacon.getRssi();
                        String Address = beacon.getBluetoothAddress().toLowerCase().replace(":", "");
                        /*//// Do we have telemetry data?
                        if (beacon.getExtraDataFields().size() > 0) {
                            long batteryMilliVolts = beacon.getExtraDataFields().get(1);
                            long telemetryVersion = beacon.getExtraDataFields().get(0);
                            long pduCount = beacon.getExtraDataFields().get(3);
                            long uptime = beacon.getExtraDataFields().get(4);
                        }*/
                        JSONObject jobj3 = new JSONObject();
                        jobj3.put("Mac", Address);
                        jobj3.put("rssi", rssi);
                        jobj3.put("Edt", df.format(Calendar.getInstance().getTime()));
                        if (beacon.getExtraDataFields().size() > 0) {
                            jobj3.put("BVolt", beacon.getExtraDataFields().get(1));
                        } else {
                            jobj3.put("BVolt", 0);
                        }
                        Jarr.put(jobj3);
                    }
                }
                jobj.put("BCNS", Jarr);
                String strLstScn = "Last scan: " + df.format(Calendar.getInstance().getTime());
                sendBroadcastMessage(strLstScn);
                publishToMQTT(jobj);
            }            
        } catch (Exception e) {
            e.printStackTrace();
        }
    }        

    /**
     * Disconnect Bluetooth
     */
    public void disconnectBluetooth() {
        try {
            mBeaconManager.stopRangingBeaconsInRegion(region);
        } catch (RemoteException e) {
            e.printStackTrace();
        }
        mBeaconManager.removeRangeNotifier(this);
    }    

Debugger is not able to come to didRangeBeaconsInRegion ?

### Mobile device model and OS version
Samsung X-Cover Pro and Android Version 11

### Android Beacon Library version
org.altbeacon:android-beacon-library:2.18

**This code is working in Android 7.1 but not working in Android 8,9,10,11**

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2021-05-05:
The library uses device storage to persist regions state so that if the app restarts it will remember the last state of each monitored beacon region.  This prevents spurious didEnterRegion callbacks when the app restarts.

The above mechanism is problematic if you are using the library to automatically start at boot and you want it to continue executing even though the user has not unlocked the device.  The OS will block reading this persisted state from storage.

Two possible solutions:

1. Disable region state persistence with `BeaconManager.getInstanceForApplication(this).setRegionStatePeristenceEnabled(false);`. You must do this before any other library functions.
2. Modify the library to access device encrypted storage as described here: https://developer.android.com/training/articles/direct-boot

Since this is a relatively unusual use of the library (the is the first issue I have heard reported in 8 years), option 2 is not a high priority for the open source project.  However, if you are interested in working on it I can advise where to make the changes and would welcome a pull request.

rakathal, 2021-05-05:
I tried with the first option and its not working for me. I would like to try second option, please let me know where to make the changes in the library.

davidgyoung, 2021-05-06:
@rakathal  please take a look at these lines of code:

https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/java/org/altbeacon/beacon/service/MonitoringStatus.java#L211-L264

You will see there how the library uses the Android filesystem is used to restore region state.  This is what will need to be adjusted to work in a pre-unlocked state.

One thing I noted when looking at this is that the filesystem timestamp is checked *before* the code bails out if you disable region state persistence here: https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/java/org/altbeacon/beacon/service/MonitoringStatus.java#L144-L148

This is what could be causing the problem where disabling region state persistence doesn't fix it for you.  Simply recording the lines in this method so it works like this might make it so that disabling region state persistence does solve the issue for you:

```
    private void restoreOrInitializeMonitoringStatus() {
        mRegionsStatesMap = new ConcurrentHashMap<Region, RegionMonitoringState>();
        if (!mStatePreservationIsOn) {
            LogManager.d(TAG, "Not restoring monitoring state because persistence is disabled");
            return;
        }
        long millisSinceLastMonitor = System.currentTimeMillis() - getLastMonitoringStatusUpdateTime();
        if (millisSinceLastMonitor > MAX_STATUS_PRESERVATION_FILE_AGE_TO_RESTORE_SECS * 1000) {
            LogManager.d(TAG, "Not restoring monitoring state because it was recorded too many milliseconds ago: "+millisSinceLastMonitor);
        }
        else {
            restoreMonitoringStatus();
            LogManager.d(TAG, "Done restoring monitoring status");
        }
    }
```

rakathal, 2021-05-07:
I am still unable to get the call back at didRangeBeaconsInRegion() method ? As soon as I unlock my the phone my app automatically start, but that's now I want.

My app is running fine in Android 7.1

davidgyoung, 2021-05-08:
You might try capturing a log after setting beaconManager.setDebug(true); to see where it hangs

rakathal, 2021-05-10:
I think by looking at this error, library is trying to save the file on credential protected storage, which is not getting retrieved on device protected storage and hence we are ending up with below error.
I am getting this error:
2021-05-10 13:02:52.799 5697-7858/com.MyOrg.MyLocation W/ContextImpl: Failed to ensure /data/user/0/com.MyOrg.MyLocation/files: mkdir failed: errno 126 (Required key not available)
2021-05-10 13:02:52.800 5697-7858/com.MyOrg.MyLocation W/ContextImpl: Failed to ensure /data/user/0/com.MyOrg.MyLocation/files: mkdir failed: errno 126 (Required key not available)
2021-05-10 13:02:52.801 5697-7858/com.MyOrg.MyLocation D/ScanState: Temp file is /data/user/0/com.MyOrg.MyLocation/files/android-beacon-library-scan-state-temp
2021-05-10 13:02:52.801 5697-7858/com.MyOrg.MyLocation D/ScanState: Perm file is /data/user/0/com.MyOrg.MyLocation/files/android-beacon-library-scan-state
2021-05-10 13:02:52.802 5697-7858/com.MyOrg.MyLocation E/ScanState: Error while saving scan status to file: Cannot delete existing file.
2021-05-10 13:02:52.802 5697-7858/com.MyOrg.MyLocation E/ScanState: Error while saving scan status to file: Cannot rename temp file.
2021-05-10 13:02:52.802 5697-7858/com.MyOrg.MyLocation D/ScanJob: ScanJob Lifecycle STOP (start fail): org.altbeacon.beacon.service.ScanJob@62e129f
2021-05-10 13:02:53.655 5697-5962/com.MyOrg.MyLocation D/DistanceConfigFetcher: response code is 200
2021-05-10 13:02:53.667 5697-5962/com.MyOrg.MyLocation D/ModelSpecificDistanceCalculator: Successfully downloaded distance models from online database
2021-05-10 13:02:53.681 5697-5962/com.MyOrg.MyLocation W/ContextImpl: Failed to ensure /data/user/0/com.MyOrg.MyLocation/files: mkdir failed: errno 126 (Required key not available)
2021-05-10 13:02:53.682 5697-5962/com.MyOrg.MyLocation W/FileUtils: Failed to chmod(/data/user/0/com.MyOrg.MyLocation/files): android.system.ErrnoException: chmod failed: ENOENT (No such file or directory)
2021-05-10 13:02:53.696 5697-5962/com.MyOrg.MyLocation W/ModelSpecificDistanceCalculator: Cannot write updated distance model to local storage
    java.io.FileNotFoundException: /data/user/0/com.MyOrg.MyLocation/files/model-distance-calculations.json: open failed: ENOENT (No such file or directory)
        at libcore.io.IoBridge.open(IoBridge.java:492)
        at java.io.FileOutputStream.<init>(FileOutputStream.java:236)
        at android.app.ContextImpl.openFileOutput(ContextImpl.java:694)
        at android.content.ContextWrapper.openFileOutput(ContextWrapper.java:232)
        at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.saveJson(ModelSpecificDistanceCalculator.java:207)
        at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.access$100(ModelSpecificDistanceCalculator.java:47)
        at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator$1.onComplete(ModelSpecificDistanceCalculator.java:249)
        at org.altbeacon.beacon.distance.ModelSpecificDistanceUpdater.doInBackground(ModelSpecificDistanceUpdater.java:30)
        at org.altbeacon.beacon.distance.ModelSpecificDistanceUpdater.doInBackground(ModelSpecificDistanceUpdater.java:16)
        at android.os.AsyncTask$3.call(AsyncTask.java:394)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:923)

This is the file and context is supposed to be device protected storage. However I am not sure where all other placed I need to do the modification in the library. Or if you can suggest the best way of doing it:

Please refer to the link below for moving the files to device encrypted storage:
https://lampham.medium.com/device-encrypted-storage-and-direct-boot-mode-in-android-6e5e25d173f3

[image]

davidgyoung, 2021-05-10:
Oh, that is a different place entirely.  I didn't think about that.

OK, rather than fixing how it saves this info, the easiest solution is to disable doing this entirely with: `beaconManager.setDistanceModelUpdateUrl("invalid");`

This will keep it from trying to save to that location in the first place.

rakathal, 2021-05-10:
Thanks David, after adding the below mentioned that you have suggested, my program proceeded without any errors and scanning is happening as per what I see from the rolling logs. 
public void createBluetoothInstance() {
        mBeaconManager = BeaconManager.getInstanceForApplication(this);
        **mBeaconManager.setRegionStatePersistenceEnabled(false);**
        mBeaconManager.setDebug(true);
        **mBeaconManager.setDistanceModelUpdateUrl("invalid");
        mBeaconManager.setEnableScheduledScanJobs(false);**
        // We use Eddystone protocol, so we have to define here
        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_UID_LAYOUT));
        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_TLM_LAYOUT));
        // Binds this activity to the BeaconService
        mBeaconManager.bind(this);
    }

However, the callback to didRangeBeaconsInRegion() is not happening. I think we are pretty close to the solution. Could you please guide us from here on how to get the callback.

davidgyoung, 2021-05-10:
Glad that worked!

I suspect the failure to detect beacons is an unrelated issue -- a pretty common one, too.  I put together a troubleshooting checklist.  The problem could be *anything* on this checklist.  https://altbeacon.github.io/android-beacon-library/detection-trouble.html

rakathal, 2021-05-11:
I really appreciate the support that your are doing.
I went through your troubleshooting checklist. Everything looks good and working while I am running the application in **unlocked** mode.

Also Today I am able to received the callback to didRangeBeaconsInRegion(), but now I am not receiving any beacons data in didRangeBeaconsInRegion method after the reboot before android ask for PIN. beacon.size() is giving me zero.

Do you know what could be the possible issue ?
Here is my code in my service class:
@Override
    public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
        try {
            if (beacons.size() > 0) {                
                Toast.makeText(getApplicationContext(), "Published!"+beacons.size(),Toast.LENGTH_SHORT).show();
            } else // If there are no beacons available in the vicinity
            {                
                Toast.makeText(getApplicationContext(), "UnPublished !"+beacons.size(),Toast.LENGTH_SHORT).show();
            }
        } catch (Exception e) {
            e.printStackTrace();
        }
    }

AndroidManifest.xml
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.example.MyApp">
    <uses-permission android:name="android.permission.RECEIVE_BOOT_COMPLETED" />
    <uses-permission android:name="android.permission.VIBRATE" />
    <uses-permission android:name="com.samsung.android.knox.permission.EVENT_BOOT_COMPLETED" />

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.WAKE_LOCK" />
    <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
    <uses-permission android:name="android.permission.READ_PHONE_STATE" />
    <uses-permission android:name="android.permission.ACCESS_WIFI_STATE"/>
    <uses-permission android:name="android.permission.READ_PHONE_STATE" />
    <uses-permission android:name="android.permission.FOREGROUND_SERVICE"></uses-permission>
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

    <application
        android:name="com.example.MyApp.App"
        android:allowBackup="true"
        android:icon="@mipmap/ic_launcher"
        android:label="MyApp"
        android:roundIcon="@mipmap/ic_launcher_round"
        android:networkSecurityConfig="@xml/network_security_config"
        android:usesCleartextTraffic="true"
        android:supportsRtl="true"
        android:theme="@style/AppTheme">
        <activity android:name="com.example.MyApp.MainActivity" android:directBootAware="true">
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />

                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>
        </activity>
        <receiver android:name=".BootBroadcastReceiver"
            android:exported="false"
            android:directBootAware="true">
            
            <intent-filter>
                <action android:name="android.intent.action.LOCKED_BOOT_COMPLETED" />
                <action android:name="android.intent.action.BOOT_COMPLETED" />
            </intent-filter>
        </receiver>
        <service android:name="org.eclipse.paho.android.service.MqttService"
[... 28 lines cut ...]
        //String input = intent.getStringExtra("inputExtra");
        Intent notificationIntent = new Intent(this, MainActivity.class);
        PendingIntent pendingIntent = PendingIntent.getActivity(this,
                0, notificationIntent, 0);
        Notification notification = new NotificationCompat.Builder(this, App.CHANNEL_ID)
                .setContentTitle("MyService is running")
                .setContentText("")
                .setSmallIcon(R.drawable.ic_baseline_directions_walk)
                //.setSmallIcon(R.drawable.ic_iinroute_48)
                .setContentIntent(pendingIntent)
                .build();
        mBeaconManager = BeaconManager.getInstanceForApplication(this);
        mBeaconManager.setRegionStatePersistenceEnabled(false);
        mBeaconManager.setDebug(true);
        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_UID_LAYOUT));
        mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_TLM_LAYOUT));
        mBeaconManager.setDistanceModelUpdateUrl("invalid");
        mBeaconManager.setEnableScheduledScanJobs(false);
        mBeaconManager.setBackgroundMode(true);
        mBeaconManager.setBackgroundBetweenScanPeriod(0);
        mBeaconManager.setBackgroundScanPeriod(1100);
        mBeaconManager.enableForegroundServiceScanning(notification, 456);
    }

Is the statement sequence in the above method correct ? Am I missing anything in the above statement.

rakathal, 2021-05-13:
I also see that the library AndroidMenifest does't have statement to start the library program execution on 
**<action android:name="android.intent.action.LOCKED_BOOT_COMPLETED"/>**
event.
Do you think, if we add this statement in Library AndroidMenifest.xml, will my app going to receive the beacon.size() after the boot before entering the PIN screen?

rakathal, 2021-06-22:
Finally I made it work, All thanks to you David !! 
Thanks for giving me an opportunity to understand and learn the internals of this library. This didn't just give me knowledge about this issue, but also helped me to learn and explored different areas of android develpment.

The issue was - To autostart the app before android ask for the PIN, in Andorid 10 and above version we have to start the foreground services to continuous run the background task. 
And to auto start the foreground services, we have to start the service in broadcast receiver like this - 
```
Intent serviceIntent = new Intent(context.getApplicationContext(), MyLocationService.class);
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
            context.getApplicationContext().**startForegroundService**(serviceIntent); // This will work.
        }
`
instead of this - 
```
Intent serviceIntent = new Intent(context.getApplicationContext(), MyLocationService.class);
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
            context.getApplicationContext().**startService**(serviceIntent); // This doesn't work on android 10 and abvoe.
        }
```
Along with the we have to start the foreground service in in the service class file in OnCreate Method.
`
BeaconManager.getInstanceForApplication(this).setRegionStatePersistenceEnabled(false);
mBeaconManager = BeaconManager.getInstanceForApplication(this);
mBeaconManager.setDistanceModelUpdateUrl("invalid");
startForeground(1,notification);
`
And onStart Method will look like this :
```
onStartCommand() {
        mBeaconManager.setEnableScheduledScanJobs(false);
        mBeaconManager.setBackgroundBetweenScanPeriod(0);
        mBeaconManager.setBackgroundScanPeriod(1100);
}
```
This will enable to receive the callback to didRangeBeaconsInRegion() method with correct count.

I know that I struggled a lot for this issue for almost 1.5 month, however I don't want my other developer friends on the internet to struggle for it.

closed 2021-06-22 by rakathal