AltBeacon/android-beacon-library #1028: Not allowed to start service Intent: BeaconIntentProcessor

romaomb, 2021-05-12:
### Expected behavior

App should receive `didEnterRegion` when waking up in background.

### Actual behavior

As soon as the process starts, the app throws an exception on `Callback.java:61`: 
```
Failed attempting to start service: org.altbeacon.beaconreference/org.altbeacon.beacon.BeaconIntentProcessor
```

After building the library locally, checking out 2.18 tag, here's the stacktrace:
```
2021-05-12 03:17:57.806 5919-5949/org.altbeacon.beaconreference E/AndroidRuntime: FATAL EXCEPTION: pool-3-thread-2
    Process: org.altbeacon.beaconreference, PID: 5919
    java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$4.done(AsyncTask.java:415)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
        at java.util.concurrent.FutureTask.run(FutureTask.java:271)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:923)
     Caused by: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.BeaconIntentProcessor (has extras) }: app is in background uid UidRecord{48b9abe u0a909 TRNB idle procs:1 seq(0,0,0)}
        at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1791)
        at android.app.ContextImpl.startService(ContextImpl.java:1736)
        at android.content.ContextWrapper.startService(ContextWrapper.java:738)
        at org.altbeacon.beacon.service.Callback.call(Callback.java:60)
        at org.altbeacon.beacon.service.MonitoringStatus.updateNewlyInsideInRegionsContaining(MonitoringStatus.java:124)
        at org.altbeacon.beacon.service.ScanHelper.processBeaconFromScan(ScanHelper.java:343)
        at org.altbeacon.beacon.service.ScanHelper.access$200(ScanHelper.java:62)
        at org.altbeacon.beacon.service.ScanHelper$ScanProcessor.doInBackground(ScanHelper.java:420)
        at org.altbeacon.beacon.service.ScanHelper$ScanProcessor.doInBackground(ScanHelper.java:386)
        at android.os.AsyncTask$3.call(AsyncTask.java:394)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:923) 
2021-05-12 03:17:57.816 5919-5949/org.altbeacon.beaconreference I/Process: Sending signal. PID: 5919 SIG: 9
```

Full log:
<details>
  <summary>Click to expand</summary>

```
2021-05-12 03:17:57.257 5919-5919/? I/beaconreferenc: Late-enabling -Xcheck:jni
2021-05-12 03:17:57.290 5919-5919/? I/beaconreferenc: Unquickening 20 vdex files!
2021-05-12 03:17:57.302 5919-5919/? D/ActivityThread: setConscryptValidator
2021-05-12 03:17:57.302 5919-5919/? D/ActivityThread: setConscryptValidator - put
2021-05-12 03:17:57.402 5919-5919/org.altbeacon.beaconreference D/NetworkSecurityConfig: No Network Security Config specified, using platform default
2021-05-12 03:17:57.441 5919-5919/org.altbeacon.beaconreference D/NetworkSecurityConfig: No Network Security Config specified, using platform default
2021-05-12 03:17:57.452 5919-5919/org.altbeacon.beaconreference I/BeaconManager: BeaconManager started up on pid 5919 named 'org.altbeacon.beaconreference' for application package 'org.altbeacon.beaconreference'.  isMainProcess=true
2021-05-12 03:17:57.457 5919-5919/org.altbeacon.beaconreference D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
2021-05-12 03:17:57.460 5919-5919/org.altbeacon.beaconreference D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2021-05-12 03:17:57.474 5919-5919/org.altbeacon.beaconreference I/BluetoothTestJob: Using BluetoothTestJob JobId from manifest: 1799803768
2021-05-12 03:17:57.501 5919-5919/org.altbeacon.beaconreference W/BeaconManager: The BeaconManager is not bound to the service.  Call beaconManager.bind(BeaconConsumer consumer) and wait for a callback to onBeaconServiceConnect()
2021-05-12 03:17:57.613 5919-5919/org.altbeacon.beaconreference I/ScanJob: Using immediateScanJobId from manifest: 208352939
2021-05-12 03:17:57.618 5919-5919/org.altbeacon.beaconreference I/ScanJob: Using periodicScanJobId from manifest: 208352940
2021-05-12 03:17:57.619 5919-5919/org.altbeacon.beaconreference W/JobInfo: Requested interval +5m10s0ms for job 208352940 is too small; raising to +15m0s0ms
2021-05-12 03:17:57.619 5919-5919/org.altbeacon.beaconreference W/JobInfo: Requested flex 0 for job 208352940 is too small; raising to +5m0s0ms
2021-05-12 03:17:57.638 5919-5919/org.altbeacon.beaconreference D/BeaconReference: didDetermineStateForRegion | state: 0 region: id1: null id2: null id3: null
2021-05-12 03:17:57.679 5919-5919/org.altbeacon.beaconreference I/ScanJob: Using immediateScanJobId from manifest: 208352939
2021-05-12 03:17:57.684 5919-5919/org.altbeacon.beaconreference I/ScanJob: Using periodicScanJobId from manifest: 208352940
2021-05-12 03:17:57.684 5919-5919/org.altbeacon.beaconreference W/JobInfo: Requested interval +5m10s0ms for job 208352940 is too small; raising to +15m0s0ms
2021-05-12 03:17:57.684 5919-5919/org.altbeacon.beaconreference W/JobInfo: Requested flex 0 for job 208352940 is too small; raising to +5m0s0ms
2021-05-12 03:17:57.727 5919-5943/org.altbeacon.beaconreference I/BluetoothTestJob: Bluetooth Test Job running
2021-05-12 03:17:57.727 5919-5943/org.altbeacon.beaconreference I/BluetoothMedic: Starting scan test
2021-05-12 03:17:57.731 5919-5943/org.altbeacon.beaconreference D/BluetoothAdapter: STATE_ON
2021-05-12 03:17:57.732 5919-5943/org.altbeacon.beaconreference D/BluetoothAdapter: STATE_ON
[... 39 lines cut ...]
        at java.lang.Thread.run(Thread.java:923) 
2021-05-12 03:17:57.816 5919-5949/org.altbeacon.beaconreference I/Process: Sending signal. PID: 5919 SIG: 9
```

</details>

The only thing that I did change was commenting the `BackgroundPowerSaver` instatiation and, inside `didDetermineStateForRegion`, I also commented the block responsible for start ranging, since I'm just looking for monitoring.

### Steps to reproduce this behavior

1. Clone either [kotlin](https://github.com/davidgyoung/android-beacon-library-reference-kotlin) or [java](https://github.com/AltBeacon/android-beacon-library-reference) version of the reference app
2. Run the app and allow access to location all the time
3. Close the app
4. Start advertising iBeacon

### Mobile device model and OS version

Samsung S21+
Android 11

### Android Beacon Library version

2.18

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

romaomb, 2021-05-12:
Debug logs:

<details>
  <summary>Click to expand</summary>

```
2021-05-12 03:41:23.059 12132-12132/? I/beaconreferenc: Late-enabling -Xcheck:jni
2021-05-12 03:41:23.093 12132-12132/? I/beaconreferenc: Unquickening 20 vdex files!
2021-05-12 03:41:23.104 12132-12132/? D/ActivityThread: setConscryptValidator
2021-05-12 03:41:23.104 12132-12132/? D/ActivityThread: setConscryptValidator - put
2021-05-12 03:41:23.225 12132-12132/org.altbeacon.beaconreference D/NetworkSecurityConfig: No Network Security Config specified, using platform default
2021-05-12 03:41:23.268 12132-12132/org.altbeacon.beaconreference D/NetworkSecurityConfig: No Network Security Config specified, using platform default
2021-05-12 03:41:23.278 12132-12132/org.altbeacon.beaconreference I/BeaconManager: BeaconManager started up on pid 12132 named 'org.altbeacon.beaconreference' for application package 'org.altbeacon.beaconreference'.  isMainProcess=true
2021-05-12 03:41:23.282 12132-12132/org.altbeacon.beaconreference D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
2021-05-12 03:41:23.284 12132-12132/org.altbeacon.beaconreference D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2021-05-12 03:41:23.292 12132-12132/org.altbeacon.beaconreference D/BluetoothMedic: Medic monitoring for transmission and scan failure notifications
2021-05-12 03:41:23.292 12132-12132/org.altbeacon.beaconreference D/BluetoothMedic: Medic scheduling periodic tests of types 3
2021-05-12 03:41:23.294 12132-12132/org.altbeacon.beaconreference I/BluetoothTestJob: Using BluetoothTestJob JobId from manifest: 1799803768
2021-05-12 03:41:23.321 12132-12132/org.altbeacon.beaconreference D/BeaconManager: updating background flag to true
2021-05-12 03:41:23.322 12132-12132/org.altbeacon.beaconreference D/BeaconManager: updating scan period to 10000, 300000
2021-05-12 03:41:23.322 12132-12132/org.altbeacon.beaconreference W/BeaconManager: The BeaconManager is not bound to the service.  Call beaconManager.bind(BeaconConsumer consumer) and wait for a callback to onBeaconServiceConnect()
2021-05-12 03:41:23.323 12132-12132/org.altbeacon.beaconreference D/BeaconManager: This consumer is not bound.  Binding now: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@ec23084
2021-05-12 03:41:23.323 12132-12132/org.altbeacon.beaconreference D/BeaconManager: Not starting beacon scanning service. Using scheduled jobs
2021-05-12 03:41:23.323 12132-12132/org.altbeacon.beaconreference D/AppStarter: Activating background region monitoring
2021-05-12 03:41:23.324 12132-12132/org.altbeacon.beaconreference D/AppStarter: Background region monitoring activated for region id1: null id2: null id3: null
2021-05-12 03:41:23.325 12132-12132/org.altbeacon.beaconreference D/BeaconManager: callback packageName: org.altbeacon.beaconreference
2021-05-12 03:41:23.346 12132-12132/org.altbeacon.beaconreference D/MonitoringStatus: Restored region monitoring state for 1 regions.
2021-05-12 03:41:23.346 12132-12132/org.altbeacon.beaconreference D/MonitoringStatus: Region  id1: null id2: null id3: null uniqueId: wildcard-region state: org.altbeacon.beacon.service.RegionMonitoringState@8ed28ee
2021-05-12 03:41:23.346 12132-12132/org.altbeacon.beaconreference D/MonitoringStatus: Done restoring monitoring status
2021-05-12 03:41:23.347 12132-12132/org.altbeacon.beaconreference D/MonitoringStatus: saveMonitoringStatusIfOn()
2021-05-12 03:41:23.352 12132-12132/org.altbeacon.beaconreference D/ScanJobScheduler: Applying settings to ScanJob
2021-05-12 03:41:23.403 12132-12132/org.altbeacon.beaconreference D/ScanState: Scan state restore regions: monitored=1 ranged=0
2021-05-12 03:41:23.404 12132-12132/org.altbeacon.beaconreference D/ScanState: ranged regions: old=0 new=0
2021-05-12 03:41:23.404 12132-12132/org.altbeacon.beaconreference D/ScanState: monitored regions: old=1 new=1
2021-05-12 03:41:23.404 12132-12132/org.altbeacon.beaconreference D/ScanState: Updated state with 0 ranging regions and 1 monitoring regions.
2021-05-12 03:41:23.419 12132-12132/org.altbeacon.beaconreference D/ScanState: Temp file is /data/user/0/org.altbeacon.beaconreference/files/android-beacon-library-scan-state-temp
2021-05-12 03:41:23.420 12132-12132/org.altbeacon.beaconreference D/ScanState: Perm file is /data/user/0/org.altbeacon.beaconreference/files/android-beacon-library-scan-state
2021-05-12 03:41:23.420 12132-12132/org.altbeacon.beaconreference D/MonitoringStatus: saveMonitoringStatusIfOn()
2021-05-12 03:41:23.423 12132-12132/org.altbeacon.beaconreference D/ScanJobScheduler: Applying scan job settings with background mode true
2021-05-12 03:41:23.423 12132-12132/org.altbeacon.beaconreference D/ScanJobScheduler: This is the first time we schedule a job and we are in background, set immediate scan flag to true in order to trigger the HW filter install.
2021-05-12 03:41:23.423 12132-12132/org.altbeacon.beaconreference D/BeaconLocalBroadcastProcessor: Register calls: global=1
2021-05-12 03:41:23.423 12132-12132/org.altbeacon.beaconreference D/ScanState: ScanState says background mode for ScanJob is true
2021-05-12 03:41:23.423 12132-12132/org.altbeacon.beaconreference D/ScanJobScheduler: We just woke up in the background based on a new scan result or first run of the app. Start scan job immediately.
2021-05-12 03:41:23.424 12132-12132/org.altbeacon.beaconreference D/ScanJobScheduler: Scheduling immediate ScanJob to run in 0 millis
[... cut ...]
2021-05-12 03:41:23.635 12132-12163/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Making new Android L scanner
2021-05-12 03:41:23.636 12132-12163/org.altbeacon.beaconreference D/BluetoothAdapter: STATE_ON
2021-05-12 03:41:23.637 12132-12164/org.altbeacon.beaconreference D/BluetoothAdapter: STATE_ON
2021-05-12 03:41:23.637 12132-12163/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 10000 milliseconds
2021-05-12 03:41:23.638 12132-12165/org.altbeacon.beaconreference I/Process: Sending signal. PID: 12132 SIG: 9
```

</details>

davidgyoung, 2021-06-07:
I reproduced this today, and put in a fix as part of #1030

closed 2021-06-07 by davidgyoung