AltBeacon/android-beacon-library #1029: Unable to build on Android 12

DarrenMiddleton, 2021-06-02:
I went to upgrade my app to work for Android 12 and got the error listed below. 

`android Manifest merger failed : Apps targeting Android 12 and higher are required to specify an explicit value for android:exported`

Ref: https://developer.android.com/about/versions/12/behavior-changes-12#security

I looked at what Android manifest was causing this issue and saw the code below. 

```
<service
     android:name="org.altbeacon.beacon.service.ScanJob"
     android:permission="android.permission.BIND_JOB_SERVICE" >
        <meta-data
         android:name="immediateScanJobId"
         android:value="208352939" />
            <meta-data
                android:name="periodicScanJobId"
                android:value="208352940" />
        </service>
        <service
            android:name="org.altbeacon.bluetooth.BluetoothTestJob"
            android:permission="android.permission.BIND_JOB_SERVICE" >
            <meta-data
                android:name="jobId"
                android:value="1799803768" />
        </service>
```

These two services are not within an activity that has the exported tag and they don't have the exported tag themselves which is required for Android 12.

davidgyoung, 2021-06-02:
Thanks for this report.  Yes, this needs to be fixed in the library.

In the meantime you can make it in your app work by adding entries to your app’s manifest to correct this like this:

<service android:name="org.altbeacon.beacon.service.ScanJob” android:exported="false" tools:replace="exported" />

<service android:name="org.altbeacon.bluetooth.BluetoothTestJob” android:exported="false" tools:replace="exported" />

davidgyoung, 2021-06-07:
Fixed in #1033

closed 2021-06-07 by davidgyoung

phattt-android-developer, 2021-06-17:
I also have the same issue when building on android 12
```06/17 13:57:16: Launching 'akerun' on Google Pixel 3.
Installation did not succeed.
The application could not be installed: INSTALL_PARSE_FAILED_MANIFEST_MALFORMED

List of apks:
[0] '/debug.apk'
Installation failed due to: 'Failed to commit install session XXX with command cmd package install-commit XXX. Error: INSTALL_PARSE_FAILED_MANIFEST_MALFORMED: Failed parse during installPackageLI: /base.apk (at Binary XML file line #615): org.altbeacon.beacon.startup.StartupBroadcastReceiver: Targeting S+ (version 10000 and above) requires that an explicit value for android:exported be defined when intent filters are present'
Retry
Failed to launch an application on all devices
```
https://developer.android.com/about/versions/12/behavior-changes-12#exported

davidgyoung, 2021-06-17:
Please try with 2.19-beta2 (available on Maven central) which has the fix for this problem and report back if you still see the issue.

tanphatstqn, 2021-06-17:
> Please try with 2.19-beta2 (available on Maven central) which has the fix for this problem and report back if you still see the issue.

Thank you for responding.
I tried again with version `2.19-beta2`, but they don't work:
```Installation failed due to: 'Failed to commit install session 1551719400 with command cmd package install-commit 1551719400. Error: INSTALL_PARSE_FAILED_MANIFEST_MALFORMED: Failed parse during installPackageLI: /data/app/vmdl1551719400.tmp/base.apk (at Binary XML file line #449): org.altbeacon.beacon.startup.StartupBroadcastReceiver: Targeting S+ (version 10000 and above) requires that an explicit value for android:exported be defined when intent filters are present'```

Looks like `exported` value has not been added in `AndroidManifest.xml` file:
[image]

Please let me know if I missed something. Thank you.