AltBeacon/android-beacon-library #1036: Outdated `BeaconConsumer` code sample in documentation

Whathecode, 2021-07-05:
The code sample in the [`BeaconManager`](https://altbeacon.github.io/android-beacon-library/javadoc/reference/org/altbeacon/beacon/BeaconManager.html) and [`BeaconConsumer` documention](https://altbeacon.github.io/android-beacon-library/javadoc/reference/org/altbeacon/beacon/BeaconConsumer.html) seems to be outdated.

It overrides different methods than defined on the interface.

davidgyoung, 2021-07-05:
@Whathecode, I agree the examples are confusing, but I do not think they are incorrect.  

The examples are confusing because they both show an Android `Activity` instance that implements the `BeaconConsumer` interface.  The methods shown with the ` @Override` annotation aren't necessarily overriding `BeaconConsumer` but instead overriding methods the `Activity` base class instance.

**Example methods that override Activity method:**
`protected void onDestroy()`
`protected void onCreate(Bundle savedInstanceState)`

**Example methods that override BeaconConsumer method:**
`abstract void	onBeaconServiceConnect()`

**Activity methods that implement BeaconConsumer methods (these are invisible):**
`abstract boolean	bindService(Intent intent, ServiceConnection connection, int mode)`
`abstract Context	getApplicationContext()`
`abstract void	unbindService(ServiceConnection connection)`

Does this make sense?

Rather than clarify the documentation, the plan for the library is to deprecate `BeaconConsumer` in a future release along with `BeaconManager#bind`, `BeaconManager#unbind`, and `RegionBootstrap` in  favor of auto-bind alternatives described in #1035 and available in the 2.19-beta3 release.  These changes will simplify the API and make `BeaconConsumer` unnecessary.

closed 2021-07-05 by davidgyoung

Whathecode, 2021-07-05:
> **Activity methods that implement BeaconConsumer methods:** (these are invisible because they are part of the class definition for `Activity`)
> `abstract boolean bindService(Intent intent, ServiceConnection connection, int mode)`
> `abstract Context getApplicationContext()`
> `abstract void unbindService(ServiceConnection connection)`

Thanks! This is what threw me off. In our codebase these were apparently implemented manually; we do not extend from `Activity`. I wonder what that indicates a code smell. 🤔

davidgyoung, 2021-07-05:
@Whathecode -- I would not worry about a code smell on your end -- if anything the code smell is with the design of the `BeaconConsumer` interface which assumes it is being used by an Android Activity or Application, which is why it has those extra methods that you have to implement if you are not using an Activity or Application.  This is one of the many reasons I want to get rid of it to simplify the API.