AltBeacon/android-beacon-library #1041: Unable to update the Android Beacon AD Type

karannassa, 2021-07-14:
When I tried to create a beacon by setting the `beaconTransmitter.setConnectable(true);` then app is generating the beacon but AD type is 02011A instead of 020106. I was expecting the following response:

02-01-06-12-FF-53-49-47-F2-34-45-4C-F6-6D-4A-0F-AD-F2-F4-91-1B-A9-00

```
02 # Number of bytes that follow in first AD structure
01 # Flags AD type
1A # Flags value 0x1A = 000011010  
   bit 0 (OFF) LE Limited Discoverable Mode
   bit 1 (ON) LE General Discoverable Mode
   bit 2 (OFF) BR/EDR Not Supported
   bit 3 (ON) Simultaneous LE and BR/EDR to Same Device Capable (controller)
   bit 4 (ON) Simultaneous LE and BR/EDR to Same Device Capable (Host)
```

As per above info, I want to set AD flag type is 0x06 (BLE_GAP_ADV_FLAG_LE_GENERAL_DISC_MODE | BLE_GAP_ADV_FLAG_BR_EDR_NOT_SUPPORTED)

And, here is the code to generate beacon:

> Beacon beacon = new Beacon.Builder().setId1("7f234454-cf6d-4a0f-adf2-f4911ba9ffa6")
                                   .setManufacturer(0x4953)
                                   .build();
            BeaconParser beaconParser = new BeaconParser()
                    .setBeaconLayout("i:2-15,p:16-16");
            BeaconTransmitter beaconTransmitter = new BeaconTransmitter(_context, beaconParser);
            beaconTransmitter.setConnectable(true);
            beaconTransmitter.startAdvertising(beacon);

Can anyone please support me to change 0x1A with 0x06?

[image]

davidgyoung, 2021-07-14:
I do not think it is possible to make Android transmit the AD Flags as 0x06.  This would be:

   bit 0 (OFF) LE Limited Discoverable Mode
   bit 1 (ON) LE General Discoverable Mode
   bit 2 (ON) BR/EDR Not Supported
   bit 3 (OFF) Simultaneous LE and BR/EDR to Same Device Capable (controller)
   bit 4 (OFF) Simultaneous LE and BR/EDR to Same Device Capable (Host)

Android OS limits the values you can configure here, and it specifically does not expose any APIs to set the raw AD Flags byte value or to turn bit 2 on and bits 3 and 4 off.  This would be effectively misstating Android's capabilities as "BR/EDR Not Supported" which is not true.  Android just doesn't let you do that.

You can certainly do this with a lower level embedded platform like ESP32 or Nordic 52832, just not with Android.

closed 2021-07-14 by davidgyoung

karannassa, 2021-07-14:
> I do not think it is possible to make Android transmit the AD Flags as 0x06. This would be:
>
> bit 0 (OFF) LE Limited Discoverable Mode
> bit 1 (ON) LE General Discoverable Mode
> bit 2 (ON) BR/EDR Not [...]

Okay thanks @davidgyoung for the update.