AltBeacon/android-beacon-library #1048: Clarify material design dependency

liedQM, 2021-08-09:
### Expected behavior
A library without UI should not depend on a UI framework like the material design library.
https://material.io/develop/android/docs/getting-started

### Actual behavior
The library depends on `'com.google.android.material:material:1.3.0'`.

### Android Beacon Library version
2.17.1 didn't have a material design dependency
Starting from `2.18.0` the library depends on material design.

We are using the beacon scanner in on of our libraries which also have no UI and the change now enforces our library to also depend on an UI framework.

I like to ask here, why the library depends on the material design library without having any UI component?

liedQM, 2021-08-09:
I can't even find any usage of `com.google.android.material` in your repository:
[com.google.android.material&type=code](https://github.com/AltBeacon/android-beacon-library/search?q=com.google.android.material&type=code)

davidgyoung, 2021-08-09:
What makes you say “The library depends on 'com.google.android.material:material:1.3.0'”?  There is certainly no intention that this be true.

liedQM, 2021-08-09:
Your pom tells me otherwise:
https://repo1.maven.org/maven2/org/altbeacon/android-beacon-library/2.19/android-beacon-library-2.19.pom

```<dependency>
<groupId>com.google.android.material</groupId>
<artifactId>material</artifactId>
<version>1.3.0</version>
</dependency>```

davidgyoung, 2021-08-09:
Fixed in #1049

closed 2021-08-09 by davidgyoung