AltBeacon/android-beacon-library #1057: Library always detect Estimote White Beacon model to a distance lower than one meter

fwermus, 2021-09-06:
### Expected behavior
Estimote white models beacons distance should reflect the actual distance of the Estimote beacon from the android cell phone.

### Actual behavior
Estimote white models beacons distance are tested below a meter no matter where the beacon is.

### Steps to reproduce this behavior

Take a Estimote White model beacon to 25 mts and got a distance below a meter.

### Mobile device model and OS version

Moto G8 on Android 11.

### Android Beacon Library version

org.altbeacon:android-beacon-library:2+

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2021-09-06:
@fwermus, can you please use the Beacon Scope app (based on this library) to take a screen shot of what you see for both a distance estimate and Rssi at one meter for your phone and beacon?

https://play.google.com/store/apps/details?id=com.davidgyoungtech.beaconscanner&hl=en_US&gl=US

fwermus, 2021-09-09:
[image]
[image]
[image]

Next comment, at 1mts of distance

davidgyoung, 2021-09-09:
So I see several different beacon identifiers scanned.  Which one is the beacon at 1 meter away?  One other request, can you please tap on the specific row of the beacon that is one meter away and take a screen shot of the detail screen.

fwermus, 2021-09-11:
It shows two beacons but actually is one. For some reason, I don't know this estimote physical beacon shows two beacons. Some extra info about these beacons are that can emit iBeacon protocol as well as Estimote protocol at same time. I have sense this beacon at a distance of 1 mt.

[image]

davidgyoung, 2021-09-11:
The screenshot shows two beacon transmissions with a distance estimate well below 1 meter and a RSSI of -52 (which is typical at one meter).  This suggests to me that whatever is transmitting is improperly calibrated.  You will need to address this with Estimote so that the measured power configuration of the beacon is set properly.  This forum answer describes how it is supposed to work:  https://forums.estimote.com/t/broadcasting-power-rssi-and-measured-power-explained/1047

closed 2021-09-11 by davidgyoung