AltBeacon/android-beacon-library #1058: Passive beacon scans seem broken

carbonaro, 2021-09-20:
### Expected behavior

When entering passive scan mode (e.g. startScan called with the SCAN_MODE_LOW_POWER setting), upon beacon detection, `StartupBroadcastReceiver#onReceive` is called with an intent having the BluetoothLeScanner.EXTRA_CALLBACK_TYPE extra typically set to CALLBACK_TYPE_ALL_MATCHES or to CALLBACK_TYPE_FIRST_MATCH. Scan results can then be processed. 

### Actual behavior

On android version => Oreo, passive beacon scans seem broken. When the app goes into background, Scans started by `ScanHelper#startAndroidOBackgroundScan` are triggering callbacks to `StartupBroadcastReceiver#onReceive` but `intent.getIntExtra(BluetoothLeScanner.EXTRA_CALLBACK_TYPE, -1)` always returns -1.

Detections do work when the app is started in the foreground.

### Steps to reproduce this behavior

With the reference app, validate that a beacon gets detected in the foreground, remove battery from the beacon, wait for the didExitRegion event, put the reference app in the background and put the battery back in the beacon. 
Log shows the following :

```
2021-09-20 14:03:25.902 21646-21646/org.altbeacon.beaconreference D/ScanJobScheduler: Scheduling periodic ScanJob (job:208352940/org.altbeacon.beaconreference/org.altbeacon.beacon.service.ScanJob) to run every 310000 millis
2021-09-20 14:03:25.911 21646-21646/org.altbeacon.beaconreference D/ScanJob: onStopJob called
2021-09-20 14:03:25.913 21646-21646/org.altbeacon.beaconreference I/ScanJob: Using periodicScanJobId from manifest: 208352940
2021-09-20 14:03:25.913 21646-21646/org.altbeacon.beaconreference I/ScanJob: onStopJob called for immediate scan org.altbeacon.beacon.service.ScanJob@3d5d673
2021-09-20 14:03:25.913 21646-21646/org.altbeacon.beaconreference I/ScanJob: ScanJob Lifecycle STOP: org.altbeacon.beacon.service.ScanJob@3d5d673
2021-09-20 14:03:25.917 21646-21646/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
2021-09-20 14:03:25.921 21646-21646/org.altbeacon.beaconreference D/CycledLeScanner: stop called
2021-09-20 14:03:25.921 21646-21646/org.altbeacon.beaconreference D/CycledLeScanner: disabling scan
2021-09-20 14:03:25.922 21646-21646/org.altbeacon.beaconreference D/CycledLeScanner: Done with scan cycle
2021-09-20 14:03:25.922 21646-21714/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Stopping LE scan on scan handler
2021-09-20 14:03:25.922 21646-21646/org.altbeacon.beaconreference D/ScanHelper: Beacon simulator not enabled
2021-09-20 14:03:25.923 21646-21714/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
2021-09-20 14:03:25.923 21646-21646/org.altbeacon.beaconreference D/CycledLeScanner: Scanning disabled. 
2021-09-20 14:03:25.923 21646-21646/org.altbeacon.beaconreference D/CycledLeScanner: cancel wakeup alarm: PendingIntent{d30dc3a: android.os.BinderProxy@fc3a0eb}
2021-09-20 14:03:25.933 21646-21646/org.altbeacon.beaconreference D/CycledLeScanner: Set a wakeup alarm to go off in 9223372036348880842 ms: PendingIntent{d30dc3a: android.os.BinderProxy@fc3a0eb}
2021-09-20 14:03:25.933 21646-21646/org.altbeacon.beaconreference D/CycledLeScanner: Stopping scanning previously left on.
2021-09-20 14:03:25.933 21646-21646/org.altbeacon.beaconreference D/CycledLeScanner: stopping bluetooth le scan
2021-09-20 14:03:25.933 21646-21646/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Stopping scan
2021-09-20 14:03:25.934 21646-21646/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: unregistering SamsungScreenOffReceiver as we stop the cycled scanner
2021-09-20 14:03:25.934 21646-21714/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Stopping LE scan on scan handler
2021-09-20 14:03:25.934 21646-21646/org.altbeacon.beaconreference D/CycledLeScanner: Destroying
2021-09-20 14:03:25.934 21646-21714/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
2021-09-20 14:03:25.934 21646-21714/org.altbeacon.beaconreference D/BluetoothLeScanner: could not find callback wrapper
2021-09-20 14:03:25.934 21646-21646/org.altbeacon.beaconreference D/ScanJob: Scanning stopped
2021-09-20 14:03:25.935 21646-21714/org.altbeacon.beaconreference D/CycledLeScanner: Quitting scan thread
2021-09-20 14:03:25.935 21646-21646/org.altbeacon.beaconreference D/ScanJob: Checking to see if we need to start a passive scan
2021-09-20 14:03:25.935 21646-21646/org.altbeacon.beaconreference D/ScanFilterUtils: making scan filter for manufacturer: 76 [B@d44be38
2021-09-20 14:03:25.935 21646-21646/org.altbeacon.beaconreference D/ScanFilterUtils: Set up a scan filter: BluetoothLeScanFilter [mDeviceName=null, mDeviceAddress=null, mUuid=null, mUuidMask=null, mServiceSolicitationUuid=null, mServiceSolicitationUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=76, mManufacturerData=[2, 21], mManufacturerDataMask=[-1, -1]]
2021-09-20 14:03:25.938 21646-21646/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
2021-09-20 14:03:25.940 21646-21646/org.altbeacon.beaconreference D/ScanHelper: Started passive beacon scan
2021-09-20 14:03:25.948 21646-21646/org.altbeacon.beaconreference D/Choreographer: simon Chor doFrame skippedFrames = 2
2021-09-20 14:03:25.963 21646-21646/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
2021-09-20 14:03:25.964 21646-21749/org.altbeacon.beaconreference D/BeaconParser: Ignoring pdu type 01
2021-09-20 14:03:25.964 21646-21749/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 02011a0cff4c001307027651e34d76fc00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 15
2021-09-20 14:03:25.965 21646-21749/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting 02 15.  The bytes I see are: 02011a0cff4c001307027651e34d76fc00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
2021-09-20 14:03:25.985 21646-21646/org.altbeacon.beaconreference D/StartupBroadcastReceiver: onReceive called in startup broadcast receiver
2021-09-20 14:03:25.985 21646-21646/org.altbeacon.beaconreference D/StartupBroadcastReceiver: Already started.  Ignoring intent: Intent { flg=0x10 cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.startup.StartupBroadcastReceiver (has extras) } of type: null
2021-09-20 14:03:26.102 21646-21646/org.altbeacon.beaconreference D/StartupBroadcastReceiver: onReceive called in startup broadcast receiver
2021-09-20 14:03:26.102 21646-21646/org.altbeacon.beaconreference D/StartupBroadcastReceiver: Already started.  Ignoring intent: Intent { flg=0x10 cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.startup.StartupBroadcastReceiver (has extras) } of type: null
2021-09-20 14:03:26.345 21646-21646/org.altbeacon.beaconreference D/StartupBroadcastReceiver: onReceive called in startup broadcast receiver
2021-09-20 14:03:26.346 21646-21646/org.altbeacon.beaconreference D/StartupBroadcastReceiver: Already started.  Ignoring intent: Intent { flg=0x10 cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.startup.StartupBroadcastReceiver (has extras) } of type: null
2021-09-20 14:03:26.468 21646-21646/org.altbeacon.beaconreference D/StartupBroadcastReceiver: onReceive called in startup broadcast receiver
2021-09-20 14:03:26.469 21646-21646/org.altbeacon.beaconreference D/StartupBroadcastReceiver: Already started.  Ignoring intent: Intent { flg=0x10 cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.startup.StartupBroadcastReceiver (has extras) } of type: null
2021-09-20 14:03:26.600 21646-21646/org.altbeacon.beaconreference D/StartupBroadcastReceiver: onReceive called in startup broadcast receiver
2021-09-20 14:03:26.601 21646-21646/org.altbeacon.beaconreference D/StartupBroadcastReceiver: Already started.  Ignoring intent: Intent { flg=0x10 cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.startup.StartupBroadcastReceiver (has extras) } of type: null
2021-09-20 14:03:26.732 21646-21646/org.altbeacon.beaconreference D/StartupBroadcastReceiver: onReceive called in startup broadcast receiver
2021-09-20 14:03:26.733 21646-21646/org.altbeacon.beaconreference D/StartupBroadcastReceiver: Already started.  Ignoring intent: Intent { flg=0x10 cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.startup.StartupBroadcastReceiver (has extras) } of type: null
```
[... cut ...]
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
```

### Mobile device model and OS version

OnePlus 5T, android 9
Pixel 3, android 12 beta 5

### Android Beacon Library version

2.19.2

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2021-10-05:
I have confirmed this problem applies to Library versions 2.19-2.19.2.  It was introduce by changes needed to support Android 12.

Android 12 requires that all `PendingIntent`s set one of two flags: `PendingIntent.FLAG_IMMUTABLE` or `PendingIntent.FLAG_MUTABLE`.  In almost all cases `PendingIntent.FLAG_IMMUTABLE` is the right choice, and that was what was made in all uses in this library for Android 12 support.  That choice was wrong.

The `PendingIntent` that is used for intent-backed scans must be mutable so that the scan results can be attached as extras.   Because this is not possible, all scan results from intent-backed scans are filtered out by this flag.

davidgyoung, 2021-10-05:
Fixed in #1059.  This will be released imminently as 2.19.3

closed 2021-10-05 by davidgyoung