AltBeacon/android-beacon-library #1062: Android 12 an explicit value for `android:exported`  required.

JimSeker, 2021-11-09:
### Expected behavior
it should run and compile.
### Actual behavior
it fails to compile with this error:
Manifest merger failed : android:exported needs to be explicitly specified for <receiver>. Apps targeting Android 12 and higher are required to specify an explicit value for `android:exported` when the corresponding component has an intent filter defined. See https://developer.android.com/guide/topics/manifest/activity-element#exported for details.

### Steps to reproduce this behavior
change any app to API 31.   
or try mine, https://github.com/JimSeker/bluetooth/tree/master/AndroidBeaconLibraryDemo

### Mobile device model and OS version
Android 12, I'm using a pixel3,

### Android Beacon Library version
org.altbeacon:android-beacon-library:2.18-beta

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2021-11-09:
These changes are already implemented in #1043 and released in library version 2.19.  Please update your library usage to the latest 2.19.3 to get these changes.

closed 2021-11-09 by davidgyoung

JimSeker, 2021-11-09:
thank you.  
Also note, for Android 12, it appears you need the new BLUETOOTH_SCAN and BLUETOOTH_CONNECT permissions.

davidgyoung, 2021-11-09:
Yes, @JimSeker that is correct that you need BLUETOOTH_SCAN if you are both targeting Android 12 and running on Android 12.   If you are targeting Android 11 or earlier the old permissions apply even on Android 12 devices.

Note that the BLUETOOTH_SCAN is an additional permission when targeting Android 12 -- you still need FINE_LOCATION and perhaps BACKGROUND_LOCATION.    So you need yet another prompt to request a permission from the user in this case.  The library does not require BLUETOOTH_CONNECT except to populate the device.name field.  Your app may also need this permission for other purposes.

I have not added the BLUETOOTH_SCAN permission to the library's manifest as doing so accomplishes nothing without accompanying code to request that permission dynamically from the user.  Android's permissions structure has become extremely complex, so I am a bit torn on how best to support this in the library.  There are other libraries out there devoted entirely to helping apps request permissions from users.  Other than documenting the requirements I'm not sure what this library should best do to help.