AltBeacon/android-beacon-library #1064: EnableForegroundServiceScanning somtimes return beacon 0

xzq2325, 2021-12-09:
### Expected behavior

Always return the exact number of beacons

### Actual behavior

Generally, the correct amount can be returned, and sometimes beacon 0 is returned, and then it has been difficult to recover. Turning on and off the Bluetooth seems to be able to return to normal

### Steps to reproduce this behavior

I haven't found the steps to reproduce

### Mobile device model and OS version

SEUIC Android 9.0

### Android Beacon Library version

2.19.3

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

I confirm that the Location and Bluetooth switches have been turned on, and the authorization is normal.
When this problem occurred, I debugged and found that there was no call onScanResult function In CycledLeScannerForLollipop.java.

davidgyoung, 2022-01-11:
You may want to try using the BluetoothMedic class to auto recover from this problem.  See: https://stackoverflow.com/a/51450865/1461050

If you find that the above works, then that may be the best we can do here.

If it does not work, then we need to collect more logs at the specific time detections stop, including system level logs.  I would recommend setting up a long-term test and sending logcat output to a file, then letting this run for many hours or days until the problem happens.

xzq2325, 2022-01-19:
> You may want to try using the BluetoothMedic class to auto recover from this problem. See: https://stackoverflow.com/a/51450865/1461050
>
> If you find that the above works, then that may be the best [...]

Very thanks for your answer. I will try using the BluetoothMedic class.

closed 2022-03-03 by davidgyoung