AltBeacon/android-beacon-library #1067: iBeacon not working with Android 12

assembleMHN, 2021-12-21:
Hi guys.

Thank you for all your hard work. I have an issue with the library i hope you can help with :)

We have an older app, using android-beacon-library to recognize ibeacons. I have a physical android 10,11,12 laying on the table in front of me running the exact same codebase. I have a beacon simulator on my iPhone. When i activate the beacon Android 10 and 11 get the signal imidiatly, but the Android 12 device never reacts. I have tried everything, and google i cannot find the reason why the exact same codebase works on 8,9,10,11 but not 12.

Im using the newest version of the library org.altbeacon:android-beacon-library:2.19.2

All the code runs ono Android 12, exactly as on 10 and 11 but 
 
           override fun didEnterRegion(region: Region) {
                ApplicationContext.addLocationTime(true, beaconStatusKeyBuilder(region))
                if (!ApplicationContext.isActivityVisible()) {
                    val date = Date(System.currentTimeMillis())
                    val cal = Calendar.getInstance()
                    cal.time = date
                    if (beaconNotificationCooldown(region)) {
                        ApplicationContext.addLocationNotification(beaconStatusKeyBuilder(region))
                        callback.sendNotification()
                    }
                }
            }

            override fun didExitRegion(region: Region) {
                ApplicationContext.addLocationTime(false, beaconStatusKeyBuilder(region))
                beaconCoolDownPeriod!!.postDelayed(beaconCoolDownPeriodRunnable, coolDownPeriodInSeconds * 1000)
            }

Is never called.

I hope somebody can help me out, or give some input :)

Best Regards
Morten

davidgyoung, 2021-12-21:
Google changed the way bluetooth permissions work for apps targeting Android 12+.    If you are targeting Android 12 (`targetSdkVersion 31` in your build.gradle file), you will also need to declare the following permission in your manifest and request it at runtime from the user:  BLUETOOTH_SCAN

An easier solution is to change your `targetSdkVersion 30` so that the same permission structure for earlier Android versions works with Android 12 devices. Read more here:

https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

closed 2021-12-21 by davidgyoung

noelzappy, 2022-10-04:
I have the permission declared in my Manifest and do request it at runtime, yet the scanning fails with the following logs when the app is killed and relaunched.

```

2022-10-04 03:14:28.482 11274-11463 BluetoothLeScanner      com.tipmeapp.tipme                   D  Start Scan with callback
2022-10-04 03:14:28.483 32139-32211 BtGatt.GattService      pid-32139                            E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_STOP, appName: com.tipmeapp.tipme, scannerId: 12, reportDelayMillis=0
2022-10-04 03:14:28.487 11274-12020 BluetoothLeScanner      com.tipmeapp.tipme                   D  onScannerRegistered() - status=0 scannerId=12 mScannerId=0
2022-10-04 03:14:28.493 32139-32211 BtGatt.GattService      pid-32139                            E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.tipmeapp.tipme, scannerId: 12, reportDelayMillis=0
2022-10-04 03:14:35.102 11274-11463 BluetoothAdapter        com.tipmeapp.tipme                   I  STATE_ON
2022-10-04 03:14:35.102 11274-11463 BluetoothLeScanner      com.tipmeapp.tipme                   D  Stop Scan with callback
2022-10-04 03:14:35.109 11274-11463 BluetoothAdapter        com.tipmeapp.tipme                   I  STATE_ON
2022-10-04 03:14:35.110 11274-11463 BluetoothAdapter        com.tipmeapp.tipme                   I  STATE_ON
2022-10-04 03:14:35.110 11274-11463 BluetoothAdapter        com.tipmeapp.tipme                   I  STATE_ON
2022-10-04 03:14:35.111 11274-11463 BluetoothLeScanner      com.tipmeapp.tipme                   D  Start Scan with callback
2022-10-04 03:14:35.111 32139-32211 BtGatt.GattService      pid-32139                            E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_STOP, appName: com.tipmeapp.tipme, scannerId: 12, reportDelayMillis=0
2022-10-04 03:14:35.116 11274-12020 BluetoothLeScanner      com.tipmeapp.tipme                   D  onScannerRegistered() - status=0 scannerId=12 mScannerId=0
2022-10-04 03:14:35.124 32139-32211 BtGatt.GattService      pid-32139                            E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.tipmeapp.tipme, scannerId: 12, reportDelayMillis=0
2022-10-04 03:14:41.732 11274-11463 BluetoothAdapter        com.tipmeapp.tipme                   I  STATE_ON
2022-10-04 03:14:41.732 11274-11463 BluetoothLeScanner      com.tipmeapp.tipme                   D  Stop Scan with callback
2022-10-04 03:14:41.740 11274-11463 BluetoothAdapter        com.tipmeapp.tipme                   I  STATE_ON
2022-10-04 03:14:41.742 11274-11463 BluetoothAdapter        com.tipmeapp.tipme                   I  STATE_ON
2022-10-04 03:14:41.742 11274-11463 BluetoothAdapter        com.tipmeapp.tipme                   I  STATE_ON
2022-10-04 03:14:41.743 11274-11463 BluetoothLeScanner      com.tipmeapp.tipme                   D  Start Scan with callback
2022-10-04 03:14:41.745 32139-32211 BtGatt.GattService      pid-32139                            E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_STOP, appName: com.tipmeapp.tipme, scannerId: 12, reportDelayMillis=0
2022-10-04 03:14:41.748 11274-11889 BluetoothLeScanner      com.tipmeapp.tipme                   D  onScannerRegistered() - status=0 scannerId=12 mScannerId=0
2022-10-04 03:14:41.757 32139-32211 BtGatt.GattService      pid-32139                            E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.tipmeapp.tipme, scannerId: 12, reportDelayMillis=0
2022-10-04 03:14:48.357 11274-11463 BluetoothAdapter        com.tipmeapp.tipme                   I  STATE_ON
2022-10-04 03:14:48.357 11274-11463 BluetoothLeScanner      com.tipmeapp.tipme                   D  Stop Scan with callback
```

Here is my declaration in the manifest

```
  <uses-permission android:name="android.permission.INTERNET"/>
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
  <uses-permission android:name="android.permission.FOREGROUND_SERVICE"/>
  <uses-permission android:name="android.permission.WAKE_LOCK"/>
  <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION"/>
  
  <uses-permission android:name="android.permission.BLUETOOTH"
      android:maxSdkVersion="30" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
      android:maxSdkVersion="30" />

  <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
  <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
```

What exactly could be the issue, please?