AltBeacon/android-beacon-library #1069: Background Intent scan stops running after enabling / disabling airplane mode

matgar, 2022-01-10:
### Expected behavior

When setting `BeaconManager.setIntentScanningStrategyEnabled(true)` and starting a scan with `startRangingBeacons(region)`, it should run until `removeRangeNotifier(region)` is called.

### Actual behavior

After scan is started, if airplane mode gets enabled, Android will stop delivering intents to `StartupBroadcastReceiver`, and will not deliver again, even after disabling airplane mode. It's like scan is cancelled by the system, without warning the application.

`ScanJob` will try to run a backup scan, but intent coordinator will prevent it, with the message: *We have detected beacons with the intent scan.  No need to do a backup scan*.

The following log shows the last scan intent received, bluetooth adapter being disabled and then enabled again after a while, and `ScanJob` trying to run a backup scan without success. 

(Sensitive data characters were replaced by *)

```
2022-01-10 12:19:52.753 2806-2806 D/StartupBroadcastReceiver: onReceive called in startup broadcast receiver
2022-01-10 12:19:52.759 2806-2806 D/StartupBroadcastReceiver: Passive background scan callback type: 1
2022-01-10 12:19:52.759 2806-2806 D/StartupBroadcastReceiver: got Android O background scan via intent
2022-01-10 12:19:52.761 2806-2806 D/IntentScanCoord: End of scan cycle
2022-01-10 12:19:52.762 2806-2806 D/ScanHelper: Beacon simulator not enabled
2022-01-10 12:19:52.765 2806-2806 D/ScanHelper: Calling ranging callback
2022-01-10 12:19:52.771 2806-2806 D/RunningAverageRssiFilter: Running average mRssi based on 5 measurements: -45.333333333333336
2022-01-10 12:19:52.778 2806-2806 D/RangedBeacon: calculated new runningAverageRssi: -45.333333333333336
2022-01-10 12:19:52.780 2806-2806 D/Callback: attempting callback via direct method call
2022-01-10 12:19:52.783 2806-2806 D/IntentHandler: got ranging data
2022-01-10 12:19:52.773 2806-2967 D/BeaconParser: Ignoring pdu type 1B
2022-01-10 12:19:52.790 2806-2967 D/BeaconParser: Ignoring pdu type 01
2022-01-10 12:19:52.800 2806-2967 D/BeaconParser: Ignoring pdu type 07
2022-01-10 12:19:52.802 2806-2967 D/BeaconParser: Ignoring pdu type 0A
2022-01-10 12:19:52.806 2806-2967 D/BeaconParser: Ignoring pdu type 00
2022-01-10 12:19:52.806 2806-2967 D/BeaconParser: No PDUs to process in this packet.
2022-01-10 12:19:52.814 2806-2967 D/BeaconParser: Ignoring pdu type 1B
2022-01-10 12:19:52.817 2806-2967 D/BeaconParser: Ignoring pdu type 01
2022-01-10 12:19:52.822 2806-2967 D/BeaconParser: Ignoring pdu type 07
2022-01-10 12:19:52.827 2806-2967 D/BeaconParser: Ignoring pdu type 0A
2022-01-10 12:19:52.833 2806-2967 D/BeaconParser: Processing pdu type FF: ****************************************************************************************************** with startIndex: 36, endIndex: 46
2022-01-10 12:19:52.837 2806-2967 D/BeaconParser: Ignoring pdu type 09
2022-01-10 12:19:52.846 2806-2967 D/BeaconParser: This is a recognized beacon advertisement -- ** ** seen
2022-01-10 12:19:52.850 2806-2967 D/BeaconParser: Bytes are: ******************************************************************************************************
2022-01-10 12:19:52.859 2806-2967 D/ScanHelper: Beacon packet detected for:  type ********* with rssi -49
2022-01-10 12:19:52.863 2806-2967 D/ScanHelper: beacon detected :  type *********
2022-01-10 12:19:52.867 2806-2967 D/ScanHelper: looking for ranging region matches for this beacon
2022-01-10 12:19:52.873 2806-2967 D/ScanHelper: matches ranging region: id1: null id2: null id3: null
2022-01-10 12:19:52.878 2806-2967 D/RangeState: adding  type ********* to existing range for: org.altbeacon.beacon.service.RangedBeacon@a3f613
2022-01-10 12:19:55.361 2806-2845 D/BluetoothAdapter: onBluetoothServiceDown: android.bluetooth.IBluetooth$Stub$Proxy@aee174e
2022-01-10 12:33:07.862 2806-2806 I/ScanJob: ScanJob Lifecycle START: org.altbeacon.beacon.service.ScanJob@4e4917c
2022-01-10 12:33:07.867 2806-3519 D/ScanJob: Scan job calling IntentScanStrategyCoordinator
2022-01-10 12:33:07.868 2806-3519 D/IntentScanCoord: End of scan cycle
2022-01-10 12:33:07.868 2806-3519 D/ScanHelper: Beacon simulator not enabled
2022-01-10 12:33:07.871 2806-3519 D/ScanHelper: Calling ranging callback
2022-01-10 12:33:07.873 2806-3519 D/RunningAverageRssiFilter: Running average mRssi based on 0 measurements: NaN
2022-01-10 12:33:07.874 2806-3519 D/RangedBeacon: calculated new runningAverageRssi: NaN
2022-01-10 12:33:07.875 2806-3519 D/RangeState: Dumping beacon from RangeState because it has no recent measurements.
2022-01-10 12:33:07.876 2806-3519 D/Callback: attempting callback via direct method call
2022-01-10 12:33:07.879 2806-3519 D/IntentHandler: got ranging data
2022-01-10 12:33:07.880 2806-3519 D/IntentScanCoord: We have detected beacons with the intent scan.  No need to do a backup scan.
2022-01-10 12:33:07.880 2806-3519 D/ScanJob: Scan job finished.  Calling jobFinished
2022-01-10 12:35:22.515 2806-2979 D/BluetoothAdapter: onBluetoothServiceUp: android.bluetooth.IBluetooth$Stub$Proxy@b8dea05
2022-01-10 12:48:07.871 2806-2806 I/ScanJob: ScanJob Lifecycle START: org.altbeacon.beacon.service.ScanJob@102148b
2022-01-10 12:48:07.876 2806-3860 D/ScanJob: Scan job calling IntentScanStrategyCoordinator
2022-01-10 12:48:07.876 2806-3860 D/IntentScanCoord: End of scan cycle
2022-01-10 12:48:07.876 2806-3860 D/ScanHelper: Beacon simulator not enabled
2022-01-10 12:48:07.878 2806-3860 D/ScanHelper: Calling ranging callback
2022-01-10 12:48:07.879 2806-3860 D/Callback: attempting callback via direct method call
2022-01-10 12:48:07.882 2806-3860 D/IntentHandler: got ranging data
2022-01-10 12:48:07.883 2806-3860 D/IntentScanCoord: We have detected beacons with the intent scan.  No need to do a backup scan.
2022-01-10 12:48:07.883 2806-3860 D/ScanJob: Scan job finished.  Calling jobFinished
2022-01-10 13:03:10.638 2806-2806 I/ScanJob: ScanJob Lifecycle START: org.altbeacon.beacon.service.ScanJob@4082181
2022-01-10 13:03:10.642 2806-3898 D/ScanJob: Scan job calling IntentScanStrategyCoordinator
2022-01-10 13:03:10.643 2806-3898 D/IntentScanCoord: End of scan cycle
2022-01-10 13:03:10.643 2806-3898 D/ScanHelper: Beacon simulator not enabled
2022-01-10 13:03:10.647 2806-3898 D/ScanHelper: Calling ranging callback
2022-01-10 13:03:10.652 2806-3898 D/Callback: attempting callback via direct method call
2022-01-10 13:03:10.654 2806-3898 D/IntentHandler: got ranging data
2022-01-10 13:03:10.655 2806-3898 D/IntentScanCoord: We have detected beacons with the intent scan.  No need to do a backup scan.
2022-01-10 13:03:10.655 2806-3898 D/ScanJob: Scan job finished.  Calling jobFinished
```

Airplane mode is enabled here:
```
2022-01-10 12:19:55.361 2806-2845 D/BluetoothAdapter: onBluetoothServiceDown: android.bluetooth.IBluetooth$Stub$Proxy@aee174e
```

And then disabled here:
```
2022-01-10 12:35:22.515 2806-2979 D/BluetoothAdapter: onBluetoothServiceUp: android.bluetooth.IBluetooth$Stub$Proxy@b8dea05
```

Also, no intent with `BluetoothLeScanner.EXTRA_ERROR_CODE` is sent to `StartupBroadcastReceiver`.

### Steps to reproduce this behavior

1. Enable intent scanning and start ranging beacons:
```kotlin
val manager = BeaconManager.getInstanceForApplication(context)

manager.setIntentScanningStrategyEnabled(true)

manager.addRangeNotifier(rangeNotifier)
manager.startRangingBeacons(defaultRegion)
```

2. Wait for the app to receive scan results.
3. Enable airplane mode on the device.
3. Make sure the app stops receiving scan results.
4. Then disable airplane mode.

### Mobile device model and OS version
Google Pixel (1st generation - 2016)
Android 10

### Android Beacon Library version
2.19.3

davidgyoung, 2022-01-11:
Good catch, @matgar.  I think the library will need to monitor for bluetooth turning from an off to an on state and then restarting that intent-backed scan.

matgar, 2022-01-12:
One thing that I observed during my tests, @davidgyoung: even if you disable the bluetooth adapter, the app will still receive new scan intents (🤯). The described behavior only happens when airplane mode is enabled.

davidgyoung, 2022-01-12:
Thanks for that clarification.  When you say “disable the Bluetooth
adapter” do you mean togging Bluetooth when swiping down from the op of the
screen?  Or do you mean going into the Settings app and disabling Bluetooth?

matgar, 2022-01-12:
The first option, swiping down from the top of the screen. Is there a difference between the two methods?

davidgyoung, 2022-01-12:
Yes, there is a big difference between the two.  The quick settings option does not disable Bluetooth in recent Android versions.  It allows scanning but disallows connections.  The full setting turns off Bluetooth.  Apparently airplane mode does the same as turning off Bluetooth in settings.