AltBeacon/android-beacon-library #1070: Add RxAndroidBle to Gradle, beacons are not scanned in didRangeBeaconsInRegion in Android 12 version.

mjkimloan, 2022-01-14:
Its version is 1.13.1 and Android 12 and lower versions scan beacons normally. 
Remove Rxandroidble from Gradle and beacons will scan normally. How can I solve this problem?****

closed 2022-01-18 by mjkimloan

SergeyLaukhin, 2022-02-09:
@mjkimloan Did you fix it? The same behavior, I really need your help.
Thank you.

zahichemaly, 2022-03-10:
I am facing the same issue. Were you able to solve this?

davidgyoung, 2022-03-10:
On Android 12, your app needs to obtain BLUETOOTH_SCAN permission or it will get the result described.  See [here](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#declare-android12-or-higher)

zahichemaly, 2022-03-10:
I already gave it that permission, and it is still not working. Only when I removed the Rxandroidble it worked again.

davidgyoung, 2022-03-10:
You might try looking at your merged manifest when RxAndridBle is added to see if it makes changes that break your permissions setup.  You can see instructions on inspecting manifest merging problems here https://developer.android.com/studio/build/manage-manifests

zahichemaly, 2022-03-10:
Thanks a bunch. I found the issue. RxAndroidBle was using `android:usesPermissionFlags="neverForLocation"` which filters beacon discovery in the scan.

I added the following permission and explicitly removed the flag:

```
    <uses-permission
        android:name="android.permission.BLUETOOTH_SCAN"
        tools:remove="android:usesPermissionFlags"
        tools:targetApi="s" /> 
```

davidgyoung, 2022-03-10:
Hey, Great, @zahichemaly.  Glad to hear you found it.