AltBeacon/android-beacon-library #1071: Unity Error AndroidJavaException:  java.lang.NoSuchMethodError:

SayChunKim, 2022-01-25:
Hi i actually trying to import aar / module of AltBeacon into Unity project but encountered certain error 
### Expected behavior
Assuming beaconManager able to get instance along with applicationContext via `CallStatic` method of `AndroidJavaClass` of Unity

```cs
public void initializeBeacon()
    {
        AndroidJavaClass unityPlayer = new AndroidJavaClass("com.unity3d.player.UnityPlayer");
        AndroidJavaObject activity = unityPlayer.GetStatic<AndroidJavaObject>("currentActivity");
        AndroidJavaObject context = activity.Call<AndroidJavaObject>("getApplicationContext");
        beaconManager = new AndroidJavaClass("org.altbeacon.beacon.BeaconManager");
//call method via Unity ScriptingAPI
        beaconManager.CallStatic("getInstanceForApplication", context);
        beaconStatusLog.text = "Initialized";
    }
```

### Actual behavior
```
Error Unity AndroidJavaException: java.lang.NoSuchMethodError: no non-static method with name='getInstanceForApplication' signature='(Landroid.app.Application;I)V' in class Ljava.lang.Object;
2022/01/26 04:03:53.329 8653 8693 Error Unity java.lang.NoSuchMethodError: no non-static method with name='getInstanceForApplication' signature='(Landroid.app.Application;I)V' in class Ljava.lang.Object;
2022/01/26 04:03:53.329 8653 8693 Error Unity 	at com.unity3d.player.ReflectionHelper.getMethodID(Unknown Source:226)
2022/01/26 04:03:53.329 8653 8693 Error Unity 	at com.unity3d.player.UnityPlayer.nativeRender(Native Method)
2022/01/26 04:03:53.329 8653 8693 Error Unity 	at com.unity3d.player.UnityPlayer.access$300(Unknown Source:0)
2022/01/26 04:03:53.329 8653 8693 Error Unity 	at com.unity3d.player.UnityPlayer$e$1.handleMessage(Unknown Source:95)
2022/01/26 04:03:53.329 8653 8693 Error Unity 	at android.os.Handler.dispatchMessage(Handler.java:103)
2022/01/26 04:03:53.329 8653 8693 Error Unity 	at android.os.Looper.loop(Looper.java:213)
2022/01/26 04:03:53.329 8653 8693 Error Unity 	at com.unity3d.player.UnityPlayer$e.run(Unknown Source:20)
2022/01/26 04:03:53.329 8653 8693 Error Unity   at UnityEngine.AndroidJNISafe.CheckException () [0x00000] in <00000000000000000000000000000000>:0 

```

### Steps to reproduce this behavior
1. Import snippets of `build.gradle`, `AndroidManifest.xml` as the android studio project in Unity Project
2. Added coding snippet as above for Button canvas `OnClick` listener in Unity Scene

### Mobile device model and OS version
Huawei Nova 4 unrooted VCE-L22
Android version 10

### Android Beacon Library version
2.19.3

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2022-01-25:
This question is out-of-scope for this forum because it is about how to use the library and not regarding a feature request or a bug report.  

My knowledge of Unity is very limited, but I don't think these two lines are correct:

        beaconManager = new AndroidJavaClass("org.altbeacon.beacon.BeaconManager");
//call method via Unity ScriptingAPI
        beaconManager.CallStatic("getInstanceForApplication", context);

It appears that the first line constructs a new instance  of the object from the class and the second line makes a call to the class.  The error message you show is about there not being an instance method called "getInstanceForApplication".  I suspect there is a different way in Unity to call a static method on a class.  Again, I could be wrong -- I am no Unity expert.

Since this is out-of-scope, I am closing this issue.  Please open a question on StackOverflow.com and I will be happy to assist there.  Since my knowledge of Unity is very limited, other members will likely be able to help more.

closed 2022-01-25 by davidgyoung

SayChunKim, 2022-01-25:
> This question is out-of-scope for this forum because it is about how to use the library and not regarding a feature request or a bug report.
> 
> My knowledge of Unity is very limited, but I don't think these two lines are correct:
> 
> ```
>     beaconManager = new AndroidJavaClass("org.altbeacon.beacon.BeaconManager");
>     //call method via Unity ScriptingAPI
>     beaconManager.CallStatic("getInstanceForApplication", context);
> ```
> 
> It appears that the first line constructs a new instance of the object from the class and the second line makes a call to the constructed object. The error message you show is about there not being an instance method called "getInstanceForApplication". I suspect there is a different way in Unity to call a static method on a class. Again, I could be wrong -- I am no Unity expert.
> 
> Since this is out-of-scope, I am closing this issue. Please open a question on StackOverflow.com and I will be happy to assist there. Since my knowledge of Unity is very limited, other members will likely be able to help more.

Thank you, i have managed to get it worked

```cs
AndroidJavaClass unityPlayer = new AndroidJavaClass("com.unity3d.player.UnityPlayer");
        AndroidJavaObject activity = unityPlayer.GetStatic<AndroidJavaObject>("currentActivity");
        AndroidJavaObject context = activity.Call<AndroidJavaObject>("getApplicationContext");
        beaconManager = new AndroidJavaClass("org.altbeacon.beacon.BeaconManager");
        beaconManager.CallStatic<AndroidJavaObject>("getInstanceForApplication", context);
        beaconStatusLog.text = "Initialized";
```

Thanks