AltBeacon/android-beacon-library #1072: MonitorNotifier does not receive callbacks when targetSdkVersion > 30

rbarzel, 2022-02-03:
### Expected behavior
The reference application should work with targetSdkVersion of 32
### Actual behavior
The reference application fails to receive callback when a beacon is detected if targetSdkVersion is set to 31 or higher.
### Steps to reproduce this behavior
In build.gradle set
```android {
    compileSdkVersion 32

    defaultConfig {
        applicationId "org.altbeacon.beaconreference"
        minSdkVersion 21
        targetSdkVersion 30
    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
}
```

In AndroidManifest.xml add the following line to the <activity> element
`android:exported="true">`

Note that the app does not see nearby beacons with the above settings.

### Mobile device model and OS version
Google Pixel 4, Android 12
### Android Beacon Library version
2.19.3
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

NimaaZx, 2022-02-17:
Have you read this "https://github.com/AltBeacon/android-beacon-library/issues/1067" ? need extra permission android 12

vishalcs106, 2022-11-25:
Facing same issue. Even after providing ACCESS_BACKGROUND_LOCATION unable to detect beacons when targetSdk is 31. If i downgrade targetSdk to 30 things are good.

davidgyoung, 2022-11-25:
When targeting Android 12+ you additionally need the user to grant BLUETOOTH_SCAN permission.  (This is a newer Android OS restriction for Bluetooth scanning and is not specific to this library).