AltBeacon/android-beacon-library #1074: Background Service stop detecting beacons Android 9

mesonerodroid, 2022-02-21:
Hello,
I have implemented Altbeacon in an application using the foreground Service, following all the steps in the tutorial. About one minute after going to background or closing the app, no beacons are detected. It only happens on one of the two devices I have.
When I go to the background or close the app, the log is as follows:

`
2022-02-21 13:05:19.523 8783-8783/ D/ Ranged: 1 beacons
2022-02-21 13:05:20.402 8783-8783/ I/BeaconService: set scan intervals received
2022-02-21 13:05:20.436 8783-8783/ I/CycledLeScanner: Adjusted nextScanStartTime to be Mon Feb 21 13:05:19 GMT+01:00 2022
2022-02-21 13:05:20.525 8783-8812/ D/BluetoothAdapter: isLeEnabled(): ON
2022-02-21 13:05:20.528 8783-8783/ E/CycledLeScannerForLollipop: Scan failed: a BLE scan with the same settings is already started by the app
.......
2022-02-21 13:06:21.567 8783-8783/ D/ Ranged: 1 beacons
2022-02-21 13:06:22.674 8783-8783/ D/ Ranged: 0 beacons
2022-02-21 13:06:24.890 8783-8783/ D/ Ranged: 0 beacons`

If I install the app on Android 8.1, beacons continue to be detected perfectly after going to background or closing app. Log:

`2022-02-21 13:03:48.683 31600-31600/ I/BeaconService: set scan intervals received
2022-02-21 13:03:48.773 31600-31600/ I/CycledLeScanner: Adjusted nextScanStartTime to be Mon Feb 21 13:03:48 GMT+01:00 2022
2022-02-21 13:03:49.412 31600-31626/ D/BluetoothAdapter: isLeEnabled(): ON
2022-02-21 13:03:49.413 31600-31600/ E/CycledLeScannerForLollipop: Scan failed: a BLE scan with the same settings is already started by the app`

### Expected behavior
Detecting beacons with app closed.

### Actual behavior
No beacons are detected after approx one minute after closing the app. the service is still active

### Steps to reproduce this behavior
Install basic app with foreground service on Android 9 (28)

### Mobile device model and OS version
Xiaomi mi 9T Android 9PKQ1 MIUI Global 11.0.1
--Working on Xiaomi mi a1 8.1 (27)

### Android Beacon Library version
2.19

davidgyoung, 2022-02-21:
This is likely a custom background restriction on Xiaomi’s MIUI 11.x operating system.  That operating system version is over three years old and unfortunately there is no public documentation on the proprietary Android customizations that may block scanning in the background.  We ou can see an analysis of restrictions in newer MIUI versions here: https://dontkillmyapp.com/xiaomi

Funding a solution will most certainly require lots of trial and error troubleshooting and to reverse engineer the secret restrictions.  

Before starting on this reverse engineering process make sure you have covered all changes needed for long term background scanning on standard Android implementations including this https://github.com/AltBeacon/android-beacon-library/pull/529

mesonerodroid, 2022-02-21:
Indeed, it is the default mode of xiaomi miui 11 "battery saving". When I change it to "no restriction", it keeps working.
Unfortunately, the only option to try is to open this menu by Intent and ask the user to switch to that mode.

 val intent = Intent()
            intent.component = ComponentName(
                "com.miui.powerkeeper",
                "com.miui.powerkeeper.ui.HiddenAppsConfigActivity"
            )
            intent.putExtra("package_name", packageName)
            intent.putExtra("package_label", getString(R.string.app_name))
            startActivity(intent)

Thanks alot.

davidgyoung, 2022-03-03:
I am sorry to say this is the best solution available.  These OEM-specific power savers are a real problem, especially from Chinese manufacturers.

closed 2022-03-03 by davidgyoung