AltBeacon/android-beacon-library #1080: Why use iPhone 1m rssi

huzhongyin, 2022-04-13:
Hello,I can't understand why the rssi at 1m of the iphone is used as the base value instead of the base value of the current phone at 1m for fitting

davidgyoung, 2022-04-13:
The expected RSSI at 1 meter is programmed into the hardware beacon and  included in the beacon Bluetooth LE Advertisement.  There is no way the beacon can know what phone(s) might be listening (a beacon is a one way transmission), so it cannot be specific to the phone model that receives it.  As a result, a "standard" phone model is used.  In reality there is no real "standard" phone model, but the iPhone 4S is the historical reference because that was the only phone in existence that supported detecting iBeacon at the time that iBeacon was released.

closed 2022-04-13 by davidgyoung

huzhongyin, 2022-04-20:
OK，I get it ,Thank you