AltBeacon/android-beacon-library #1084: Monitoring won't work when app is in background and foreground service is running in Android 10+

albertRCell, 2022-05-03:
### Expected behavior
Monitoring should keep working after app is in the background

### Actual behavior
Monitor stops when app is in background specifically when backgroundBetweenScanPeriod is grater than 0 and background location permission is not granted

### Steps to reproduce this behavior
1. Set up this config 
iBeaconManager.enableForegroundServiceScanning(
BeaconNotificationManager.createNotification(
        context
    ), BeaconNotificationManager.notificationId
)
iBeaconManager.setEnableScheduledScanJobs(false)
iBeaconManager.backgroundBetweenScanPeriod = 1000
iBeaconManager.backgroundScanPeriod = 1500
iBeaconManager.bind()
2. Location permission is granted. (background location permission is not)

### Mobile device model and OS version
Motorola Moto G Power OS version 11

### Android Beacon Library version
2.19.2

###More notes 
This problem stops when backgroundBetweenScanPeriod is 0 or background location permission is granted

I have posted the question in stackoverflow as well.
https://stackoverflow.com/q/72104470/7034213

Thanks.

davidgyoung, 2022-05-03:
You do need background permission.  See here:

https://stackoverflow.com/a/72104832/1461050

Metolinam, 2022-05-03:
> ### Beklenen davranış
> Uygulama arka planda olduktan sonra izleme çalışmaya devam etmelidir
> 
> ### Gerçek davranış
> Uygulama arka plandayken özellikle backgroundBetweenScanPeriod 0'dan rende olduğunda ve arka plan konumu izni verilmediğinde monitör durur
> 
> ### Bu davranışı yeniden oluşturma adımları
> 1. Bu yapılandırmayı ayarlayın
> 
> ```
> iBeaconManager.beaconParsers.add(
>     BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_UID_LAYOUT)
> )
> iBeaconManager.beaconParsers.add(
>     BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_TLM_LAYOUT)
> )
> BeaconManager.setRegionExitPeriod(3000)
> iBeaconManager.enableForegroundServiceScanning(
> BeaconNotificationManager.createNotification(
>         context
>     ), BeaconNotificationManager.notificationId
> )
> iBeaconManager.setEnableScheduledScanJobs(false)
> iBeaconManager.backgroundBetweenScanPeriod = 1000
> iBeaconManager.backgroundScanPeriod = 1500
> iBeaconManager.foregroundBetweenScanPeriod = 1000
> iBeaconManager.foregroundScanPeriod = 1500
> iBeaconManager.bind()
> ```
> 
> 2. Konum izni verilir. (arka plan konumu izni değil)
> 
> ### Mobil cihaz modeli ve işletim sistemi sürümü
> Motorola Moto G Power OS sürüm 11
> 
> ### Android Beacon Library sürümü
> 3.19.2
> 
> ###More notlar backgroundBetweenScanPeriod 0 olduğunda veya arka plan konumu izni verildiğinde bu sorun durur
> 
> Soruyu stackoverflow'da da yayınladım. https://stackoverflow.com/q/72104470/7034213
> 
> Teşekkürler.

.circleci/config.yml

closed 2022-05-22 by davidgyoung