AltBeacon/android-beacon-library #1085: 2.16.3 version don't exist on Maven Central

mm-ssuzuki, 2022-05-09:
We use 2.16.3 version now. It is necessary to change Jcenter to Maven Central because JCenter has been shutdown.
2.16.3 version is not upload on Maven Central, only Jcenter as far as I know. 

Could you please upload 2.16.3 version on Maven Central?

Best regards.

davidgyoung, 2022-05-22:
@mm-ssuzuki I looked into doing this for you but unfortunately it is not easy.  Here's why:

Library version 2.16.3 is tagged in Git, but the  build scripts for that version of the library don't support publishing to Maven Central, only to JCenter.   Backporting the maven publishing functionality to that library version's build scripts would be time consuming and would risk making an unintended change to 2.16.3 binary that could affect functionality.

I do have an archived binary of the AAR file here:  https://github.com/AltBeacon/android-beacon-library/releases/tag/2.16.3

A few options:

1. You can embed this car file in your project so that you can keep using that version.
2. If you can find instructions on how to manually post an AAR file to Maven Central, I will try to do that.  I did a quick search and could not find a procedure right away.

mm-ssuzuki, 2022-05-24:
@davidgyoung 
Thank you for your explanation.
I tried to embed archived binary of the AAR as You explained and It worked without problem for now.

I found reference information below. I would appreciate it you reference.
http://gmariotti.blogspot.com/2013/09/publish-aar-file-to-maven-central-with.html

davidgyoung, 2022-06-01:
Glad to hear the AAR file worked.  Unfortunately, the instructions in the link you shared are for how to modify the gradle file to automate this process, so it unfortunately does not help.  I can't use those instructions to publish old copies of the library without also making changes to the old version of the library.