AltBeacon/android-beacon-library #1092: Altbecon ranging stops detecting beacons after phone's bluetooth is turned off and back on

albertRCell, 2022-06-08:
### Expected behavior
Ranging should return beacons after phone's bluetooth is turned off and back on

### Actual behavior
Scanner keeps running but beacons are not reported

### Steps to reproduce this behavior
1. Start Beacon Manager and add a ranging notifier
2. didRangeBeaconsInRegion is getting called with beacons detected
3. Turn off phone's bluetooth
4. Turn on  phone's bluetooth
5. didRangeBeaconsInRegion is still getting called but with no beacons

### Mobile device model and OS version
Moto G Power 2021 Android 11

### Android Beacon Library version
2.19.2

I opened a question in stackoerflow with more details on my config
https://stackoverflow.com/questions/72552856/altbecon-ranging-stops-detecting-beacons-after-phones-bluetooth-is-turned-off-a

Thanks

davidgyoung, 2022-06-11:
This problem is largely a consequence of a setting shown in the question:

1. Set 
`beaconManager.backgroundBetweenScanPeriod = 0`

It probably also only happens if you set `beaconManager.eneableScheduledScanJobs =false` because otherwise scanning would be auto restarted every 15-25 mins.

The reason is that when the background between scan period is 0, scanning never stops and also never restarts.  If you turn off Bluetooth and turn it back on, that effectively kills the original scan.  And because the library was told to never stop that original scan, it never detects again.

Clearly, this is not intended library behavioral — it is a bug.  The library should register for Bluetooth off/ on events and restart scanning in the on event.  

Until this is fixed, you can work around this by doing one of two things:

1. Set the between scan period to something small like 100 ms.
2. Register for callbacks on Bluetooth state n your app, and if it goes off, stop ranging, and if it starts again, restart ranging.

albertRCell, 2022-06-11:
Thanks @davidgyoung I will try the work around