AltBeacon/android-beacon-library #1107: I am able to start the app in background in Android 10 but not in Android 12

fanwu8184, 2022-10-12:
### Expected behavior
The app should be waked up and run in background periodically in Android 12 same as Android 10.

### Actual behavior
The app wakes up and run periodically in Android 10 but not in Android 12.

### Steps to reproduce this behavior
Step 1: open the app
Step 2: kill the app
Step 3: observer logs in the logcat

### Mobile device model and OS version
Both are Samsung phone, one is OS Android 10 , the other one is OS Android 12

### Android Beacon Library version
v2.19.4

My code are below:

class HewApplication: Application(), RangeNotifier {
    private var beaconManager: BeaconManager? = null
    private val hewRegion = Region("hewRegion", null, null, null)

override fun onCreate() {
        super.onCreate()
        println("ttt HewApplication onCreate...")
        beaconManager =  BeaconManager.getInstanceForApplication(applicationContext)
        beaconManager?.addRangeNotifier(this)
        beaconManager?.startRangingBeacons(hewRegion)
    }

override fun didRangeBeaconsInRegion(beacons: MutableCollection<Beacon>?, region: Region?) {
        println("ttt didRangeBeaconsInRegion:  beacons: $beacons")
    }
}

About the permission:
android.permission.ACCESS_FINE_LOCATION: true
android.permission.BLUETOOTH_CONNECT: true
android.permission.ACCESS_BACKGROUND_LOCATION: true
android.permission.BLUETOOTH_SCAN: true

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

AleksArm, 2022-10-20:
I have the same issue, the issue is

 android.app.ForegroundServiceStartNotAllowedException: startForegroundService() not allowed due to mAllowStartForeground false: service com.vozni.finder/org.altbeacon.beacon.service.BeaconService
 
 On Android 12 there is a restriction about starting foreground service when the app works in the background.
 
 https://stackoverflow.com/questions/70044393/fatal-android-12-exception-startforegroundservice-not-allowed-due-to-mallows

davidgyoung, 2022-10-20:
Please try library version 2.19.5-beta6 which contains a fix for a gentle fallback as described in the *Restrictions on Foreground Service Start* section of this page: https://altbeacon.github.io/android-beacon-library/foreground-service.html

You can see a detailed technical description of the change in the beta release here: https://github.com/AltBeacon/android-beacon-library/pull/1088

davidgyoung, 2022-10-20:
closing this issue as there is an existing fix in #1088 and a beta release 2.19.5-beta6.  If you have problems with that fix, please open a new issue describing any problems specific to that release.

closed 2022-10-20 by davidgyoung