AltBeacon/android-beacon-library #1110: Monitoring not working after app is killed

NickLou26, 2022-11-01:
### Expected behavior
I should receive a post request to my api when the device detects ble beacons. I receive the request correctly when the app is opened, but when it is killed, I don't receive the call.

### Actual behavior
I don't receive any request.

### Steps to reproduce this behavior

class CustomApplication : FlutterApplication() {
    companion object {
        private const val TAG = "CustomApplication"
    }

    override fun onCreate() {
        super.onCreate()
        //Works here, i receive the call on my api properly
        //sendRequest() 
        Log.d(TAG, "App started up")
         val beaconManager: BeaconManager = BeaconManager.getInstanceForApplication(this)
         val region = Region("FetchVision", null, null, null)
         beaconManager.getRegionViewModel(region).regionState.observeForever(monitoringObserver)
         beaconManager.startMonitoring(region)
    }

    private val monitoringObserver = Observer<Int> { state ->
        if (state == MonitorNotifier.INSIDE) {
            Log.d(TAG, "Got a didEnterRegion call")
            //Doesn't work when app is killed, but works when is open
            sendRequest()
        }
        else {
            Log.d(TAG, "No beacons detected")
        }
    }
}

### Mobile device model and OS version
Google pixel 5, Android 13

davidgyoung, 2022-11-01:
Please see this page from the library documentation:  https://altbeacon.github.io/android-beacon-library/resume-after-terminate.html

Is what you are witnessing different from what is described there?  Please note the specific ways apps can be stopped and later restarted by the library as described on that page, then update this issue with details.

NickLou26, 2022-11-02:
After consulting the page, I tested it again and it seems to be launching the app correctly after termination. However, if we look at my code : 
override fun onCreate() {
    super.onCreate()
    //Works here, i receive the call on my api properly
    //sendRequest() 
    Log.d(TAG, "App started up")
     val beaconManager: BeaconManager = BeaconManager.getInstanceForApplication(this)
     val region = Region("FetchVision", null, null, null)
     beaconManager.getRegionViewModel(region).regionState.observeForever(monitoringObserver)
     beaconManager.startMonitoring(region)
}

private val monitoringObserver = Observer<Int> { state ->
    if (state == MonitorNotifier.INSIDE) {
        Log.d(TAG, "Got a didEnterRegion call")
        //Doesn't work when app is killed, but works when is open
        sendRequest()
    }
    else {
        Log.d(TAG, "No beacons detected")
    }
}

When the app is launched automatically in background, the sendRequest() in onCreate() is called correctly but the sendRequest in monitoring observer is never called. Is this normal? I'm trying to make an app where I will be running ble scans in the background.

davidgyoung, 2022-11-02:
Let's clarify this: `sendRequest in monitoring observer is never called`.  Two questions:

1. Do you see the monitoringObserver get called shortly after app re-launch (regardless of the value of state)?  
2. If after app re-launch if you turn off your beacon, wait at least 25 minutes, then turn on your beacon and wait another 25 minutes, do you get any calls to monitoringObserver?  (Waiting > 25 minutes is necessary to trigger background/foregrond transitions is necessary on Android 8+ using default settings because this is how often the job scheduler -- the default scan strategy with the library -- allows scheduling scans.)

NickLou26, 2022-11-02:
sendRequest in monitoring observer is called when the app is opened, but not when killed (after re-launch).

1. I tried calling sendRequest like this but it is still not called : 
```
private val monitoringObserver = Observer<Int> { state ->
        sendRequest()
        if (state == MonitorNotifier.INSIDE) {
            Log.d(TAG, "Got a didEnterRegion call")
        }
        else {
            Log.d(TAG, "No beacons detected")
        }
    }
```

2. I tried to do as you said and i still didn't get any calls from sendRequest in monitoringObserver. I do get the call from sendRequest in onCreated though.

davidgyoung, 2023-01-04:
I think the easiest way to address this problem is for you to call:

beaconManager.setRegionStatePeristenceEnabled(false)

You can see details here:  https://altbeacon.github.io/android-beacon-library/state-persistence.html