AltBeacon/android-beacon-library #1111: getting IllegalStateException: schedule()/enqueue() called more than 250 times in the past 60000ms on android 12+

octoberv29, 2022-11-02:
### Expected behavior
not getting IllegalStateException when using the library

### Actual behavior
after calling startMonitoring(region) I'm getting IllegalStateException: schedule()/enqueue() called more than 250 times in the past 60000ms

this started after updating to 2.1.9 version and updating app's targetSdkVersion to 31

crash log:

Caused by: java.lang.IllegalStateException: schedule()/enqueue() called more than 250 times in the past 60000ms. See the documentation for more information.
at
android.os.Parcel.createExceptionOrNull(Parcel.java:2434)
at
android.os.Parcel.createException(Parcel.java:2410)
at
android.os.Parcel.readException(Parcel.java:2393)
at
android.os.Parcel.readException(Parcel.java:2335)
at
android.app.job.IJobScheduler$Stub$Proxy.schedule(IJobScheduler.java:312)
at
android.app.JobSchedulerImpl.schedule(JobSchedulerImpl.java:47)
at
org.altbeacon.beacon.service.ScanJobScheduler.schedule(ScanJobScheduler.java:205)
at
org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob(ScanJobScheduler.java:96)
at
org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob(ScanJobScheduler.java:103)
at
org.altbeacon.beacon.BeaconManager.applyChangesToServices(BeaconManager.java:1251)
at
org.altbeacon.beacon.BeaconManager.startMonitoringBeaconsInRegion(BeaconManager.java:1097)
at
org.altbeacon.beacon.BeaconManager$1.onBeaconServiceConnect(BeaconManager.java:1783)
at
org.altbeacon.beacon.BeaconManager.bindInternal(BeaconManager.java:425)
at
org.altbeacon.beacon.BeaconManager.autoBind(BeaconManager.java:1808)
at
org.altbeacon.beacon.BeaconManager.startMonitoring(BeaconManager.java:1136)

### Steps to reproduce this behavior
I start scanning for beacons on Android 12 device (Google Pixel 6), then I background an app, then come back to it, do a couple of clicks in the UI and get this IllegalStateException.

### Mobile device model and OS version
Android 12, Google Pixel 6

### Android Beacon Library version
2.1.9

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2022-11-02:
The stack trace starts with this API method: `BeaconManager.startMonitoring(...)`. Can you please show the code that calls this method in your app?  How is this code triggered?  It seems something may be calling it over 250 times in 60 seconds... this method is not designed to be called that frequently.

octoberv29, 2022-11-03:
Hi, sorry, we have finally found an issue. Yeah, it was on our side. 

After update to Target API = 31, we only added a check for bluetooth permissions and we thought that the error came from beacon library, because we couldn't find an error for almost a week. But yesterday we found that we forgot to add distinctUntilChanged() operator to behaviourSubject that was responsible for posting changes in bluetooth permissions.

Thus, previously we would make a call every time the subject is updated, even if it's the same value, but we wanted to only make a call when we receive a value (true) different to a previous one. Hope this helps to anyone else facing the same issue.

btw, thanks for quick response!

closed 2022-11-03 by octoberv29

davidgyoung, 2022-11-03:
Great!  Good to hear you found the issue