AltBeacon/android-beacon-library #1113: background scanning does not start (version: 2.19.4)

eral09, 2022-11-17:
### Expected behavior
When our app is no longer in foreground, background time between scans and scan time should be used.

### Actual behavior
BeaconManager backgroundMode, background time between scans and scan time is correct, but ranging still uses the foreground parameters.

### Steps to reproduce this behavior
Setup time between scans (foreground/background). Minimize the app / lock the screen.

### Mobile device model and OS version
Tested devices:
Pixel 6 android 13
Samsung A52 android 12 

### Android Beacon Library version
2.19.4
Rolling back to 2.19.2 works fine

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2022-11-22:
@eral09, I was not able to reproduce this problem with the official reference app here: https://github.com/davidgyoung/android-beacon-library-reference-kotlin

I modified this reference app to make these minimal changes to match what you described (but I changed the background timings to make it easier to see the result in the logs):

```
diff --git a/app/build.gradle b/app/build.gradle
index 0436a15..803d3d3 100644
--- a/app/build.gradle
+++ b/app/build.gradle
@@ -3,13 +3,13 @@ apply plugin: 'kotlin-android'
 apply plugin: 'kotlin-android-extensions'
 
 android {
-    compileSdkVersion 32
+    compileSdkVersion 33
     buildToolsVersion "32.0.0"
 
     defaultConfig {
         applicationId "org.altbeacon.beaconreference"
         minSdkVersion 26
-        targetSdkVersion 31
+        targetSdkVersion 33
         versionCode 1
         versionName "1.0"
 
diff --git a/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.kt b/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.kt
index 521ecd1..c15c1cf 100644
--- a/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.kt
+++ b/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.kt
@@ -61,10 +61,10 @@ class BeaconReferenceApplication: Application() {
         // If you want to continuously range beacons in the background more often than every 15 mintues,
         // you can use the library's built-in foreground service to unlock this behavior on Android
         // 8+.   the method below shows how you set that up.
-        //setupForegroundService()
-        //beaconManager.setEnableScheduledScanJobs(false);
-        //beaconManager.setBackgroundBetweenScanPeriod(0);
-        //beaconManager.setBackgroundScanPeriod(1100);
+        setupForegroundService()
+        beaconManager.setEnableScheduledScanJobs(false);
+        beaconManager.setBackgroundBetweenScanPeriod(10000);
+        beaconManager.setBackgroundScanPeriod(1100);
 
         // Ranging callbacks will drop out if no beacons are detected
         // Monitoring callbacks will be delayed by up to 25 minutes on region exit
```

Then put the app to the background and viewed the following in LogCat:

```
2022-11-21 19:40:21.377 16658-16658/org.altbeacon.beaconreference D/BeaconReference: id1: 00e69745-943a-45de-bf29-f80586c46c99 id2: 1 id3: 1 about 13.654204655794677 meters away
2022-11-21 19:40:22.377 16658-16658/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 8978 milliseconds
2022-11-21 19:40:23.380 16658-16658/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 7976 milliseconds
2022-11-21 19:40:24.382 16658-16658/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 6974 milliseconds
2022-11-21 19:40:25.384 16658-16658/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 5972 milliseconds
2022-11-21 19:40:26.386 16658-16658/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 4970 milliseconds
2022-11-21 19:40:27.388 16658-16658/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 3968 milliseconds
2022-11-21 19:40:28.390 16658-16658/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 2966 milliseconds
2022-11-21 19:40:29.392 16658-16658/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 1964 milliseconds
2022-11-21 19:40:30.394 16658-16658/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 962 milliseconds
2022-11-21 19:40:31.357 16658-16658/org.altbeacon.beaconreference D/CycledLeScanner: starting a new scan cycle
```

As you can see from the above, the library pauses scans for 10 seconds as directed by the backgroundBetweenScanPeriod setting.

You may wish to try the same with the reference app and confirm you see the same.  If so, you may want to look for other differences between your app and the reference app and see if you can isolate the cause of the issue.

If you can reproduce a problem with the reference app, please add instructions on how to do so and re-open this issue.

eral09, 2022-11-22:
I had to use beaconManager.setEnableScheduledScanJobs(false); for it to work on 2.19.4.

Thank you!

eral09, 2022-11-24:
Update:
I gave you the wrong information earlier regarding how we setup the beacon scanning in our application.

In our application we run the beacon scanning in a **background** service within our application. Going from altBeacon 2.19.2 to 2.19.4 scanning does not work in the background anymore.

I did some further testing in the Reference app with following changes:

````
diff --git a/app/build.gradle b/app/build.gradle
index 7571d78..6c2dc48 100644
--- a/app/build.gradle
+++ b/app/build.gradle
@@ -3,13 +3,13 @@ apply plugin: 'kotlin-android'
 apply plugin: 'kotlin-android-extensions'

 android {
-    compileSdkVersion 32
+    compileSdkVersion 33
     buildToolsVersion "32.0.0"

     defaultConfig {
         applicationId "org.altbeacon.beaconreference"
         minSdkVersion 26
-        targetSdkVersion 31
+        targetSdkVersion 33
         versionCode 1
         versionName "1.0"
````

````
diff --git a/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.kt b/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.kt
index 521ecd1..9179a71 100644
--- a/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.kt
+++ b/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.kt
@@ -62,20 +62,28 @@ class BeaconReferenceApplication: Application() {
         // you can use the library's built-in foreground service to unlock this behavior on Android
         // 8+.   the method below shows how you set that up.
         //setupForegroundService()
-        //beaconManager.setEnableScheduledScanJobs(false);
-        //beaconManager.setBackgroundBetweenScanPeriod(0);
-        //beaconManager.setBackgroundScanPeriod(1100);
+        beaconManager.setForegroundBetweenScanPeriod(15000);
+        beaconManager.setForegroundScanPeriod(5000);
+        beaconManager.setBackgroundBetweenScanPeriod(60000 * 15);
+        beaconManager.setBackgroundScanPeriod(6000);
````

This works fine on version 2.19.2 but not in 2.19.4

davidgyoung, 2022-11-24:
OK, please clarify "Going from altBeacon 2.19.2 to 2.19.4 scanning does not work in the background anymore."  How specifically does it not work?  Are the timings of the scan cycles incorrect as originally noted in this issue?  Or are you saying that no devices are detected regardless of scan cycles?

The expected behavior (regardless of library version) on Android 10+ is that a background service will be active to scan for beacons only for ~10 minutes after transitioning to the background, at which point Android will kill the service and no beacons can be detected anymore until the next time the app is launched manually.  The background scans should run as configured in the scan cycles, and will return the results on Android < 12 only if FINE_LOCATION and BACKGROUND_LOCATION permissions are granted by the user, and on Android 12+ only if FINE_LOCATION, BACKGROUND_LOCATION and  BLUETOOTH_SCAN permissions are granted by the user.   

How does what you see differ from the above?

eral09, 2022-11-24:
The time between scans are incorrect.

**2.19.4:** When app is in background (e.g. pressing home button and/or locking the screen) time between scan value is incorrect. Setting `beaconManager.setBackgroundBetweenScanPeriod` value has no effect.

**2.19.2:** When app is in background (e.g. pressing home button or locking and/or locking the screen) setting `beaconManager.setBackgroundBetweenScanPeriod` works as expected.

davidgyoung, 2022-11-27:
Below is log excerpt of a test of the reference app on a Pixel 4a with Android 13.  The reference app includes the changes described above (compile/targetSDK 33), setting a long background between scan period of 15 minutes, I see the 15 scan periods being respected within the +/-5 minute flex of the Android OS (The case below actually shows the next scan starting just under 11 minutes later.)

Based on this test, I believe 2.19.4 works as designed.  If you see otherwise, please repeat this same test attaching logs showing what you see differently.

```
2022-11-27 10:50:16.820 22988-22988/org.altbeacon.beaconreference D/MainActivity: onPause
2022-11-27 10:50:16.821 22988-22988/org.altbeacon.beaconreference D/BackgroundPowerSaver: activity paused: org.altbeacon.beaconreference.MainActivity@fb42c4 active activities: 0
2022-11-27 10:50:16.821 22988-22988/org.altbeacon.beaconreference D/BackgroundPowerSaver: setting background mode
2022-11-27 10:50:16.821 22988-22988/org.altbeacon.beaconreference D/BeaconManager: API setBackgroundMode true
2022-11-27 10:50:16.821 22988-22988/org.altbeacon.beaconreference D/BeaconManager: API setBackgroundModeIternal true
2022-11-27 10:50:16.821 22988-22988/org.altbeacon.beaconreference D/BeaconManager: API updateScanPeriods
2022-11-27 10:50:16.821 22988-22988/org.altbeacon.beaconreference D/BeaconManager: updating background flag to true
2022-11-27 10:50:16.821 22988-22988/org.altbeacon.beaconreference D/BeaconManager: updating scan periods to 6000, 900000
2022-11-27 10:50:16.821 22988-22988/org.altbeacon.beaconreference D/ScanJobScheduler: Applying settings to ScanJob
2022-11-27 10:50:16.828 22988-22988/org.altbeacon.beaconreference D/ScanState: Scan state restore regions: monitored=1 ranged=1
2022-11-27 10:50:16.828 22988-22988/org.altbeacon.beaconreference D/BeaconManager: API getBeaconParsers, current count 1
2022-11-27 10:50:16.828 22988-22988/org.altbeacon.beaconreference D/ScanState: ranged regions: old=1 new=1
2022-11-27 10:50:16.828 22988-22988/org.altbeacon.beaconreference D/ScanState: monitored regions: old=1 new=1
2022-11-27 10:50:16.828 22988-22988/org.altbeacon.beaconreference D/ScanState: Updated state with 1 ranging regions and 1 monitoring regions.
2022-11-27 10:50:16.850 22988-22988/org.altbeacon.beaconreference D/ScanState: Temp file is /data/user/0/org.altbeacon.beaconreference/files/android-beacon-library-scan-state-temp
2022-11-27 10:50:16.850 22988-22988/org.altbeacon.beaconreference D/ScanState: Perm file is /data/user/0/org.altbeacon.beaconreference/files/android-beacon-library-scan-state
2022-11-27 10:50:16.851 22988-22988/org.altbeacon.beaconreference D/MonitoringStatus: saveMonitoringStatusIfOn()
2022-11-27 10:50:16.854 22988-22988/org.altbeacon.beaconreference D/ScanJobScheduler: Applying scan job settings with background mode true
2022-11-27 10:50:16.854 22988-22988/org.altbeacon.beaconreference D/BeaconLocalBroadcastProcessor: Register calls: global=1
2022-11-27 10:50:16.854 22988-22988/org.altbeacon.beaconreference D/ScanState: ScanState says background mode for ScanJob is true
2022-11-27 10:50:16.854 22988-22988/org.altbeacon.beaconreference D/ScanJobScheduler: Not scheduling an immediate scan because we are in background mode.   Cancelling existing immediate ScanJob.
2022-11-27 10:50:16.857 22988-22988/org.altbeacon.beaconreference I/ScanJob: Using immediateScanJobId from manifest: 208352939
2022-11-27 10:50:16.860 22988-22988/org.altbeacon.beaconreference I/ScanJob: Using periodicScanJobId from manifest: 208352940
2022-11-27 10:50:16.861 22988-22988/org.altbeacon.beaconreference W/JobInfo: Requested flex 0 for job 208352940 is too small; raising to +5m0s0ms
2022-11-27 10:50:16.862 22988-22988/org.altbeacon.beaconreference D/ScanJobScheduler: Scheduling periodic ScanJob (job:208352940/org.altbeacon.beaconreference/org.altbeacon.beacon.service.ScanJob) to run every 906000 millis
2022-11-27 10:50:16.876 22988-22988/org.altbeacon.beaconreference D/ScanJob: onStopJob called
2022-11-27 10:50:16.878 22988-22988/org.altbeacon.beaconreference I/ScanJob: Using periodicScanJobId from manifest: 208352940
2022-11-27 10:50:16.878 22988-22988/org.altbeacon.beaconreference I/ScanJob: onStopJob called for immediate scan org.altbeacon.beacon.service.ScanJob@564980e
2022-11-27 10:50:16.878 22988-22988/org.altbeacon.beaconreference I/ScanJob: ScanJob Lifecycle STOP: org.altbeacon.beacon.service.ScanJob@564980e
2022-11-27 10:50:16.881 22988-22988/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
2022-11-27 10:50:16.883 22988-22988/org.altbeacon.beaconreference D/CycledLeScanner: stop called
2022-11-27 10:50:16.883 22988-22988/org.altbeacon.beaconreference D/CycledLeScanner: disabling scan
2022-11-27 10:50:16.883 22988-22988/org.altbeacon.beaconreference D/CycledLeScanner: Done with scan cycle
2022-11-27 10:50:16.883 22988-27294/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Stopping LE scan on scan handler
2022-11-27 10:50:16.883 22988-22988/org.altbeacon.beaconreference D/ScanHelper: Beacon simulator not enabled
2022-11-27 10:50:16.883 22988-27294/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
2022-11-27 10:50:16.883 22988-27294/org.altbeacon.beaconreference D/BluetoothLeScanner: could not find callback wrapper
2022-11-27 10:50:16.884 22988-22988/org.altbeacon.beaconreference D/ScanHelper: Calling ranging callback
[... cut ...]
2022-11-27 11:01:20.068 22988-22988/org.altbeacon.beaconreference D/CurveFittedDistanceCalculator: avg mRssi: -65.66666666666667 distance: 1.4354929997337451
2022-11-27 11:01:20.068 22988-22988/org.altbeacon.beaconreference D/BeaconReference: id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 1 id3: 1 about 1.4354929997337451 meters away
2022-11-27 11:01:20.078 22988-22988/org.altbeacon.beaconreference D/ScanJob: Checking to see if we need to start a passive scan
2022-11-27 11:01:20.078 22988-22988/org.altbeacon.beaconreference I/ScanJob: We are inside a beacon region.  We will not scan between cycles.

```

closed 2022-11-27 by davidgyoung