AltBeacon/android-beacon-library #1114: Parser priority is required for interoperability

Maximvdw, 2022-12-06:
AltBeacon is meant to be an 'open' interoperable standard. I have a use case where I have a layout for an alternative beacon that has similarities with AltBeacon but adds additional features. I still want to detect AltBeacon's, but I want to give my parser a priority for trying to find variants on AltBeacon.

_To clarify: I have the AltBeacon parser and my own parser. My own parser will detect a subset of beacons detected by the AltBeacon parser_

### Expected behavior
I originally expected the parser priority to depend on the order on which it was added to the `BeaconManager`, so I cleared the `getBeaconParsers` to remove the default AltBeacon parser. I added my own parser and added the AltBeacon parser again.

### Actual behavior
Due to the HashSet https://github.com/AltBeacon/android-beacon-library/blob/db4622cc4baad43d100333f85e40b4d2a4ae0637/lib/src/main/java/org/altbeacon/beacon/service/ScanHelper.java#L75
the order is not determined
If a beacon is parsed by "a" parser: https://github.com/AltBeacon/android-beacon-library/blob/db4622cc4baad43d100333f85e40b4d2a4ae0637/lib/src/main/java/org/altbeacon/beacon/service/ScanHelper.java#L436-L442 other parsers are ignored

I can manage to get the parser working by not 'readding' the AltBeacon parser. However, I still want to detect AltBeacons that are not a subset of my own parser.

### Android Beacon Library version
[2.19.4](https://mvnrepository.com/artifact/org.altbeacon/android-beacon-library/2.19.4)
I have tried the beta with the same result.

### Possible Fix
Add a priority to the beacon parser or maintain priority based on its order in the beacon manager.

davidgyoung, 2022-12-06:
On first glance, I don't see why the list of BeaconParsers couldn't be maintained in an ArrayList.  It seems like that is the simplest solution to give you what you are looking for.  If this were the case, you could clear the list of parsers, and re-add them in the desired order.

Maximvdw, 2022-12-07:
yes, an arraylist with the same order as the one from BeaconManager would be perfect

Maximvdw, 2023-01-19:
Is this something that could be implemented in a new release or does this break compatibility with other features?

davidgyoung, 2023-01-19:
I think it is safe to make this change as there is no API contract for the order of evaluation -- right not it is indeterminate based on the implementation of Set inside the JVM, which is also subject to change.   I also think it is very unlikely this will affect the vast majority of apps which use common beacon parsers for which there is not any overlap between any two.