AltBeacon/android-beacon-library #1115: beacon distance value showing negative distance on Motorola phones (moto edge 20 pro & moto g5+)

gptshubham595, 2022-12-19:
### Expected behavior
    Distance observed via beacon should be positive

### Actual behavior
       Distance observed via beacon is negative at any distance even if there's a little shake

### Steps to reproduce this behavior
      1. Install altbeacon reference app on mototrola phone and observe the distance shown
      2. shake the phone

### Mobile device model and OS version
     Tested on moto edge 20 pro & moto g5+
     OS Version: 12
     
### Android Beacon Library version
   altbeacon_version = "2.19.4"

davidgyoung, 2022-12-19:
Thanks for this report.   Can you please provide a few samples of these two values when you see the negative distance?

1. beacon.runningAverageRssi
2. beacon.distance

gptshubham595, 2022-12-20:
Tested on Moto G5 S+
<google-sheets-html-origin>

13:37:28 | Distance: | 1.358780418
-- | -- | --
13:37:28 | Running | -87
13:37:29 | Ranged: | 1
13:37:29 | Distance: | 1.042936817
13:37:29 | Running | -85
13:37:30 | Ranged: | 0
13:37:30 | Ranged: | 0
13:37:31 | Ranged: | 0
13:37:32 | Ranged: | 1
13:37:32 | Distance: | 0.7765247283
13:37:32 | Running | -83
13:37:32 | Ranged: | 1
13:37:32 | Distance: | -0.1508527346
13:37:32 | Running | -70
13:37:33 | Ranged: | 1
13:37:33 | Distance: | -0.1133826044
13:37:33 | Running | -71
13:37:34 | Ranged: | 1
13:37:34 | Distance: | -0.2150709093
13:37:34 | Running | -68
13:37:35 | Ranged: | 0
13:37:35 | Ranged: | 1
13:37:35 | Distance: | 0.9506274682
13:37:35 | Running | -65.66666667
13:37:36 | Ranged: | 0
13:37:37 | Ranged: | 1
13:37:37 | Distance: | -0.2590933289
13:37:37 | Running | -66.33333333
13:37:38 | Ranged: | 0
13:37:39 | Ranged: | 1
13:37:39 | Distance: | -0.2052836873
13:37:39 | Running | -68.33333333
13:37:40 | Ranged: | 1
13:37:40 | Distance: | -0.242427232
13:37:40 | Running | -67
13:37:40 | Ranged: | 0
13:37:41 | Ranged: | 1
13:37:41 | Distance: | -0.242427232
13:37:41 | Running | -67
13:37:42 | Ranged: | 1
13:37:42 | Distance: | -0.226366312
13:37:42 | Running | -67.6
13:37:43 | Ranged: | 0
13:37:44 | Ranged: | 1
13:37:44 | Distance: | -0.1846436131
13:37:44 | Running | -69
13:37:45 | Ranged: | 0
13:37:46 | Ranged: | 1
13:37:46 | Distance: | -0.1508527346
13:37:46 | Running | -70
13:37:46 | Ranged: | 0
13:37:47 | Ranged: | 1
13:37:47 | Distance: | -0.2669835118
13:37:47 | Running | -66
13:37:48 | Ranged: | 0
13:37:49 | Ranged: | 0
13:37:50 | Ranged: | 0
13:37:50 | Ranged: | 1
13:37:50 | Distance: | 0.4189038879
13:37:50 | Running | -60.5
13:37:51 | Ranged: | 0
13:37:52 | Ranged: | 1
13:37:52 | Distance: | 0.2746885648
13:37:52 | Running | -58
13:37:53 | Ranged: | 0

gptshubham595, 2022-12-20:
Also i've noted that the beacon reading (Ranged 0 beacons) even if the beacon advertiser and receiver are in close vicinity
can you help how we can read beacons consistently

```
foregroundScanPeriod = 700 
bacgroundScanPeriod = 700
RunningAverageRssiFilter.setSampleExpirationMiliSeconds(5000L)
```

RenisDudhat, 2022-12-20:
Hi @davidgyoung 

I faced the same issue that sometimes getting negative distance on Motorola devices.

Explanation : 

- As you have explained in the documentation that "The library uses a different formula for calculating distance for different Android device models and if no match is found for the device, the library will fall back to the default device calculation for the Nexus 5".
- But in the case of Motorola devices, it will always find a model from the list of available [models](https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/resources/model-distance-calculations.json) as there is an object with  **"manufacturer": "Motorola"** key-value pair(let's call it _moto-device-model_).
- Next, with this matched model library, will calculate the distance using the formula... ( Two cases )

     - If the ratio of RSSI measured by a device to the reference RSSI is less than 1 it will use the formula 
          - **distance = Math.pow(ratio,10);**
          - In this case, no negative distance is measured. (Distance will be same as other devices)
          
     - Else  will use 
          - **distance = (mCoefficient1) x Math.pow(ratio,mCoefficient2) + mCoefficient3;**
          - In this case, we are getting negative distance for Motorola devices because of the negative value of mCoefficient3 (-0.45324519) in the moto-device-model.
          
     - Something more than the above two cases...
          - Let me divide 2nd equation into two halves. The first is "(mCoefficient1)*Math.pow(ratio,mCoefficient2)" and the second is mCoeffiient3.
          - Now if the first half yields the output which is >= 0.45324519 , again the distance will be positive for the second equation itself.  

In short, The negative distance measured in Motorola devices is because of the negative value of mCoefficient3.

gptshubham595, 2022-12-20:
are there any option to replace this json instead of loading complete library?

gptshubham595, 2022-12-21:
It worked, but can you tell how we can get these three coefficient values for different phones ?

coz I know we can get distance via this formulae but for 2nd equation how we can get variables:
```
txpower = -40   
ratio =(txpower-rssi)/40
distance= math.pow(10,ratio)
print(distance)
```

davidgyoung, 2022-12-23:
Yes, it sounds like the Motorola coefficients should be replaced in the library's built-in and web hosted device map file.  As @RenisDudhat described above, there is a certain range of RSSI values input to the formula that will yield negative values and this should not happen.  

There are instructions for calculating the constants in the last section of this page:  https://altbeacon.github.io/android-beacon-library/distance-calculations.html

If you can take the measurements and come up with a set of coefficients that does not give negative values, we can replace the existing Motorola coefficients with a new one.

gptshubham595, 2022-12-27:
Hey, I tried these two formulas to get the values; if I get the C as -ve, I chose different formulae so that C should be positive or zero always 

-  for C > =0 
    d = A * (r/t) ^ B + C
- for C < 0
    d = A * (r/t) ^ B 
    
For Motorola device I got these values

Method 1
---
| A | B | C |
| -- | -- | -- |
| 2.33 | 4.75 | -1.85|

Method 2 (C as 0)
---
| A | B | C |
| -- | -- | -- |
|1.60433254 | 5.33443408 | 0|

```
import numpy, scipy, matplotlib
import matplotlib.pyplot as plt
from scipy.optimize import curve_fit
from scipy.optimize import differential_evolution
import warnings

yData = numpy.array([1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 18, 20, 27, 34, 41])
xData = numpy.array([0.9607843137,1.117647059,1.176470588,1.254901961,1.274509804,1.333333333,1.411764706,1.529411765,1.509803922,1.431372549,1.470588235,1.568627451,1.392156863,1.411764706,1.62745098,1.490196078,1.68627451,1.529411765,1.784313725,1.647058824])

def func(x, a, b, c): 
    return (a * (x**b)) + c

# def func(x, a, b): 
#    return (a * (x**b)) 

# function for genetic algorithm to minimize (sum of squared error)
def sumOfSquaredError(parameterTuple):
    warnings.filterwarnings("ignore") # do not print warnings by genetic algorithm
    val = func(xData, *parameterTuple)
    return numpy.sum((yData - val) ** 2.0)

def generate_Initial_Parameters():
    # min and max used for bounds
    maxX = max(xData)
    minX = min(xData)
    maxY = max(yData)
    minY = min(yData)

    minData = min(minX, minY)
    maxData = max(maxX, maxY)

    parameterBounds = []
    parameterBounds.append([0, maxData]) # search bounds for a
    parameterBounds.append([0, maxData]) # search bounds for b
    parameterBounds.append([0, maxData]) # search bounds for Offset

    # "seed" the numpy random number generator for repeatable results
    result = differential_evolution(sumOfSquaredError, parameterBounds, seed=5)
    return result.x

# by default, differential_evolution completes by calling curve_fit() using parameter bounds
geneticParameters = generate_Initial_Parameters()

# now call curve_fit without passing bounds from the genetic algorithm,
# just in case the best fit parameters are aoutside those bounds
#fittedParameters, pcov = curve_fit(func, xData, yData, geneticParameters, maxfev=3000)
fittedParameters, pcov = curve_fit(func, xData, yData, geneticParameters, maxfev=3000)
print('Fitted parameters:', fittedParameters)
print()

modelPredictions = func(xData, *fittedParameters) 

absError = modelPredictions - yData

SE = numpy.square(absError) # squared errors
MSE = numpy.mean(SE) # mean squared errors
RMSE = numpy.sqrt(MSE) # Root Mean Squared Error, RMSE
Rsquared = 1.0 - (numpy.var(absError) / numpy.var(yData))

print()
print('RMSE:', RMSE)
print('R-squared:', Rsquared)

print()

##########################################################
# graphics output section
def ModelAndScatterPlot(graphWidth, graphHeight):
    f = plt.figure(figsize=(graphWidth/100.0, graphHeight/100.0), dpi=100)
    axes = f.add_subplot(111)

    # first the raw data as a scatter plot
    axes.plot(xData, yData,  'D')

    # create data for the fitted equation plot
    xModel = numpy.linspace(min(xData), max(xData))
    yModel = func(xModel, *fittedParameters)

    # now the model as a line plot
    axes.plot(xModel, yModel)

    axes.set_xlabel('X Data') # X axis data label
    axes.set_ylabel('Y Data') # Y axis data label

    plt.show()
    plt.close('all') # clean up after using pyplot

graphWidth = 800
graphHeight = 600
ModelAndScatterPlot(graphWidth, graphHeight)
```

davidgyoung, 2022-12-28:
@gptshubham595  and @RenisDudhat can you do some testing with these values and see how they work on your Motorola devices in practice?
 
Method 2 (C as 0)
---
| A | B | C |
| -- | -- | -- |
|1.60433254 | 5.33443408 | 0|

In order to properly put this into the config file we really need to know the actual values reported by android.os.Build for:

            Build.VERSION.RELEASE,
            Build.ID,
            Build.MODEL,
            Build.MANUFACTURER

The CSV file relies on those actual values for matching purposes.

You can use this URL to configure the library with this equation for initial testing: https://android-beacon-library.s3.amazonaws.com/android-distance2.json

I made it the first equation for Motorola which should match that for your device if you device reports it Build.MANUFACTURER as "Motorola"

gptshubham595, 2022-12-28:
I used these values and all worked perfectly fine
https://github.com/gptshubham595/beacon/blob/main/android-distance.json

gptshubham595, 2023-01-05:
Yeah sure you can use these:

```
{
  "models":
  [
    {
      "coefficient1": 0.42093,
      "coefficient2": 6.9476,
      "coefficient3": 0.54992,
      "version":"4.4.2",
      "build_number":"KOT49H",
      "model":"Nexus 4",
      "manufacturer":"LGE"
    },
    {
      "coefficient1": 0.42093,
      "coefficient2": 6.9476,
      "coefficient3": 0.54992,
      "version":"4.4.2",
      "build_number":"LPV79",
      "model":"Nexus 5",
      "manufacturer":"LGE",
      "default": true
    },
    {
      "coefficient1": 0.9401940951,
      "coefficient2": 6.170094565,
      "coefficient3": 0.0,
      "version":"5.0.2",
      "build_number":"LXG22.67-7.1",
      "model":"Moto X Pro",
      "manufacturer":"XT1115",
      "default": false
    },
    {
      "coefficient1": 1.60433254,
      "coefficient2": 5.33443408,
      "coefficient3": 0,
      "version":"6.0",
      "build_number":"MPE24.49-18",
      "model":"XT1092",
      "manufacturer":"Motorola",
      "default": false
    },
    {
      "coefficient1": 2.65,
      "coefficient2": 3.91,
      "coefficient3": 1.61,
      "version":"12.0",
      "build_number":"SP1A.210812.016",
      "model":"CPH2401",
      "manufacturer":"Oneplus",
      "default": false
    },
    {
      "coefficient1": 0.000475,
      "coefficient2": 25.65,
      "coefficient3": 5.86,
      "version":"6.0",
      "build_number":"",
      "model":"",
      "manufacturer":"Google",
      "default": false
    },
    {
      "coefficient1": 0.102139,
      "coefficient2": 13.18,
      "coefficient3": 2.38,
      "version":"6.0",
      "build_number":"",
      "model":"",
      "manufacturer":"Oppo",
      "default": false
    },
    {
      "coefficient1":1.660582	,
      "coefficient2": 6.05,
      "coefficient3": 0.67,
      "version":"6.0",
      "build_number":"",
      "model":"",
      "manufacturer":"Redmi",
      "default": false
    },
    {
      "coefficient1": 1.35070638	,
      "coefficient2": 7.31428284,
      "coefficient3": 0,
      "version":"6.0",
      "build_number":"",
      "model":"",
      "manufacturer":"samsung",
      "default": false
    },
    {
      "coefficient1": 1.9504739,
      "coefficient2": 5.42879314,
      "coefficient3": 0,
      "version":"6.0",
      "build_number":"",
      "model":"",
      "manufacturer":"Vivo",
      "default": false
    }
  ]
}
```