AltBeacon/android-beacon-library #1116: Beacon values reading is not consistent, gap between two beacon reading is around 8 to 10s depening on phone

gptshubham595, 2022-12-20:
### Expected behavior
    Beacon reaading should be consistent gap between two beacon reading should be less than 2 sec

### Actual behavior
    Beacon reaading is inconsistent gap between two beacon reading is more than 8 to 10 sec in some phones

### Steps to reproduce this behavior
    Install altbeacon reference apk and take readings

### Mobile device model and OS version
   Device: Oppo F1s
   OS: Android 12

### Android Beacon Library version
   altbeacon_version = "2.19.4"

davidgyoung, 2022-12-23:
A few questions:

1. Do you experience this with the app visible in the foreground, or only in the background?

2. What settings are you using for beaconManager.setScanPeriod(...) and beaconManager.setBetweenScanPeriod(...)?  (Do you customize this or use the defaults)

3. For the beacon you are using to test, if you install the BeaconScope app from the Google Play store, what does it tell you that the advertising rate is?  (The app will show this as a value in Hz)

gptshubham595, 2022-12-28:
[image]

davidgyoung, 2022-12-28:
Does the screenshot labelled "Vivo Mobile" correspond to "Device: Oppo F1s, OS: Android 12" as mentioned in the issue?

gptshubham595, 2022-12-28:
Means this issue was observed majorly in vivo and oppo

gptshubham595, 2023-01-03:
I read this, and found OS Bluetooth chips are not configurable
https://stackoverflow.com/questions/62609250/android-beacon-library-long-search-time

so we cannot override or do anything?

davidgyoung, 2023-01-03:
The problem is most likely specific to a customization of the oppo/vivo/oneplus Bluetooth stacks in the closed-source  code forked from the Android Open Source project.  The code you show is likely modified in unknown ways by the manufacturer and this may be the driver of your observations.

It is possible there are workarounds to improve this, but finding them requires time consuming low-level tests with the Android Bluetooth scanning APIs in various conditions (foreground, background, screen on, screen off, etc) to document differences from the AOSP behavior seen on Google Pixel devices when scanning for a reference beacon.

closed 2023-09-02 by davidgyoung