AltBeacon/android-beacon-library #1126: How to stop Beacon service or stop scanning via Beacon library

Spectra6170, 2023-01-20:
I tried with this lines :  beaconManager.stopMonitoring(region), beaconManager.stopRangingBeacons(region). but nothing happens. It continuously scanning reader device. Is there any other way to stop this service?

davidgyoung, 2023-01-20:
Why do you say scanning did not stop?  What specifically do you see to suggest otherwise?

Spectra6170, 2023-01-20:
I have two functionalities in my application. 

1. NFC 
2. Bluetooth LE
So when I Scan using beacon library it works perfect except android Samsung version 13 and 12. When I Switch on NFC then I implemented below lines :
beaconManager.stopMonitoring(region), beaconManager.stopRangingBeacons(region).
But it still Scanning via bluetooth.

closed 2023-06-22 by Spectra6170