AltBeacon/android-beacon-library #1128: Beacon scanning stopped when screen turned off

TyagiAnkit-100, 2023-01-30:
### Expected behavior

Scanning should continue even if screen turn off 

### Actual behavior

It stopped scanning when device screen turned off and start printing "ScanHelper: Beacon simulator not enabled" . It work absolutely fine before screen off and did not print this message. 

### Steps to reproduce this behavior

Simply turn off the screen lock

### Mobile device model and OS version

Android OS 13, Samsung A23S

### Android Beacon Library version

Latest 2.19.5

TyagiAnkit-100, 2023-01-30:
### Logs of my device:

2023-01-30 20:07:19.777 30029-30927/com.aswat.carrefouruae.debug D/BeaconParser: Processing pdu type FF: 1aff4c00021581cbb42c73a096606c7d2fe94627f3a300010001bf0000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 26
2023-01-30 20:07:19.777 30029-30927/com.aswat.carrefouruae.debug D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
2023-01-30 20:07:19.778 30029-30927/com.aswat.carrefouruae.debug D/BeaconParser: Bytes are: 1aff4c00021581cbb42c73a096606c7d2fe94627f3a300010001bf0000000000000000000000000000000000000000000000000000000000000000000000
2023-01-30 20:07:19.785 30029-30927/com.aswat.carrefouruae.debug D/ScanHelper: Beacon packet detected for: id1: 81cbb42c-73a0-9660-6c7d-2fe94627f3a3 id2: 1 id3: 1 with rssi -66
2023-01-30 20:07:19.788 30029-30927/com.aswat.carrefouruae.debug D/RangeState: adding id1: 81cbb42c-73a0-9660-6c7d-2fe94627f3a3 id2: 1 id3: 1 to existing range for: org.altbeacon.beacon.service.RangedBeacon@9994198
2023-01-30 20:07:20.062 30029-30929/com.aswat.carrefouruae.debug D/BeaconParser: Processing pdu type FF: 1aff4c00021581cbb42c73a096606c7d2fe94627f3a300010001bf0000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 26
2023-01-30 20:07:20.064 30029-30929/com.aswat.carrefouruae.debug D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
2023-01-30 20:07:20.064 30029-30929/com.aswat.carrefouruae.debug D/BeaconParser: Bytes are: 1aff4c00021581cbb42c73a096606c7d2fe94627f3a300010001bf0000000000000000000000000000000000000000000000000000000000000000000000
2023-01-30 20:07:20.074 30029-30929/com.aswat.carrefouruae.debug D/ScanHelper: Beacon packet detected for: id1: 81cbb42c-73a0-9660-6c7d-2fe94627f3a3 id2: 1 id3: 1 with rssi -67
2023-01-30 20:07:20.077 30029-30929/com.aswat.carrefouruae.debug D/RangeState: adding id1: 81cbb42c-73a0-9660-6c7d-2fe94627f3a3 id2: 1 id3: 1 to existing range for: org.altbeacon.beacon.service.RangedBeacon@9994198
2023-01-30 20:07:20.345 30029-30932/com.aswat.carrefouruae.debug D/BeaconParser: Processing pdu type FF: 1aff4c00021581cbb42c73a096606c7d2fe94627f3a300010001bf0000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 26
2023-01-30 20:07:20.346 30029-30932/com.aswat.carrefouruae.debug D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
2023-01-30 20:07:20.347 30029-30932/com.aswat.carrefouruae.debug D/BeaconParser: Bytes are: 1aff4c00021581cbb42c73a096606c7d2fe94627f3a300010001bf0000000000000000000000000000000000000000000000000000000000000000000000
2023-01-30 20:07:20.354 30029-30932/com.aswat.carrefouruae.debug D/ScanHelper: Beacon packet detected for: id1: 81cbb42c-73a0-9660-6c7d-2fe94627f3a3 id2: 1 id3: 1 with rssi -69
2023-01-30 20:07:20.356 30029-30932/com.aswat.carrefouruae.debug D/RangeState: adding id1: 81cbb42c-73a0-9660-6c7d-2fe94627f3a3 id2: 1 id3: 1 to existing range for: org.altbeacon.beacon.service.RangedBeacon@9994198
2023-01-30 20:07:20.624 30029-30933/com.aswat.carrefouruae.debug D/BeaconParser: Processing pdu type FF: 1aff4c00021581cbb42c73a096606c7d2fe94627f3a300010001bf0000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 26
2023-01-30 20:07:20.625 30029-30933/com.aswat.carrefouruae.debug D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
2023-01-30 20:07:20.626 30029-30933/com.aswat.carrefouruae.debug D/BeaconParser: Bytes are: 1aff4c00021581cbb42c73a096606c7d2fe94627f3a300010001bf0000000000000000000000000000000000000000000000000000000000000000000000
2023-01-30 20:07:20.633 30029-30933/com.aswat.carrefouruae.debug D/ScanHelper: Beacon packet detected for: id1: 81cbb42c-73a0-9660-6c7d-2fe94627f3a3 id2: 1 id3: 1 with rssi -72
2023-01-30 20:07:20.636 30029-30933/com.aswat.carrefouruae.debug D/RangeState: adding id1: 81cbb42c-73a0-9660-6c7d-2fe94627f3a3 id2: 1 id3: 1 to existing range for: org.altbeacon.beacon.service.RangedBeacon@9994198
2023-01-30 20:07:20.912 30029-30938/com.aswat.carrefouruae.debug D/BeaconParser: Processing pdu type FF: 1aff4c00021581cbb42c73a096606c7d2fe94627f3a300010001bf0000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 26
2023-01-30 20:07:20.912 30029-30938/com.aswat.carrefouruae.debug D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
2023-01-30 20:07:20.913 30029-30938/com.aswat.carrefouruae.debug D/BeaconParser: Bytes are: 1aff4c00021581cbb42c73a096606c7d2fe94627f3a300010001bf0000000000000000000000000000000000000000000000000000000000000000000000
2023-01-30 20:07:20.920 30029-30938/com.aswat.carrefouruae.debug D/ScanHelper: Beacon packet detected for: id1: 81cbb42c-73a0-9660-6c7d-2fe94627f3a3 id2: 1 id3: 1 with rssi -70
2023-01-30 20:07:20.924 30029-30938/com.aswat.carrefouruae.debug D/RangeState: adding id1: 81cbb42c-73a0-9660-6c7d-2fe94627f3a3 id2: 1 id3: 1 to existing range for: org.altbeacon.beacon.service.RangedBeacon@9994198
2023-01-30 20:07:21.198 30029-30945/com.aswat.carrefouruae.debug D/BeaconParser: Processing pdu type FF: 1aff4c00021581cbb42c73a096606c7d2fe94627f3a300010001bf0000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 26
2023-01-30 20:07:21.198 30029-30945/com.aswat.carrefouruae.debug D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
[... cut ...]
2023-01-30 20:07:38.970 30029-30029/com.aswat.carrefouruae.debug D/RangedBeacon: calculated new runningAverageRssi: -73.73469387755102
2023-01-30 20:07:48.977 30029-30029/com.aswat.carrefouruae.debug D/ScanHelper: Beacon simulator not enabled
2023-01-30 20:07:48.979 30029-30029/com.aswat.carrefouruae.debug D/IntentHandler: Calling monitoring notifier: com.aswat.carrefouruae.feature.beacon.BeaconHandler@4c034fd
2023-01-30 20:07:59.000 30029-30029/com.aswat.carrefouruae.debug D/ScanHelper: Beacon simulator not enabled
2023-01-30 20:08:09.011 30029-30029/com.aswat.carrefouruae.debug D/ScanHelper: Beacon simulator not enabled

TyagiAnkit-100, 2023-01-30:
It was detecting beacons and was printing "Beacon packet detected for: id1: 81cbb42c-73a0-9660-6c7d-2fe94627f3a3" but once my screen locked it has not detected any beacons and start printing Beacon simulator not enabled.

I have implement foreground service to san beacons if app is in background which works fine until device screen off but doesn't works after screen off and start printing "Beacon simulator not enabled"

davidgyoung, 2023-01-30:
Can you reproduce this with the official reference app?  It has an option to use a foreground service as well.

davidgyoung, 2023-09-02:
closed due to inactivity

closed 2023-09-02 by davidgyoung