AltBeacon/android-beacon-library #1140: Not able to send beacon data to flutter when app is killed.

PriyankaInoid, 2023-05-05:
### Expected behavior  Need beacon scanning when app is terminated.

### Actual behavior    Giving Error when app killed and not able to send beacon scanning data to flutter(Tried to send a platform message to Flutter, but FlutterJNI was detached from native C++. Could not send. Channel: beacons_plugin_stream. Response ID: 309)

### Steps to reproduce this behavior    

### Mobile device model and OS version   Galaxy M30 and OS version is 10

### Android Beacon Library version  implementation 'org.altbeacon:android-beacon-library:2.19'
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2023-05-05:
I suspect what is going on here is this:

1. The Flutter app stops for some reason (the app crashes or the operating system kills it the app due to memory pressure or another reason)
2. The Android Beacon Library starts the app back up due to beacon detection based on OS-level APIs that allow re-launch of app when this happens. 
3. The Android Beacon Library tries to deliver the beacon detection results to Flutter, but Flutter has not been properly initialized in this case, so the error you mentioned happens.

In order to fix this, you probably need to adjust the way that Flutter is initialized in this case.  In the Android lifecycle, the a native class [Application](https://developer.android.com/reference/android/app/Application) has an onCreate method that is called when the app is re-launched in this case.  Somehow, you need to get Flutter reinitialized in this method.

I am not a Flutter expert so I am afraid I cannot advise further, but this should be a pretty common thing to do.

davidgyoung, 2023-09-02:
closed due to inactivity

closed 2023-09-02 by davidgyoung