AltBeacon/android-beacon-library #1147: feature query

ilker-aktuna, 2023-06-18:
This is not actually an issue, but just a question.
I am trying to set some flags on an Advertisement data for my app.
For example , LE limited Discoverable mode as seen here:
```

1A # Flags value 0x1A = 000011010  
   bit 0 (OFF) LE Limited Discoverable Mode
   bit 1 (ON) LE General Discoverable Mode
   bit 2 (OFF) BR/EDR Not Supported
   bit 3 (ON) Simultaneous LE and BR/EDR to Same Device Capable (controller)
   bit 4 (ON) Simultaneous LE and BR/EDR to Same Device Capable (Host)

```
default Android BLE library does not allow that. Can I use this library to set these flags ?
if possible, how ?
if not, could it be possible by wrapping the Advertiser part of the Android BLE library ?

davidgyoung, 2023-06-18:
Unfortunately, this is not possible with this or any library.  Android Bluetooth APIs do not support configuring these flags.

closed 2023-06-18 by davidgyoung

ilker-aktuna, 2023-06-18:
thank you for the quick answer. Would it be possible, writing a custom library in C++ ?

davidgyoung, 2023-06-18:
You would need to modify the Android Bluetooth stack  to expose a configuration mechanism and flash a custom ROM on a rooted Android device.  His might be possible for your own phone or for phones you distribute, but it won’t work for an app to distribute to others to run on their own phones.

ilker-aktuna, 2023-06-19:
thank you. So we'll expect Android/Google to add these flags and make their stack more flexible.
Which is not possible in near future...