AltBeacon/android-beacon-library #1152: Support for Android 14

masanori-taniguchi, 2023-07-18:
When will you be able to support android14?

I am currently using 2.19, but I get the following error when using the enableForegroundServiceScanning on Android 14

Unable to create service org.altbeacon.beacon.service.BeaconService: java.lang.SecurityException: Starting FGS with type location callerApp= ProcessRecord{38f1531 18832:jp.beacrew.loco.demoapp_staging/u0a284} targetSDK=34 requires permissions: all of the permissions allOf=true [ android.permission.FOREGROUND_SERVICE_LOCATION] any of the permissions allOf=false [android.permission.ACCESS_COARSE_LOCATION, android. permission.ACCESS_FINE_LOCATION] and the app must be in the eligible state/exemptions to access the foreground only permission

stephenruda, 2023-08-02:
When targeting Android 14, every Foreground Service is required to specify a `foregroundServiceType` in the manifest.

I agree that the library needs to be updated to add this specification.  However, you can actually get around this problem by setting the `foregroundServiceType` in your own manifest.  
Try adding this to your `AndroidManifest.xml`
```
<service
    android:name="org.altbeacon.beacon.service.BeaconService"
    android:foregroundServiceType="location" />
```

Manifest entries our automatically merged so the result in the merged version would look something like this...
[image]
The lines that are unhighlighted are the lines we have in our main manifest and the lines that are highlighted are the lines merged from the manifest of a dependency.

This should solve your problem until the library is officially updated to support targeting Android 14.  Hope it helps!

davidgyoung, 2023-08-02:
Thanks for this, @stephenruda.  This is consistent with what I had planned to do to the library for Android 14 — essentially just adding that same `foregroundServiceType=“location”` attribute to the library’s included manifest.   My understanding is that this is the only change needed.

I hope to have this tested and in the library before Sept. 1, 2023 — I have been delayed testing Android 14 myself due to my eligible test phones being dedicated to other projects.  But it is good to know your testing suggests this is a viable solution.

For anyone else waiting for official library support, please use Stephen’s solution above in the meantime.

davidgyoung, 2023-09-02:
Addressed in https://github.com/AltBeacon/android-beacon-library/pull/1155

closed 2023-09-02 by davidgyoung