AltBeacon/android-beacon-library #1157: disable foreground service scanning

broarr, 2023-09-20:
### Expected behavior
I have an application that needs to be able to toggle foreground scans on and off. I expect the foreground scans to stop when I call `beaconManager.disableForegroundServiceScanning()`. The documentation states that `disableForegroundServiceScanning()` should be called after `beaconManager.unbind(this)`. My understanding is that the bind APIs are deprecated in favor of the newer autobind API. Is there a way to stop a foreground scan when using the autobind APIs?

### Actual behavior
The scan continues without interruption.

### Steps to reproduce this behavior
Create a foreground service like in the reference. Call `disableForegroundServiceScanning()` after the foreground scanning has begun

### Mobile device model and OS version
Pixel 3a -- Android 12

### Android Beacon Library version
2.19.6

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2023-09-20:
Yes, you can stop scanning with a foreground service simply by calling `beaconManager.stopRanging(region)` and  `beaconManager.stopMonitoring(region)` for all regions being ranged and monitored.  Doing so will cause the foreground service to exit.   After you have done that, you can call `disableForegroundServiceScanning()` and the start ranging and monitoring again.    

Because the APIs to stop a service on Android are asynchronous, it will take a non-zero amount of time for the service to stop.  Unfortunately there is no callback, so your best bet is to insert a delay of 100ms or so between stopping ranging/monitoring and restarting.

I will leave this issue open as a reminder to update the documentation to include the above.

broarr, 2023-09-21:
Great! Thank you so much for your quick response!

closed 2023-10-27 by davidgyoung