AltBeacon/android-beacon-library #1158: Android 13 support : Beacon service crash

pravingaikwad07, 2023-09-28:
Is there any update on this issue : https://github.com/AltBeacon/android-beacon-library/issues/1141#issue-1710209091

pravingaikwad07, 2023-09-28:
Crash report 1:
`Fatal Exception: android.app.RemoteServiceException$ForegroundServiceDidNotStartInTimeException: Context.startForegroundService() did not then call Service.startForeground(): ServiceRecord{6170cc7 u0 com.wrkspot.employee.prod/org.altbeacon.beacon.service.BeaconService}
       at android.app.ActivityThread.generateForegroundServiceDidNotStartInTimeException(ActivityThread.java:2245)
       at android.app.ActivityThread.throwRemoteServiceException(ActivityThread.java:2216)
       at android.app.ActivityThread.-$$Nest$mthrowRemoteServiceException()
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2508)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loopOnce(Looper.java:226)
       at android.os.Looper.loop(Looper.java:313)
       at android.app.ActivityThread.main(ActivityThread.java:8762)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:604)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)`

closed 2023-09-30 by davidgyoung

pravingaikwad07, 2023-09-30:
So, does it mean that the improvements are available in 'org.altbeacon:android-beacon-library:**2.19.6** ?

davidgyoung, 2023-09-30:
Yes

pravingaikwad07, 2023-09-30:
Thanks for the quick reply. I appreciate it.

I will update on this thread if I find any solution to reduce the frequency of this issue. Also, I will try to post the frequency in decrease of crashes after upgrading 2.19.5 to 2.19.6 as you suggested.

davidgyoung, 2023-09-30:
The improvement was introduced in 2.19, so you will only see a difference if compared inf 2.18.x with 2.19.x

pravingaikwad07, 2023-09-30:
Ok. So I am currently using 2.19.5 and if I upgrade it to 2.19.6, then do you think that there will be improvements in overall frequency of this crash?

pravingaikwad07, 2023-09-30:
Also, wanted to check that is there any plan to release the version 2.20 stable version? I can see that it's in beta version as of now.