AltBeacon/android-beacon-library #1163: Stop detecting iBeacons when app is terminated after 30 minutes

SERCHAT, 2023-10-12:
### Expected behavior
Scan and detect beacons in order to scan cycles even when app is terminated
### Actual behavior
Stops detecting beacons when app is terminated after 30 minutes
### Steps to reproduce this behavior
I am referenced to the example in this `link(https://altbeacon.github.io/android-beacon-library/samples-java.html)`.
I want to scan beacons without using foreground services to show foreground service notification to users. My code is :

```java
 // my beaconsManager's on create method called on MainApplication onCreate method
public void onCreate() {

        beaconsLocalStorage = new BeaconsLocalStorage(cMainApplication);
        BeaconLocalData beaconLocalData = beaconsLocalStorage.getBeaconData();
        if (beaconLocalData.isStarted == true) {

            String uuid = beaconLocalData.uuid;
            String identifier = cMainApplication.getPackageName();
            BeaconManager.getInstanceForApplication(cMainApplication).getBeaconParsers()
                    .add(new BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
            Region region = new Region(identifier,
                    Identifier.parse(uuid), null, null);
            BeaconManager.getInstanceForApplication(cMainApplication).setBackgroundScanPeriod(SCAN_PERIOD);
            BeaconManager.getInstanceForApplication(cMainApplication).setForegroundScanPeriod(SCAN_PERIOD);
            try {
                BeaconManager.getInstanceForApplication(cMainApplication).updateScanPeriods();
            } catch (RemoteException e) {
                throw new RuntimeException(e);
            }
            BeaconManager.getInstanceForApplication(cMainApplication).startMonitoring(region);
            BeaconManager.getInstanceForApplication(cMainApplication).addMonitorNotifier(cMainApplication);

        }
    }
```

And my application class already implements MonitorNotifier to get didEnter and didExit events

```java

public class MainApplication extends MultiDexApplication implements MonitorNotifier {
.
.
.
.

@Override
  public void onCreate() {
    super.onCreate();
    
    // BeaconsManager class is my helper class is not related with android beacons library's BeaconManager class
    beaconsManager = new BeaconsManager(getApplicationContext());
    beaconsManager.onCreate();
  }

 
@Override
public void didEnterRegion(Region region) {

    beaconHelper.didEnterRegion(region);
}

@Override
public void didExitRegion(Region region) {
    
    beaconHelper.didExitRegion(region);
}

.
.
.
.
}
````

With this approach I get the didEnter event even the app is backgrounded/foregrounded. Even my app is terminated in 30 minutes already get this event as expected but after 30 minutes I 'am not getting anything. Also I never get didExit event terminated/foregrounded/backgrounded.
I have spent time over 2 weeks for that but no results please help me, I accept any suggestion for different approaches. 

### Mobile device model and OS version

Android 10
Xiaomi mi A2

### Android Beacon Library version
2.19.5
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2023-10-12:
Please read about the way Xiaomi customizes Android to restrict apps from running in the background [here](https://dontkillmyapp.com/xiaomi).  You might consider adding instructions to your app to show users how to turn off these app killing features.

SERCHAT, 2023-10-12:
Sorry for that I forgot to write on phone model my Xiaomi mi A2 has a google rom it is android one.