AltBeacon/android-beacon-library #1168: Range beacons when app is terminating

SERCHAT, 2023-10-31:
### Expected behavior
Range beacons while app is terminated
### Actual behavior
Getting empty beacon list on didRangeBeaconsInRegion callback.
### Steps to reproduce this behavior
```Java
public class MainApplication extends MultiDexApplication implements MonitorNotifier {

 @Override
    public void onCreate() {
        super.onCreate();
        beaconsManager = new BeaconsManager(getApplicationContext());
        beaconsManager.onCreate();
    }

 @Override
    public void didDetermineStateForRegion(int state, Region region) {

        try {
            if (state == MonitorNotifier.OUTSIDE) {

                beaconsManager.didExitRegion(region);

            }

            if (state == MonitorNotifier.INSIDE) {
               beaconsManager.didEnterRegion(region);
            }

        } catch (Exception e) {
            throw new RuntimeException(e);
        }

    }
}
```

My own BeaconsManager class
```Java

public class BeaconsManager {

    public BeaconsLocalStorage beaconsLocalStorage;
    public MainApplication cMainApplication;

    public static long FOREGROUND_SCAN_PERIOD = 3500l; // ms
    public static long FOREGROUND_SCAN_BETWEEN_PERIOD = 60000l; // ms

    public static long REGION_EXIT_PERIOD = 60000l;//ms

    private final String DID_ENTER_EVENT = "beaconRegionDidEnter";
    private final String DID_EXIT_EVENT = "beaconRegionDidExit";

    private BeaconManager bm;

    public BeaconsManager(Context appContext) {
        beaconsLocalStorage = new BeaconsLocalStorage(appContext);
        MainApplication application = (MainApplication) appContext.getApplicationContext();
        cMainApplication = application;
        BeaconManager.setRegionExitPeriod(REGION_EXIT_PERIOD);
        bm = BeaconManager.getInstanceForApplication(cMainApplication.getApplicationContext());
        bm.getBeaconParsers()
                .add(new BeaconParser()
                        .setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
        bm.setForegroundScanPeriod(FOREGROUND_SCAN_PERIOD);
        bm.setForegroundBetweenScanPeriod(FOREGROUND_SCAN_BETWEEN_PERIOD);
        bm.setBackgroundMode(false);
        try {
            bm.updateScanPeriods();
        } catch (RemoteException e) {
[... 260 lines cut ...]
2- user exits from app(app now is terminated and monitoring started)
3- user enters with it's own device in beacon region
4- monitor notifier detect beacon and call didDetermineState for inside state
--- So far my code works like a charm
5- when state is inside I want to range beacons to get minor and major values (keep in mind app is terminated )
6- start ranging inside a worker which is run on mainThread 
7- I 'am getting here inside the didRangeBeaconsInRegion method an empty beacon list or nothing 

I have been added all required permissions
ACCESS_BACKGROOUND_LOCATION
FINE_LOCATION
BLUETOOTH
BLUETOOTH_ADMIN
BLUETOOTH_SCAN

I want to range beacons when app is terminated without show a foreground notification to the user because my ranging  process will terminated with the first success result of ranging.

I now my phone is in don't kill my app list. I implement to request disable the battery optimizations permissions and user can disable on UI. All my tests done with disabled battery optimizations. 
Please help me!
Thank you for your time and helps!

### Mobile device model and OS version
Xiaomi mi A2 android one - (API 29)
### Android Beacon Library version
2.20

davidgyoung, 2023-10-31:
Based on the description, it sounds like the Xiaomi mi A2 is blocking Bluetooth scans after after app termination.  Because Xiaomi Android extensions are proprietary, it's unfortunately secret and undocumented what is going on here.  Your best bet is to do a variety of tests to reverse engineer what they do in this case.  Here are some things you might try:

1. turn on debug logging in library with beaconManager.setDebug(true)
2. Look for log lines where the app restarts after termination and the library starts a bluetooth scan.
3. Look for operating system log lines after (2) above which may give clues to whether or why the scans are blocked
4. Experiment with bringing the app UI to the foreground to see if that changes the ability to do scans, and see if any operating system log lines give further clues.

Unfortunately, I do not have a Xioami device for experimentation, but I am happy to review any logs you can produce with the info above.

SERCHAT, 2023-10-31:
@davidgyoung Let me explain something. When the device is plugged to the PC to get logs from library all is working fine. After 12 or 13 hours when I 'am entering in a ble region the MonitorNotifier fires the **didDetermineState** with **MonitorNotifier.Iside** . After that I wan to do a **range** to get _**minor**_ and **_major_** value from beacons. Can you tell me can I range beacons when app is terminated ? And if yes how can I do? Monitoring works like a charm when app is **backgrounded/terminated**

davidgyoung, 2023-10-31:
Sorry, I missed your point 4 where you said you get the monitoring entered event successfully.  OK, so it is good that scans are working in the background after termination.

You say this happens after 12 or 13 hours.  Are you saying the ranging callback is not empty fine if you only wait 30 minutes after app termination before triggering a region entry event?  If you do not need to wait 12 hours to reproduce the problem that would make testing much faster!

I note that you have   `bm.setEnableScheduledScanJobs(true);` which means that the library will rely on the Android job scheduler which really overrides your between scan period of 60000l and makes background scans be scheduled every 15 minutes +/-  10 minutes when the phone is not in doze node.  The library also uses a filtered low power scan to get a rapid detection for region entry events.  So this is what should happen:

1. Your app is in the background with no beacons visible and a filtered low power scan going.
2. A beacon is detected by this filtered low power scan, triggering a didEnterRegion event
3. The library will start a new scan for an "extra" ranging cycle that lasts for the duration of your custom scan period (3500 ms).  At the end of this cycle, it will send your app a didRangeBeacons event with a list of those detected.  I will note that this 3.5 seconds is way too short -- the library default is 30 seconds, because in the background, Android turns off the radio receiver most of the time and beacons can take a long time to be detected.  It is possible that 3.5 seconds is just not long enough.
4. After that, I would expect your app to get additional didRangeBeacons events every 15 +/- 10 minutes as allowed by the Android job scheduler.

Can you confirm the above?  Do you get a didRangeBeacons event ~3500ms after the region entry event?  Is this the one that has an empty list of beacons?  And how long does it take for  the next didRangeBeacons callback arrive?  Is it empty as well?

It would also be helpful if you could try the official reference application (Java version is here: https://github.com/AltBeacon/android-beacon-library-reference)  on your phone, which also uses the JobScheduler by default and do both monitoring and ranging in the background.  It uses a default background scan period that is a much longer 30000ms.    I would like to know if you run that app under the same conditions if its ranging results are non-empty.  You'd have to look at LogCat output to see if that is true.

SERCHAT, 2023-11-01:
@davidgyoung  thank you for your reply again. As far as I observed, sometimes  the didRange method is never called  and sometimes is called but result is an empty array, all this happens after a certain period of time. I mean when a install the app recently everything works fine. Tested after 30 minutes 15 minutes 20 minutes and ranging works fine. After a couple of hours I don't know why but I can not range beacons in region. If you see I have some notifications for testing purposes after a couple of a time I see "Ranging started from worker" what is mean startRanging function called successfully and stuck it here. I do not get notifications from inside the didRangeBeaconsInRegion.
I want do short scans to get result very fast and show result quickly. Please tell me @davidgyoung  what am I do wrong ?

davidgyoung, 2023-11-01:
I understand that you want to get ranging detections rapidly after region entry, but that may not be possible in 3.5 seconds.  Different OEMs configure the Bluetooth chip to use a "scan window" that is very small in certain background modes that makes rapid detections impossible.  I would encourage you to experiment with my suggested changes above -- specifically a much longer library SCAN_PERIOD to see if that does make a difference. Even if it is not what you want, it would be a helpful troubleshooting step.

Secondly, I really need you to reproduce this problem with the reference app and attach logs with beaconModule.setDebug(true); showing the empty range result right after a region entry.  Please understand that this form is for reporting and resolving problems with the library, not with third-party apps, so troubleshooting a third party app is out of scope.  Please let me know if you can provide logs reproducing the problem with the reference app and I am happy to assist further.

SERCHAT, 2023-11-02:
@davidgyoung  there are some logs. When I 'am exited and enter a region gives me these logs 
```
2023-11-02 09:03:39.300  3423-27979 BluetoothAdapter        io.test.compnay                      D  isLeEnabled(): ON
2023-11-02 09:03:39.304  3423-28155 BluetoothAdapter        io.test.company                      D  isLeEnabled(): ON
2023-11-02 09:03:39.309  3423-28495 BluetoothAdapter        io.test.company                      D  isLeEnabled(): ON
2023-11-02 09:03:39.311  3423-3445  BluetoothLeScanner      io.test.company                      D  onScannerRegistered() - status=0 scannerId=5 mScannerId=0
2023-11-02 09:03:39.313  3423-31147 BluetoothAdapter        io.test.company                      D  isLeEnabled(): ON
2023-11-02 09:03:39.316  3423-26091 BluetoothAdapter        io.test.company                      D  isLeEnabled(): ON
2023-11-02 09:03:39.318  3423-27527 BluetoothAdapter        io.test.company                      D  isLeEnabled(): ON
2023-11-02 09:03:39.320  3423-19307 BluetoothAdapter        io.test.company                      D  isLeEnabled(): ON
2023-11-02 09:03:39.331  3423-3445  BluetoothLeScanner      io.test.company                      D  onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2023-11-02 09:03:39.342  3423-3445  BluetoothLeScanner      io.test.company                      D  onScannerRegistered() - status=0 scannerId=8 mScannerId=0
2023-11-02 09:03:39.354  3423-3445  BluetoothLeScanner      io.test.company                      D  onScannerRegistered() - status=0 scannerId=12 mScannerId=0
2023-11-02 09:03:39.367  3423-3445  BluetoothLeScanner      io.test.company                      D  onScannerRegistered() - status=0 scannerId=13 mScannerId=0
2023-11-02 09:03:39.376  3423-3445  BluetoothLeScanner      io.test.company                      D  onScannerRegistered() - status=0 scannerId=14 mScannerId=0
2023-11-02 09:03:39.385  3423-3445  BluetoothLeScanner      io.test.company                      D  onScannerRegistered() - status=0 scannerId=15 mScannerId=0
2023-11-02 09:03:39.512  3423-28004 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.514  3423-28157 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.515  3423-28498 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.518  3423-31535 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.520  3423-26097 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.521  3423-27534 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.522  3423-19310 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.791  3423-27981 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.792  3423-28159 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.794  3423-28508 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.797  3423-31544 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.798  3423-26100 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.800  3423-27536 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.801  3423-19360 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.805  3423-27985 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.807  3423-28166 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.808  3423-28509 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.811  3423-31565 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2023-11-02 09:03:39.812  3423-26102 ScanHelper              io.test.company                      D  Decoding beacon. First parser layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
[... cut ...]
2023-11-02 09:07:49.345  2285-3494  BtGatt.GattService      com.android.bluetooth                E  App 'io.test.company' is scanning too frequently
2023-11-02 09:07:49.346  3423-20801 BluetoothLeScanner      io.test.company                      D  onScannerRegistered() - status=6 scannerId=-1 mScannerId=0
2023-11-02 09:08:02.059  3423-9785  newrelic                io.test.company                      I  Connecting, saved state is not available: DataToken{accountId=0, agentId=0}
2023-11-02 09:08:02.176  3423-9785  newrelic                io.test.company                      I  Clearing harvest configuration.
```