AltBeacon/android-beacon-library #1172: startForeground Exception when app is not in foreground

gptshubham595, 2023-11-16:
in BeacoManager getting this exception

```
android.app.ForegroundServiceStartNotAllowedException: startForegroundService() not allowed due to mAllowStartForeground false: service com.olaelectric.companion/org.altbeacon.beacon.service.BeaconService
    android.app.ForegroundServiceStartNotAllowedException$1.createFromParcel(ForegroundServiceStartNotAllowedException.java:54)
    android.app.ForegroundServiceStartNotAllowedException$1.createFromParcel(ForegroundServiceStartNotAllowedException.java:50)
    android.os.Parcel.readParcelableInternal(Parcel.java:4787)
    android.os.Parcel.readParcelable(Parcel.java:4755)
    android.os.Parcel.createExceptionOrNull(Parcel.java:3018)
    android.os.Parcel.createException(Parcel.java:3007)
    android.os.Parcel.readException(Parcel.java:2990)
    android.os.Parcel.readException(Parcel.java:2932)
    android.app.IActivityManager$Stub$Proxy.startService(IActivityManager.java:6214)
    android.app.ContextImpl.startServiceCommon(ContextImpl.java:1967)
    android.app.ContextImpl.startForegroundService(ContextImpl.java:1933)
    android.content.ContextWrapper.startForegroundService(ContextWrapper.java:839)
    org.altbeacon.beacon.BeaconManager.bindInternal(BeaconManager.java:447)
    org.altbeacon.beacon.BeaconManager.autoBind(BeaconManager.java:1854)
    org.altbeacon.beacon.BeaconManager.startRangingBeacons(BeaconManager.java:996)
    com.olaelectric.ble.data.proximity.ProximityRepository.startRanging(ProximityRepository.java:195)
    com.olaelectric.ble.data.proximity.ProximityRepository.access$startRanging(ProximityRepository.java:30)
    com.olaelectric.ble.data.proximity.ProximityRepository$readBeacons$2.invokeSuspend(ProximityRepository.kt:150)
    kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
    kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
    kotlinx.coroutines.internal.LimitedDispatcher.run(LimitedDispatcher.kt:42)
    kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:95)
    kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
    kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
    kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
    kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)

```

This seems to happen for the following reason

https://stackoverflow.com/a/70666991

It is happening here [see // <---- AT THIS PLACE ITS HAPPENING !! ---> ] 

```
public void bindInternal(@NonNull InternalBeaconConsumer consumer) {
        if (!isBleAvailableOrSimulated()) {
            LogManager.w(TAG, "Method invocation will be ignored.");
            return;
        }
        synchronized (consumers) {
            ConsumerInfo newConsumerInfo = new ConsumerInfo();
            ConsumerInfo alreadyBoundConsumerInfo = consumers.putIfAbsent(consumer, newConsumerInfo);
            if (alreadyBoundConsumerInfo != null) {
                LogManager.d(TAG, "This consumer is already bound");
            }
            else {
                LogManager.d(TAG, "This consumer is not bound.  Binding now: %s", consumer);
                if (mIntentScanStrategyCoordinator != null) {
                    mIntentScanStrategyCoordinator.start();
                    consumer.onBeaconServiceConnect();
                }
                else if (mScheduledScanJobsEnabled) {
                    LogManager.d(TAG, "Not starting beacon scanning service. Using scheduled jobs");
                    consumer.onBeaconServiceConnect();
                }
                else {
                    LogManager.d(TAG, "Binding to service");
                    Intent intent = new Intent(consumer.getApplicationContext(), BeaconService.class);
                    if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O &&
                            this.getForegroundServiceNotification() != null) {
                        if (isAnyConsumerBound()) {
                            LogManager.i(TAG, "Not starting foreground beacon scanning" +
                                    " service.  A consumer is already bound, so it should be started");
                        }
                        else {
                            LogManager.i(TAG, "Starting foreground beacon scanning service.");
                            mContext.startForegroundService(intent); // <---- AT THIS PLACE ITS HAPPENING !! --->
                        }
                    }
                    else {
                    }
                    consumer.bindService(intent, newConsumerInfo.beaconServiceConnection, Context.BIND_AUTO_CREATE);
                }
                LogManager.d(TAG, "consumer count is now: %s", consumers.size());
            }
        }
    }
```

I can pick this up.

davidgyoung, 2023-11-16:
> public void ***@***.*** InternalBeaconConsumer consumer) {

What version of the library are you running?  There have been fixes put in
for this as of version 2.20.2.  Please make sure you have tested with that
version.

See details of fixes in #1165 #1162 #1155

gptshubham595, 2023-11-30:
tried multiple attempts to import 2.20.2 but implementation only fails for this version

https://jitpack.io/#AltBeacon/android-beacon-library/

closed 2023-12-08 by davidgyoung