AltBeacon/android-beacon-library #1173: StartupBroadcastReceiver throws NullPointerException

rahul-kumawat-vts, 2023-11-24:
**Expected behavior**
StartupBroadcastReceiver should start without exception

### Actual behavior
StartupBroadcastReceiver is throwing NullPointer exception while trying to retrieve action from Intent as Action is null.

Line 33.
        if (intent.getAction().equals(Intent.ACTION_BOOT_COMPLETED)) {

`Fatal Exception: java.lang.RuntimeException: Unable to start receiver org.altbeacon.beacon.startup.StartupBroadcastReceiver: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean java.lang.String.equals(java.lang.Object)' on a null object reference
       at android.app.ActivityThread.handleReceiver(ActivityThread.java:4903)
       at android.app.ActivityThread.-$$Nest$mhandleReceiver()
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2420)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loopOnce(Looper.java:226)
       at android.os.Looper.loop(Looper.java:313)
       at android.app.ActivityThread.main(ActivityThread.java:8762)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:604)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)`

`Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'boolean java.lang.String.equals(java.lang.Object)' on a null object reference
       at org.altbeacon.beacon.startup.StartupBroadcastReceiver.onReceive(StartupBroadcastReceiver.java:33)
       at android.app.ActivityThread.handleReceiver(ActivityThread.java:4894)
       at android.app.ActivityThread.-$$Nest$mhandleReceiver()
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2420)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loopOnce(Looper.java:226)
       at android.os.Looper.loop(Looper.java:313)
       at android.app.ActivityThread.main(ActivityThread.java:8762)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:604)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)`

### Steps to reproduce this behavior

### Mobile device model and OS version

[image]

### Android Beacon Library version
org.altbeacon:android-beacon-library:2.20.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

juliankotrba, 2023-11-28:
@rahul-kumawat-vts it looks like they already fixed this issue (https://github.com/AltBeacon/android-beacon-library/commit/9207ad205b927db4ece2eec5f3b76cf402cfefe3) with version 2.20.2.

davidgyoung, 2023-12-08:
Fixed in https://github.com/AltBeacon/android-beacon-library/commit/9207ad205b927db4ece2eec5f3b76cf402cfefe3 and release version 2.20.2

closed 2023-12-08 by davidgyoung