AltBeacon/android-beacon-library #1174: No beacons after locked screen

juliankotrba, 2023-11-28:
### Behavior + steps to reproduce this behavior

I am creating this ticket because we experience a weird problem with only one of our test devices (Pixel 7). The issue is one of the classic ones: as soon as we lock the phone and the display goes dark, the ranging stops.

I already tried to change all the different library settings in our app but the behavior stayed the same. To simplify the analysis of the problem for you, I checked out the [android-beacon-library-reference-kotlin](https://github.com/davidgyoung/android-beacon-library-reference-kotlin) project ([#2aa42a44](https://github.com/davidgyoung/android-beacon-library-reference-kotlin/commit/2aa42a44bae20f2abcdd29847fcea4d39fa9ab11)) and tried to reproduce the problem there, which was actually reproducible. To prove that the problem occurs, I made some small modifications to the project.

1) I updated the unique region id to match our beacon scheme.

2) I uncommented the following lines

`````
beaconManager.setBackgroundBetweenScanPeriod(0);
beaconManager.setBackgroundScanPeriod(1100);
`````

3) I added a beep tone when central ranging observer is called:

````
...
if (rangeAgeMillis < 10000) {
    val toneGen1 = ToneGenerator(AudioManager.STREAM_MUSIC, 100)
    toneGen1.startTone(ToneGenerator.TONE_CDMA_PIP, 150)
    Log.d(MainActivity.TAG, "Ranged: ${beacons.count()} beacons")
    ...
}
...
````

4) I set the debug mode to true and added a custom logger, which is a replica of the VerboseAndroidLogger except that I prefixed the tag with "DEBUG" in order to make it easily filterable in the Logcat.

Then I ran the app on our Pixel 8 (Android 14) and on our Pixel 7. On the Pixel 8, the beeping continued after locking the phone while it stopped on the Pixel 7. I suspect that the problem is due to a device setting, but I haven't been able to identify it yet.

While reading through the GH issues I stumbled across your comment here where you mention three possible reasons why beacons are not detected when the screen is off: https://github.com/AltBeacon/android-beacon-library/issues/944#issuecomment-579956716

Ad 1. If the hardware code hasn't been set correctly, this would then lead to problems on different devices right? Since it works on almost all other phones I assume this one is correct.

Ad 2. In case the beacon parser layout is incorrect, I also assume it would also affect other devices.

Ad 3. I uninstalled all BLE scanning apps and tried again, no change in behavior

After the screen becomes active again, the beeping  and therefore the ranging starts again. 

### Mobile device model and OS version

Pixel 7 
Android 14
Build number: AP11.231020.016.A1
Security update: 1 November 2023
This devices is enrolled in the beta program

### Library Logs:
[altbeacon-logs.txt](https://github.com/AltBeacon/android-beacon-library/files/13486563/altbeacon-logs.txt)

We would really appreciate it if you could shed some light on this behavior.
Please let me know if any other important information is missing or if there is anything I can assist you with.

Thanks 
Julian

davidgyoung, 2023-12-05:
Hi @juliankotrba thank you for this report. 

I was able to reoproduce your results.  At first glance, this appears to be a change to Android 14.  It seems that with the screen off, scans with empty Bluetooth LE scan filters return no results.  (This  is how Samsung forks of Android behaved since 8.1, but Pixels did not appear to work this way as recently as Android 13).

  I tried modifying the Android Beacon Library code to use non-empty scan filters (like the library already does for Samsung) and the ranging started working even with the screen off:  https://github.com/AltBeacon/android-beacon-library/pull/1176
  
  Does your app target Android 14?  Do you know if this matters?

juliankotrba, 2023-12-05:
Hi @davidgyoung. 

> I tried modifying the Android Beacon Library code to use non-empty scan filters (like the library already does for Samsung) and the ranging started working even with the screen off: https://github.com/AltBeacon/android-beacon-library/pull/1176

Nice, I will test your fix and give you feedback later!

> Does your app target Android 14? Do you know if this matters?
Yes, we target Android 14 but I am not sure if this matters. As far as I remember I did some tests were I went back in the git history of the project and it also did not work when we did not target Android 14.

When you were able to reproduce the issue, on which Pixel did you test? For us, the issue only appeared on our Pixel 7, but not Pixel 8 (tested on one Pixel 7 and two Pixel 8).

Thanks for your help!

juliankotrba, 2023-12-06:
@davidgyoung Pixel 7 is working again! :clap:

davidgyoung, 2023-12-06:
Good to hear.  For the record, my only tests have been with the Pixel 7
which is the only Android 14 device I have hand now.  I can try others next
week.  It would be good to fully characterize so the library can implement
a proper fix for all.

davidgyoung, 2023-12-08:
I did a few more tests with Pixel 7 Android 14, varying the target SDK in the app:

```
Model      OS         Library Version Target SDK   Ranges with Screen Off  Notes
--------  ----------  --------------- ----------   ----------------------  -----
Pixel 7   Android 14  2.20.2          33           YES                     Clean install
Pixel 7   Android 14  2.20.2          34           NO                      Clean install
Pixel 7   Android 14  2.20.2          34           YES                     Upgrade from above
```

So the problem only happens on the Pixel 7 if set the targetSdk to 34.   Further, if you set the targetSdk to 33, run the app, then change the targetSdk to 34 and run the app again (without uninstall), the behavior is not seen.  So it seems the behavior only applies to apps targeting Android 14 and newly installed on Pixel 7s.

davidgyoung, 2024-01-09:
@kangjjang, this thread is dedicated to detection problems with the screen off ONLY on Android 14 on Pixel phones.  It sounds like you need general help configuring an app for fast background detections.  Please see this page: [Alternate Scan Strategies] (https://altbeacon.github.io/android-beacon-library/detection_times.html).  You will need to configure the library to use either a [Foreground Service](https://altbeacon.github.io/android-beacon-library/foreground-service.html) or the Intent Scan Strategy described at the very bottom of [this page](https://altbeacon.github.io/android-beacon-library/detection_times.html).

If you need further help configuring your custom app, please open a new question on StackOverflow.com, as that is the proper forum for assistance with using the library with third party apps.  This thread is ONLY for trouble detecting with the screen off on Android 14 on Pixel phones.

akoskuczi-bw, 2024-01-31:
Hi @davidgyoung, 
Same issue with Motorola g54 even with Android version 13.
Build: 5.10.168-android12-9-00003-g828a3b1f7462-ab10564101

From logcat:  "BtGatt.ScanManager com.android.bluetooth W  Cannot start unfiltered scan in screen-off. This scan will be resumed later: 14"

davidgyoung, 2024-01-31:
@akoskuczi-bw, the logcat line you mention has been around since Android 8.1, and as originally created, applied only to the case where there are zero scan filters attached to the scan:

https://android.googlesource.com/platform/packages/apps/Bluetooth/+/319aeae6f4ebd13678b4f77375d1804978c4a1e1%5E%21/#F1

When do you see this log line with Android 13?  Does it happen every time you try to start a scan with the screen off or just sometimes?    Is in an app using this library?

akoskuczi-bw, 2024-01-31:
Yes, it is on Android 13. The scanning stops working consistently when the screen is off, and it restarts when the screen is on (not necessarily unlocked). I am using it with the following Flutter plugin: https://github.com/Lukangi/beacon_scanner/tree/main/beacon_scanner

davidgyoung, 2024-01-31:
Researching AOSP commits, I found this change from July 2023 that is now part of Android 14:

https://android-review.googlesource.com/c/platform/packages/modules/Bluetooth/+/2654980

Interestingly the commit is from a Motorola engineer, so perhaps the change went in to an earlier proprietary change on Motorola that does not show up in AOSP.

OK, so this sounds believable.  However, this does not explain the OP's report that the restriction did not apply to Pixel 8 with Android 14, and my observation above that it did not apply to Pixel 6a with Android 14.  Could it be that this AOSP change was not in the Android 14 builds for the Pixel 6a and Pixel 8?  What else could explain this?

davidgyoung, 2024-02-01:
I have merged the change to fix this for Android 14. This does not address the problem for Motorola Android 13 as reported by @akoskuczi-bw.  Does anybody else on this thread have a Motorola device with Android 13 to confirm the problem there?

If we can get a confirmation of this on a second Motorola Android 13 device, I will make a new change that creates a scan filter for those devices, too.  (I am reluctant to do this without confirmation since it could be a breaking change for apps that have been running on Motorola phones for the past year.)  Until this is resolved I will leave this issue open.

akoskuczi-bw, 2024-02-01:
@davidgyoung  I absolutely understand your position regarding Motorola. I'm wondering if anyone can confirm the issue with Motorola. Thank you very much for your time and efforts.

juliankotrba, 2024-02-28:
@akoskuczi-bw @davidgyoung FYI I can't confirm the issue on the Motorola, however, I can reproduce it on our "new" Asus Zenfone 9 running with Android 13. Maybe also on a OnePlus But for that I need to do some more investigation.

Adding `Build.MANUFACTURER.equalsIgnoreCase("asus")` to the scan filter check if statement solves the issue on my Asus.

devirtua, 2024-07-29:
I am using Realme GT2, Android 14.
Same problem appears.

alexmicleaacc, 2024-10-31:
> I have merged the change to fix this for Android 14. This does not address the problem for Motorola Android 13 as reported by @akoskuczi-bw. Does anybody else on this thread have a Motorola device [...]

@davidgyoung I can confirm this for Motorola g42, Android 13. 
The reference app does not detect any beacon immediately after putting the app in background mode, and resumes detection when the screen is on.
I've downloaded the library locally, added (Build.MANUFACTURER.equalsIgnoreCase("motorla") to the code and the reference app is now working as expected when in background mode. 

I can create a pull request with the fix as well, just let me know.