AltBeacon/android-beacon-library #1178: BeaconService.startForegroundIfConfigured

rahul-kumawat-vts, 2024-01-19:
```
Fatal Exception: java.lang.RuntimeException
Unable to create service org.altbeacon.beacon.service.BeaconService: android.app.ForegroundServiceStartNotAllowedException: Service.startForeground() not allowed due to mAllowStartForeground false: service com.risesoftware.storybrampton/org.altbeacon.beacon.service.BeaconService
```
```

Caused by android.app.ForegroundServiceStartNotAllowedException: Service.startForeground() not allowed due to mAllowStartForeground false: service com.risesoftware.storybrampton/org.altbeacon.beacon.service.BeaconService
       at android.app.ForegroundServiceStartNotAllowedException$1.createFromParcel(ForegroundServiceStartNotAllowedException.java:54)
       at android.app.ForegroundServiceStartNotAllowedException$1.createFromParcel(ForegroundServiceStartNotAllowedException.java:50)
       at android.os.Parcel.readParcelableInternal(Parcel.java:4870)
       at android.os.Parcel.readParcelable(Parcel.java:4852)
       at android.os.Parcel.createExceptionOrNull(Parcel.java:3052)
       at android.os.Parcel.createException(Parcel.java:3041)
       at android.os.Parcel.readException(Parcel.java:3024)
       at android.os.Parcel.readException(Parcel.java:2966)
       at android.app.IActivityManager$Stub$Proxy.setServiceForeground(IActivityManager.java:6761)
       at android.app.Service.startForeground(Service.java:775)
       at org.altbeacon.beacon.service.BeaconService.startForegroundIfConfigured(BeaconService.java:277)
       at org.altbeacon.beacon.service.BeaconService.onCreate(BeaconService.java:201)
       at android.app.ActivityThread.handleCreateService(ActivityThread.java:4671)
       at android.app.ActivityThread.-$$Nest$mhandleCreateService()
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2269)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loopOnce(Looper.java:205)
       at android.os.Looper.loop(Looper.java:294)
       at android.app.ActivityThread.main(ActivityThread.java:8194)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
```

### Mobile device model and OS version

[image]

### Android Beacon Library version

**org.altbeacon:android-beacon-library:2.20.3**

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2024-01-19:
Thanks for this report.

We added the code below to prevent that in #1162 released in October in 2.20.1, but it looks like it will not work in your case:

```
            try {
                this.startForeground(notificationId, notification);
            }
            catch (SecurityException exception) {
                // https://issuetracker.google.com/issues/294408576
                LogManager.w(TAG, "Call to service startForeground() threw a SecurityException.  The Foreground Service for beacon scanning may have started anyway, but this behavior might change in  different conditions or a future Android version.");
            }
```

The reason is because you are getting a [android.app.ForegroundServiceStartNotAllowedException](https://developer.android.com/reference/android/app/ForegroundServiceStartNotAllowedException)  which does not inherit from SecurityException, it inherits from IllegalStateException.

We probably ned to add another catch here for that exception.  This will prevent the crash reports, but it won't change the fact that we won't be able to start the foreground service in these cases.

@stephenruda any thoughts here?

stephenruda, 2024-01-19:
Hi!  Yes I have some thoughts.

`ForegroundServiceStartNotAllowedException` actually dates back to Android 12.  After re-reading the Android 14 docs, I don't think there was anything new in Android 14 that would have caused this exception that wasn't already happening in Android 12 and 13.
This exception is actually a child of `ServiceStartNotAllowedException` [which you are already catching here.](https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/java/org/altbeacon/beacon/BeaconManager.java#L488)
Catching the exception using the parent class should be sufficient in Java.  It should catch both the `ServiceStartNotAllowedException` and `ForegroundServiceStartNotAllowedException`.

My understanding was that this exception was only ever thrown in calls to `startForegroundService()` and not calls to `startForeground()` like we are seeing in the mentioned stacktrace.

I have other foreground services in my app (unrelated to the `android-beacon-library`) that catches this exception.  Just like in the `BeaconManager`, I am catching this exception on `startForegroundService()` and not on `startForeground()`.  I have never had an issue with it.

I finally just shipped my Android 14 changes for my app last week.  I have not noticed this exception yet in my app but I will keep my eye out for it.

My conclusion would be...
1)  Maybe something has changed in Android 14 and now this exception is being thrown from calls to `startForeground()` as well as `startForegroundService()`.  I haven't experienced that yet myself so maybe we should wait until we can collect more data.  My app has 4 other foreground services alongside the `BeaconService` so if this is the case it shouldn't take me too long before I see it.  If that ends up being the case, we will have to [fallback to the job scheduler](https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/java/org/altbeacon/beacon/BeaconManager.java#L491) just like we would when catching the exception from the `BeaconManager`.  I don't think we could just "eat" the error and act like nothing happened like we do with the `SecurityException`.
2)  Maybe this user has the library implemented in a unique way.  For example, maybe they are calling `startForegroundService()` themselves rather than allowing the `BeaconManager` to do it for them?  If this is something they can recreate consistently, it would be great to see some code or have them recreate the problem in the reference application.

davidgyoung, 2024-01-19:
@rahul-kumawat-vts can you please share two things:

1. Any entry in your AndroidManifest.xml that refers to "BeaconService".  Below is what it might look like:

```
        <service 
            android:name=".service.BeaconService"
            android:enabled="true"
            android:exported="false"
            android:isolatedProcess="false"
            android:foregroundServiceType="location"
            android:label="beacon"            
            />
```

2. The code that configures the foreground service and starts ranging or monitoring.   Below is what the official Kotlin reference app does to set this up to give you a clue what we are looking to see.  The key part is the code that calls `enableForegroundServiceScanning`  and then `startRangingBeacons` or `startMonitoring`.

```
        val builder = Notification.Builder(this, "BeaconReferenceApp")
        builder.setSmallIcon(R.drawable.ic_launcher_background)
        builder.setContentTitle("Scanning for Beacons")
        val intent = Intent(this, MainActivity::class.java)
        val pendingIntent = PendingIntent.getActivity(
                this, 0, intent, PendingIntent.FLAG_UPDATE_CURRENT + PendingIntent.FLAG_IMMUTABLE
        )
        builder.setContentIntent(pendingIntent);
        val channel =  NotificationChannel("beacon-ref-notification-id",
            "My Notification Name", NotificationManager.IMPORTANCE_DEFAULT)
        channel.setDescription("My Notification Channel Description")
        val notificationManager =  getSystemService(
                Context.NOTIFICATION_SERVICE) as NotificationManager
        notificationManager.createNotificationChannel(channel);
        builder.setChannelId(channel.getId());
        Log.d(TAG, "Calling enableForegroundServiceScanning")
        BeaconManager.getInstanceForApplication(this).enableForegroundServiceScanning(builder.build(), 456);
        Log.d(TAG, "Back from  enableForegroundServiceScanning")

        beaconManager.startMonitoring(region)
        beaconManager.startRangingBeacons(region)
```

stephenruda, 2024-01-19:
[Looking a bit more into this issue](https://stackoverflow.com/questions/77680265/service-startforeground-not-allowed-due-to-mallowstartforeground-false-androi?noredirect=1&lq=1), my leading theory is that there is something new in Android 14 that is causing this.

Because the `BeaconService` is sticky, it will attempt to restart itself if it is ever killed.  This is likely why we are seeing calls to `startForeground()` without any calls from the `BeaconManager`.
[The docs say that it does not restrict restarting a sticky foreground service.](https://developer.android.com/reference/android/app/Service#START_STICKY) [Maybe that is an undocumented change or bug in Android 14](https://issuetracker.google.com/issues/307329994)?

Finally, the reason that I have not seen this in my app is due to the fact that I force users to turn off battery optimizations for my app prior to using my foreground services.  This allows me to [start foreground services from the background](https://developer.android.com/develop/background-work/services/foreground-services#background-start-restriction-exemptions) and therefore my app would never throw an exception when this scenario occurs.

I am not suggesting that the Beacon Library should force users to turn this setting off - I am just explaining why I haven't seen the exception in my app.

Ultimately, what do we do about this issue?  I am not really sure.  To me it does feel like an Android 14 bug but I doubt that it will be fixed any time soon.  I suppose falling back to the job scheduler is better than crashing?

davidgyoung, 2024-01-19:
Yes, I agree that falling back to the JobScheduler is probably the solution in this case, but it would be good to understand the specific sequence that triggers it.  @stephenruda do you think you might be able to re-enable background optimizations on your app on a test phone and let it run for a day or so to see if you can reproduce a crash?  

I am going to try running the reference app on the latest Android 14 to see if I can reproduce, but I may not be able to, because I suspect the foreground service will just keep running for days.  I wonder if I can force a crash of the foreground service somehow to test your sticky restart theory.

stephenruda, 2024-01-19:
Yes, I can do that.

I am unsure how to consistently reproduce the problem.  I don't think there is a way we can manually trigger 'killing' the service in the same way that the OS does.  I think the OS kills the service when system resources get low.  As you suggested, it might take some time before we actually see the problem occur.

davidgyoung, 2024-01-21:
I was unable to reproduce the problem running the Kotlin reference app (configured to use a foreground service) on a Pixel 6a on the latest Android 14 (Security update January 5, 2014), then crashing the app connected to `adb` on my workstation with this command:

`adb shell am crash org.altbeacon.beaconreference`

The app crashed, then restarted as expected, and the foreground service started successfully as shown in the logs  below.

```
2024-01-20 23:04:51.338 24146-24146 AndroidRuntime          org.altbeacon.beaconreference        D  Shutting down VM
2024-01-20 23:04:51.342 24146-24146 AndroidRuntime          org.altbeacon.beaconreference        E  FATAL EXCEPTION: main
                                                                                                    Process: org.altbeacon.beaconreference, PID: 24146
                                                                                                    android.app.RemoteServiceException$CrashedByAdbException: shell-induced crash
                                                                                                    	at android.app.ActivityThread.throwRemoteServiceException(ActivityThread.java:2115)
                                                                                                    	at android.app.ActivityThread.-$$Nest$mthrowRemoteServiceException(Unknown Source:0)
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2394)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:106)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:205)
                                                                                                    	at android.os.Looper.loop(Looper.java:294)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8248)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
2024-01-20 23:04:51.371 24146-24146 Process                 org.altbeacon.beaconreference        I  Sending signal. PID: 24146 SIG: 9
---------------------------- PROCESS ENDED (24146) for package org.altbeacon.beaconreference ----------------------------
---------------------------- PROCESS STARTED (24626) for package org.altbeacon.beaconreference ----------------------------
2024-01-20 23:04:51.510 24626-24626 Compatibil...geReporter org.altbeacon.beaconreference        D  Compat change id reported: 242716250; UID 10244; state: ENABLED
2024-01-20 23:04:51.520 24626-24626 nativeloader            org.altbeacon.beaconreference        D  Configuring clns-4 for other apk /data/app/~~Jv2bdx7ebRgX_okn7mVOOw==/org.altbeacon.beaconreference-fJ-Z1Z_BvOyOuJs1M4fQaA==/base.apk. target_sdk_version=34, uses_libraries=, library_path=/data/app/~~Jv2bdx7ebRgX_okn7mVOOw==/org.altbeacon.beaconreference-fJ-Z1Z_BvOyOuJs1M4fQaA==/lib/arm64, permitted_path=/data:/mnt/expand:/data/user/0/org.altbeacon.beaconreference
2024-01-20 23:04:51.529 24626-24626 GraphicsEnvironment     org.altbeacon.beaconreference        V  Currently set values for:
2024-01-20 23:04:51.529 24626-24626 GraphicsEnvironment     org.altbeacon.beaconreference        V    angle_gl_driver_selection_pkgs=[]
2024-01-20 23:04:51.529 24626-24626 GraphicsEnvironment     org.altbeacon.beaconreference        V    angle_gl_driver_selection_values=[]
2024-01-20 23:04:51.529 24626-24626 GraphicsEnvironment     org.altbeacon.beaconreference        V  ANGLE GameManagerService for org.altbeacon.beaconreference: false
2024-01-20 23:04:51.530 24626-24626 GraphicsEnvironment     org.altbeacon.beaconreference        V  org.altbeacon.beaconreference is not listed in per-application setting
2024-01-20 23:04:51.530 24626-24626 GraphicsEnvironment     org.altbeacon.beaconreference        V  Neither updatable production driver nor prerelease driver is supported.
2024-01-20 23:04:51.535 24626-24626 BeaconParser            org.altbeacon.beaconreference        D  Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2024-01-20 23:04:51.546 24626-24626 BeaconManager           org.altbeacon.beaconreference        I  BeaconManager started up on pid 24626 named 'org.altbeacon.beaconreference' for application package 'org.altbeacon.beaconreference'.  isMainProcess=true
2
```

davidgyoung, 2024-01-21:
App also restarted fine  automatically after reboot with foreground service

rahul-kumawat-vts, 2024-01-22:
@davidgyoung We got so many crash on over firebase. i have tried to replicate but not able to replicate. 

In our app we are running a service to handle the beacon 

`import android.app.*
import android.content.Intent
import android.content.pm.ServiceInfo
import android.os.Build
import android.os.IBinder
import android.os.RemoteException
import androidx.annotation.RequiresApi
import androidx.core.app.NotificationCompat
import androidx.core.app.NotificationManagerCompat
import dagger.hilt.android.AndroidEntryPoint
import org.altbeacon.beacon.*
import timber.log.Timber

@AndroidEntryPoint
class MyAppBeaconService : Service(), InternalBeaconConsumer, MonitorNotifier, RangeNotifier {
    
    private var beaconManager: BeaconManager? = null
    private val region = Region("backgroundRegion", null, null, null)

    private var lastShownNotificationId = 0
    private var builder: NotificationCompat.Builder? = null

    companion object {
        const val IBEACON_LAYOUT = "IBEACON_LAYOUT_ACTUAL_VALUE"
        const val EDDYSTONE_UID_LAYOUT = BeaconParser.EDDYSTONE_UID_LAYOUT
        const val EDDYSTONE_URL_LAYOUT = BeaconParser.EDDYSTONE_URL_LAYOUT
        const val EDDYSTONE_TLM_LAYOUT = BeaconParser.EDDYSTONE_TLM_LAYOUT
    }

    override fun onCreate() {
        super.onCreate()
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
            startBeaconForegroundService(NOTIFICATION_ID)
        }
    }

    override fun onStartCommand(intent: Intent?, flags: Int, startId: Int): Int {
        Timber.d("MyAppBeaconService onStartCommand")
        if (intent?.action == STOP_BEACON_FOREGROUND_ACTION) {
            stopForeground(true)
            stopSelf()
        } else {
            try {
                beaconManager = BeaconManager.getInstanceForApplication(this)
                beaconManager?.beaconParsers?.add(BeaconParser().setBeaconLayout(IBEACON_LAYOUT))
                beaconManager?.beaconParsers?.add(BeaconParser().setBeaconLayout(EDDYSTONE_UID_LAYOUT))
                beaconManager?.beaconParsers?.add(BeaconParser().setBeaconLayout(EDDYSTONE_URL_LAYOUT))
                beaconManager?.beaconParsers?.add(BeaconParser().setBeaconLayout(EDDYSTONE_TLM_LAYOUT))

                beaconManager?.backgroundBetweenScanPeriod = 1000L
                beaconManager?.foregroundBetweenScanPeriod = 1000L
                beaconManager?.enableForegroundServiceScanning(builder?.build(), NOTIFICATION_ID)
                beaconManager?.setEnableScheduledScanJobs(false)
                beaconManager?.backgroundScanPeriod = 1100L
                beaconManager?.bindInternal(this)
                beaconManager?.updateScanPeriods()

            } catch (e: Exception) {
                e.printStackTrace()
            }
        }
        return START_STICKY
    }

    override fun onBind(intent: Intent?): IBinder? {
        return null
[... 110 lines cut ...]
        beaconManager?.addRangeNotifier(this)
        try {
            beaconManager?.startMonitoring(region)
            beaconManager?.startRangingBeacons(region)
        } catch (e: RemoteException) {
            e.printStackTrace()
            Timber.d("MyAppBeaconService - onBeaconServiceConnect - errStartMonitoringRanging: ${e.message}, exception: ${e.printStackTrace()}")
        }
    }

    private fun startBeaconService() {
        try {
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
                context.startForegroundService(Intent(context, MyAppBeaconService::class.java))
            } else {
                context.startService(Intent(context, MyAppBeaconService::class.java))
            }
        } catch (exception: Exception) {
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.S
                && exception is ForegroundServiceStartNotAllowedException) {
                Timber.d("MyAppBeaconService - startBeaconService - ForegroundServiceStartNotAllowedException: ${exception.message}")
            }
        }
    }
}`

rahul-kumawat-vts, 2024-01-22:
@davidgyoung We got so many crash on over firebase. i have tried to replicate but not able to replicate. 

In our app we are running a service to handle the beacon. Please check the code we are using in our app.

````
<service
  android:name=".MyAppBeaconService"
  android:enabled="true"
  android:exported="false"
  android:foregroundServiceType="location" />

import android.app.*
import android.content.Intent
import android.content.pm.ServiceInfo
import android.os.Build
import android.os.IBinder
import android.os.RemoteException
import androidx.annotation.RequiresApi
import androidx.core.app.NotificationCompat
import androidx.core.app.NotificationManagerCompat
import dagger.hilt.android.AndroidEntryPoint
import org.altbeacon.beacon.*
import timber.log.Timber

@AndroidEntryPoint
class MyAppBeaconService : Service(), InternalBeaconConsumer, MonitorNotifier, RangeNotifier {
    
    private var beaconManager: BeaconManager? = null
    private val region = Region("backgroundRegion", null, null, null)

    private var lastShownNotificationId = 0
    private var builder: NotificationCompat.Builder? = null

    companion object {
        const val IBEACON_LAYOUT = "IBEACON_LAYOUT_ACTUAL_VALUE"
        const val EDDYSTONE_UID_LAYOUT = BeaconParser.EDDYSTONE_UID_LAYOUT
        const val EDDYSTONE_URL_LAYOUT = BeaconParser.EDDYSTONE_URL_LAYOUT
        const val EDDYSTONE_TLM_LAYOUT = BeaconParser.EDDYSTONE_TLM_LAYOUT
    }

    override fun onCreate() {
        super.onCreate()
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
            startBeaconForegroundService(NOTIFICATION_ID)
        }
    }

    override fun onStartCommand(intent: Intent?, flags: Int, startId: Int): Int {
        Timber.d("MyAppBeaconService onStartCommand")
        if (intent?.action == STOP_BEACON_FOREGROUND_ACTION) {
            stopForeground(true)
            stopSelf()
        } else {
            try {
                beaconManager = BeaconManager.getInstanceForApplication(this)
                beaconManager?.beaconParsers?.add(BeaconParser().setBeaconLayout(IBEACON_LAYOUT))
                beaconManager?.beaconParsers?.add(BeaconParser().setBeaconLayout(EDDYSTONE_UID_LAYOUT))
                beaconManager?.beaconParsers?.add(BeaconParser().setBeaconLayout(EDDYSTONE_URL_LAYOUT))
                beaconManager?.beaconParsers?.add(BeaconParser().setBeaconLayout(EDDYSTONE_TLM_LAYOUT))

                beaconManager?.backgroundBetweenScanPeriod = 1000L
                beaconManager?.foregroundBetweenScanPeriod = 1000L
                beaconManager?.enableForegroundServiceScanning(builder?.build(), NOTIFICATION_ID)
                beaconManager?.setEnableScheduledScanJobs(false)
                beaconManager?.backgroundScanPeriod = 1100L
                beaconManager?.bindInternal(this)
                beaconManager?.updateScanPeriods()

            } catch (e: Exception) {
                e.printStackTrace()
[... 117 lines cut ...]
        beaconManager?.addRangeNotifier(this)
        try {
            beaconManager?.startMonitoring(region)
            beaconManager?.startRangingBeacons(region)
        } catch (e: RemoteException) {
            e.printStackTrace()
            Timber.d("MyAppBeaconService - onBeaconServiceConnect - errStartMonitoringRanging: ${e.message}, exception: ${e.printStackTrace()}")
        }
    }

    private fun startBeaconService() {
        try {
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
                context.startForegroundService(Intent(context, MyAppBeaconService::class.java))
            } else {
                context.startService(Intent(context, MyAppBeaconService::class.java))
            }
        } catch (exception: Exception) {
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.S
                && exception is ForegroundServiceStartNotAllowedException) {
                Timber.d("MyAppBeaconService - startBeaconService - ForegroundServiceStartNotAllowedException: ${exception.message}")
            }
        }
    }
}

davidgyoung, 2024-01-22:
Interesting.  So your app starts its own custom foreground service which
then starts the Android Beacon Library’s built-in foreground service using
a single common Notification.

I am a bit surprised this works — but I presume it does at least on Android
13 and below.

I wonder if for some reason this unusual setup has an impact on Android
14’s new rules on foreground services (whether intended or not).

Have you or your team done any testing on Android 14 yourself to see if
this technique works in general?

stephenruda, 2024-01-22:
Over the weekend, I was not able to reproduce the crash in my app after re-enabling battery optimizations.

I agree that this is an interesting approach.  I have never done anything exactly like this.  In my app, I do have instances where I start a foreground service from within another foreground service.  My understanding of foreground services is that it _should_ work.  But there are so many intricacies and every new Android update adds more.

I am having a hard time wrapping my head around every possible scenario and how this situation might have occurred.

My best guess is that you are trying to start the `BeaconService` when it is not allowed - just as the exception states.  You cannot start a foreground service while the app is in the background.
Is it possible that you are trying to start the service after sending the `STOP_BEACON_FOREGROUND_ACTION` flag?

You are also catching a lot of exceptions in the `MyAppBeaconService`.  It is possible that you catch an exception within this service so it never makes it into the foreground - but then you try starting the `BeaconService` which will fail because nothing was in the foreground.

Here are some links that might be helpful:
https://developer.android.com/develop/background-work/services/foreground-services#bg-access-restrictions
https://developer.android.com/develop/background-work/services/foreground-services#wiu-restrictions
https://developer.android.com/develop/background-work/services/fg-service-types#location

Due to the amount of crashes you are seeing, I am guessing this issue is related to your unique set up.

davidgyoung, 2024-01-28:
@rahul-kumawat-vts, as suggested by @stephenruda,  I suspect that the core issue here may be that Android 14 simply does not support your unusual foreground service use case.  I would suggest you do one of two things:

1. Remove the catch of the exceptions inside `startBeaconForegroundService` method and see if you get exception reports (indicating your foreground service is also being blocked from starting by Android.)

- or - 

2. Add a new member variable to `MyForegroundService` called `foregroundServiceStartFailed` that gets set to true if any of the above exceptions inside  `startBeaconForegroundService`  get caught, and if this flag is true,  do not do any of the  call  any of the `BeaconManager` setup which causes a secondary exception to crash the app.   Then see if the crash reports go away.

If you choose option (2) above, I'd suggest you also add something to send stack traces or more detailed logs to a server so you can debug why your foreground service is  not being allowed by Andorid to start, as option 2 may otherwise make your service start failures silent.

I am going to close this issue for now, as I do not believe there is anything the library can do to resolve these Android 14 restrictions.  However, if you find evidence to the contrary in the above tests, feel free to re-open this issue.

closed 2024-01-28 by davidgyoung