AltBeacon/android-beacon-library #1180: Do we have sample app to implement the library ?

rahul-kumawat-vts, 2024-02-19:


rahul-kumawat-vts, 2024-02-19:
My requirement is i have to detect the beacon in each state either App in background, App is removed from Background and App is running in foreground.

davidgyoung, 2024-02-19:
The library supports a number of different "Scan strategies" that can be used to optimize background detections depending on your use case.  All of these scan strategies are shown in the Application class of the reference app you mention above (all but one are commented out, but the comments provide instructions on configuring the other scan strategies.)

See this table for the pros and cons of each scan strategy:

https://altbeacon.github.io/android-beacon-library/detection_times.html

For your purpose, you probably want to consider the Foreground Service scan strategy or the Intent scan strategy.  They are designed to give you the best results for background detections within the limits allowed by the Android operating system.  Neither are perfect, but one should get you as close as possible to your goal.

rahul-kumawat-vts, 2024-02-25:
@davidgyoung Which function is used to stop library foreground service from our app?

davidgyoung, 2024-02-25:
Thy this to stop a previously configured foreground service: 

```
fun stopForegroundService(context: Context) {
        val beaconManager = BeaconManager.getInstanceForApplication(context)
        for (region in beaconManager.monitoredRegions) {
            beaconManager.stopMonitoring(region)
        }
        for (region in beaconManager.rangedRegions) {
            beaconManager.stopRangingBeacons(region)
        }
        BeaconManager.getInstanceForApplication(this).disableForegroundServiceScanning()
    }
```

The last line disables the foreground service configuration.  If you want to restart the foreground service later, you can omit that line -- that way the next time you start ranging or monitoring it will start up again. 

 Be a advised that Android 13 and 14 only allow starting a foreground service when the app is in the foreground (or other specific situations like on BOOT_COMPLETED).