AltBeacon/android-beacon-library #1188: Can we use this with tauri mobile?

egeres, 2024-04-17:
[Tauri](https://github.com/tauri-apps/tauri) is a framework to build desktop applications with many interesting advantages over electron (mainly, it uses rust as a backend language and a more lightweight frontend). After years of works, the developers have made their way into application development with the same technology stack but bundled as an application https://beta.tauri.app/blog/tauri-2-0-0-beta. This could be a replacement for react native in the future, as of today It's still on beta

I'm not entirely familiar on how to link or use android libraries in other development environments like for example react native. Would we need to recompile this library in a special way? Are there limitations in the way this package has been built?

(I'm sorry if I sound vague, I'm a bit lost on this area of development!)

egeres, 2024-04-17:
Alright, so the integration of android libraries seems to be specific of the framework

- Tauri: https://beta.tauri.app/guides/plugins/develop-mobile/#android
- React native: https://reactnative.dev/docs/native-modules-android

davidgyoung, 2024-04-22:
I am not familiar with Tauri, but I do not believe you will be able to control Bleutooth on desktop computers using a library that uses Android APIs, because the way Bluetooth APIs work on Desktop operating systems is very different than on Android.

closed 2024-04-22 by davidgyoung

egeres, 2024-04-23:
They are releasing a version for mobile where you can compile for IOS and Android!!