AltBeacon/android-beacon-library #1197: altbeacon does not detect beacon devices on Android 12 and later

G-Roman1988, 2024-06-21:
Hello! The Android beacon/altbeacon library (version 2.20.6) does not identify any beacon devices on Android 12 and above, but on Android 11 and below the same code works (it identifies). I am working on an application to help visually impaired people identify public transport and I am stuck with this problem. I have tried many solutions I found on the internet and in the documentation, but with no success. Steps to reproduce the problem: create a standard iBeacon and set a UUID for personal proximity, then try to scan beacon devices with the Android beacon/altbeacon library version 2.20.6 on Android 12+. In the log, it shows something about the scan job immediately and the periodic scan time. I will leave the code and manifest file below. Please, anyone who can help, any solution is welcome.  package com.gtdvm.echopoint

import android.annotation.SuppressLint
import android.app.Activity
import android.bluetooth.BluetoothAdapter
import android.bluetooth.BluetoothDevice
import android.bluetooth.BluetoothGatt
import android.bluetooth.BluetoothGattCallback
import android.bluetooth.BluetoothManager
import android.bluetooth.le.ScanCallback
import android.bluetooth.BluetoothGattService
import android.bluetooth.BluetoothGattCharacteristic
import android.bluetooth.BluetoothGattCharacteristic.PERMISSION_READ
import android.bluetooth.BluetoothGattCharacteristic.PERMISSION_WRITE
import android.bluetooth.BluetoothGattCharacteristic.PROPERTY_WRITE
import android.bluetooth.BluetoothGattCharacteristic.PROPERTY_NOTIFY
import android.bluetooth.BluetoothGattCharacteristic.WRITE_TYPE_DEFAULT
import android.bluetooth.BluetoothGattCharacteristic.WRITE_TYPE_NO_RESPONSE
import android.bluetooth.BluetoothGattDescriptor
import com.polidea.rxandroidble3.RxBleClient
import com.polidea.rxandroidble3.RxBleConnection
import com.polidea.rxandroidble3.RxBleDevice
import com.polidea.rxandroidble3.RxBleDeviceServices
import com.polidea.rxandroidble3.scan.ScanSettings
import com.polidea.rxandroidble3.scan.ScanFilter
import com.polidea.rxandroidble3.scan.ScanResult
import com.polidea.rxandroidble3.RxBleScanResult
import com.polidea.rxandroidble3.exceptions.BleException
import com.polidea.rxandroidble3.exceptions.BleDisconnectedException
import com.polidea.rxandroidble3.NotificationSetupMode
import com.polidea.rxandroidble3.RxBleConnection.RxBleConnectionState
import android.location.LocationManager
import android.content.Intent
import androidx.appcompat.app.AppCompatActivity
import android.content.pm.PackageManager
import androidx.core.app.ActivityCompat
import androidx.core.content.ContextCompat
import android.content.Context
import android.content.ServiceConnection
import android.icu.number.IntegerWidth
import android.os.Build
import android.os.Looper
import android.os.ParcelUuid
import android.os.RemoteException
import android.provider.Settings
import android.util.Log
import androidx.activity.result.contract.ActivityResultContracts
import androidx.compose.animation.defaultDecayAnimationSpec
import androidx.compose.ui.res.integerResource
import androidx.compose.ui.res.stringArrayResource
import androidx.lifecycle.ViewModelProvider
import io.reactivex.rxjava3.disposables.Disposable
import io.reactivex.rxjava3.core.Completable
import io.reactivex.rxjava3.core.Observable
import io.reactivex.rxjava3.core.Single
import io.reactivex.rxjava3.core.SingleSource
import io.reactivex.rxjava3.disposables.CompositeDisposable
import io.reactivex.rxjava3.internal.operators.observable.ObservableIgnoreElementsCompletable
import io.reactivex.rxjava3.internal.operators.observable.ObservableJust
import io.reactivex.rxjava3.internal.operators.observable.ObservableScan
import io.reactivex.rxjava3.schedulers.Schedulers
import kotlinx.coroutines.newFixedThreadPoolContext
import java.lang.StringBuilder
import java.nio.ByteBuffer
import java.util.UUID
//import com.gtdvm.echopoint.Utils.getMajor
//import com.gtdvm.echopoint.Utils.getMinor
//import com.gtdvm.echopoint.Utils.getUUID
import org.altbeacon.beacon.BeaconManager
import org.altbeacon.beacon.BeaconParser
[... 309 lines cut ...]
        <activity
            android:name=".ListDevices"
            android:exported="false" />
        <activity
            android:name=".UnderCategory"
            android:exported="false"
            android:label="@string/title_activity_under_category"
            android:theme="@style/Theme.EchoPoint.NoActionBar" />
        <activity
            android:name=".MainActivity"
            android:exported="true"
            android:theme="@style/Theme.EchoPoint">
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />

                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>
        </activity>

        <service android:name="org.altbeacon.beacon.service.ScanJob" android:permission="android.permission.BIND_JOB_SERVICE" />
        <meta-data android:name="android.app.job.JobScheduler.SCHEDULE_IMMEDIATE" android:value="true"/>
        
    </application>

</manifest>  I don't know and don't understand if I am doing something wrong or if there is a problem with the altbeacon library or with Android. So please, anyone who has experience and knowledge, help me understand how I can solve this problem.

VolodaUa, 2024-06-26:
@G-Roman1988 Could you please attach minimal reproducible sample? I can confirm that this library detects IBeacon Beacons  on Android 12+.

G-Roman1988, 2024-06-28:
Hello! Thanks for the reply. of course I attach some logs. these logs are recent and from google pixely 6 android 14 phone.  2024-06-28 15:30:40.528  9015-9015  BeaconManager           com.gtdvm.echopoint                  I  BeaconManager started up on pid 9015 named 'com.gtdvm.echopoint' for application package 'com.gtdvm.echopoint'.  isMainProcess=true
    2024-06-28 15:30:40.531  9015-9015  BeaconParser            com.gtdvm.echopoint                  D  Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
    2024-06-28 15:30:40.546  9015-9015  Compatibil...geReporter com.gtdvm.echopoint                  D  Compat change id reported: 263076149; UID 10316; state: ENABLED
    2024-06-28 15:30:40.549  9015-9015  BeaconParser            com.gtdvm.echopoint                  D  Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
    2024-06-28 15:30:40.565  9015-9015  LOCATION-MANAGER        com.gtdvm.echopoint                  D  locationManager is : true
    2024-06-28 15:30:40.568  9015-9015  BackgroundPowerSaver    com.gtdvm.echopoint                  I  Background mode not set.  We assume we are in the foreground.
    2024-06-28 15:30:40.605  9015-9015  ScanJob                 com.gtdvm.echopoint                  I  Using immediateScanJobId from manifest: 208352939
    2024-06-28 15:30:40.605  9015-9015  Compatibil...geReporter com.gtdvm.echopoint                  D  Compat change id reported: 194532703; UID 10316; state: ENABLED
    2024-06-28 15:30:40.606  9015-9015  Compatibil...geReporter com.gtdvm.echopoint                  D  Compat change id reported: 253665015; UID 10316; state: ENABLED
    2024-06-28 15:30:40.606  9015-9015  Compatibil...geReporter com.gtdvm.echopoint                  D  Compat change id reported: 311402873; UID 10316; state: DISABLED
    2024-06-28 15:30:40.606  9015-9015  Compatibil...geReporter com.gtdvm.echopoint                  D  Compat change id reported: 323349338; UID 10316; state: DISABLED
    2024-06-28 15:30:40.606  9015-9015  JobInfo                 com.gtdvm.echopoint                  W  Job 'com.gtdvm.echopoint/org.altbeacon.beacon.service.ScanJob#208352939' has a deadline with no functional constraints. The deadline won't improve job execution latency. Consider removing the deadline.
    2024-06-28 15:30:40.608  9015-9015  ScanJob                 com.gtdvm.echopoint                  I  Using periodicScanJobId from manifest: 208352940
    2024-06-28 15:30:40.608  9015-9015  JobInfo                 com.gtdvm.echopoint                  W  Requested interval +5m0s0ms for job 208352940 is too small; raising to +15m0s0ms
    2024-06-28 15:30:40.608  9015-9015  JobInfo                 com.gtdvm.echopoint                  W  Requested flex 0 for job 208352940 is too small; raising to +5m0s0ms
    2024-06-28 15:30:40.608  9015-9015  BeaconMonitoring        com.gtdvm.echopoint                  D  Outside region: myUniqueBeaconRegion
    2024-06-28 15:30:40.621  9015-9015  ScanJob                 com.gtdvm.echopoint                  I  Using immediateScanJobId from manifest: 208352939
    2024-06-28 15:30:40.621  9015-9015  JobInfo                 com.gtdvm.echopoint                  W  Job 'com.gtdvm.echopoint/org.altbeacon.beacon.service.ScanJob#208352939' has a deadline with no functional constraints. The deadline won't improve job execution latency. Consider removing the deadline.
    2024-06-28 15:30:40.623  9015-9015  ScanJob                 com.gtdvm.echopoint                  I  Using periodicScanJobId from manifest: 208352940
    2024-06-28 15:30:40.623  9015-9015  JobInfo                 com.gtdvm.echopoint                  W  Requested interval +5m0s0ms for job 208352940 is too small; raising to +15m0s0ms
    2024-06-28 15:30:40.623  9015-9015  JobInfo                 com.gtdvm.echopoint                  W  Requested flex 0 for job 208352940 is too small; raising to +5m0s0ms
    2024-06-28 15:30:40.627  9015-9015  ContentCaptureHelper    com.gtdvm.echopoint                  I  Setting logging level to OFF
    2024-06-28 15:30:40.679  9015-9015  ScanJob                 com.gtdvm.echopoint                  I  ScanJob Lifecycle START: org.altbeacon.beacon.service.ScanJob@e59b6e5
    2024-06-28 15:30:40.724  9015-9686  ScanHelper              com.gtdvm.echopoint                  D  BeaconParsers set to  count: 2
    2024-06-28 15:30:40.724  9015-9686  ScanHelper              com.gtdvm.echopoint                  D  First parser layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
    2024-06-28 15:30:40.725  9015-9686  CycledLeScanner         com.gtdvm.echopoint                  I  Using Android O scanner
    2024-06-28 15:30:40.728  9015-9686  ScanJob                 com.gtdvm.echopoint                  I  Using immediateScanJobId from manifest: 208352939
    2024-06-28 15:30:40.728  9015-9686  ScanJob                 com.gtdvm.echopoint                  I  Running immediate scan job: instance is org.altbeacon.beacon.service.ScanJob@e59b6e5
    2024-06-28 15:30:40.728  9015-9686  ScanJob                 com.gtdvm.echopoint                  I  scanJob version 2.20.6 is starting up on the main process
    2024-06-28 15:30:40.742  9015-9686  ModelSpeci...Calculator com.gtdvm.echopoint                  W  Cannot find match for this device.  Using default
    2024-06-28 15:30:40.742  9015-9686  ModelSpeci...Calculator com.gtdvm.echopoint                  W  Cannot find match for this device.  Using default
[... cut ...]
                    <category android:name="android.intent.category.LAUNCHER" />
                </intent-filter>
            </activity>
    
            <service
                android:name="org.altbeacon.beacon.service.ScanJob"
                android:permission="android.permission.BIND_JOB_SERVICE" />
    
            <meta-data
                android:name="android.app.job.JobScheduler.SCHEDULE_IMMEDIATE"
                android:value="true" />
        </application>
    
    </manifest>   please help me to get out of this impasse/problem.

VolodaUa, 2024-07-02:
Hi @G-Roman1988 

You don't need your foreground service. The library already declared everthing and you just need to build notification, notification channel, grant notification permission(if I'm not mistaken).

This code starts the foreground service. 
beaconManager.enableForegroundServiceScanning(builder.build(), 456)
beaconManager.setEnableScheduledScanJobs(false)

ForegroundService -> https://altbeacon.github.io/android-beacon-library/foreground-service.html

If you need to have BLE comunications with BLE device when the beacons are found:
1) Clean up your code, remove all extra logic.
2) Create your own foreground service, start this service.

val builder = Notification.Builder(this)
....
3) Configure BeaconManager 
Notificaiton notification = builder.build(), 
**beaconManager.enableForegroundServiceScanning(notification ), 456)
beaconManager.setEnableScheduledScanJobs(false)**

In your service -> startForeground() with the same notification id and notification.

So, you will foregound service scanning in library and your own.
Your foreground service type should be location if you want to have it worked for a long time

4) Start monitoring and I suggest to call "requestStateForRegion()" to get the determinate state callbacl.
5) I also suggest startiing ranging when you enetered in the region(Monitori notifier). Do in both "didDeterminateState" and "didEntered".
6) Once you get it worked, proceed with connection and read characteristics. 

Side notes:

  override fun onCreate() {
        super.onCreate()
        appContext = this.applicationContext
        **bluetoothServices = BluetoothServices(appContext as AppCompatActivity)**
        
It looks strange and not sure that it can work - cast applicationContext to Activity.

G-Roman1988, 2024-07-08:
Hello @VolodaUa! I dropped the personal foreground service and set altbeacon to use its own foreground service, which is declared. I created the channel for notification and notification but it doesn't work and the  notification doesn't show either. to app info on notification is not active to allow notifications. on this link : https://altbeacon.github.io/android-beacon-library/foreground-service.html there is some information but i am new to android and i don't understand much there. I am leaving the code below modified and please help me out of this deadlock.   package com.gtdvm.echopoint

import android.annotation.SuppressLint
import android.app.Activity
import android.app.Notification
import android.app.NotificationChannel
import android.app.NotificationManager
import android.os.IBinder
import android.app.Service
import android.bluetooth.BluetoothAdapter
import android.bluetooth.BluetoothDevice
import android.bluetooth.BluetoothGatt
import android.bluetooth.BluetoothGattCallback
import android.bluetooth.BluetoothManager
import android.bluetooth.le.ScanCallback
import android.bluetooth.BluetoothGattService
import android.bluetooth.BluetoothGattCharacteristic
import android.bluetooth.BluetoothGattCharacteristic.PERMISSION_READ
import android.bluetooth.BluetoothGattCharacteristic.PERMISSION_WRITE
import android.bluetooth.BluetoothGattCharacteristic.PROPERTY_WRITE
import android.bluetooth.BluetoothGattCharacteristic.PROPERTY_NOTIFY
import android.bluetooth.BluetoothGattCharacteristic.WRITE_TYPE_DEFAULT
import android.bluetooth.BluetoothGattCharacteristic.WRITE_TYPE_NO_RESPONSE
import android.bluetooth.BluetoothGattDescriptor
import com.polidea.rxandroidble3.RxBleClient
import com.polidea.rxandroidble3.RxBleConnection
import com.polidea.rxandroidble3.RxBleDevice
import com.polidea.rxandroidble3.RxBleDeviceServices
import com.polidea.rxandroidble3.scan.ScanSettings
import com.polidea.rxandroidble3.scan.ScanFilter
import com.polidea.rxandroidble3.scan.ScanResult
import com.polidea.rxandroidble3.RxBleScanResult
import com.polidea.rxandroidble3.exceptions.BleException
import com.polidea.rxandroidble3.exceptions.BleDisconnectedException
import com.polidea.rxandroidble3.NotificationSetupMode
import com.polidea.rxandroidble3.RxBleConnection.RxBleConnectionState
import android.location.LocationManager
import android.content.Intent
import androidx.appcompat.app.AppCompatActivity
import android.content.pm.PackageManager
import androidx.core.app.ActivityCompat
import androidx.core.content.ContextCompat
import android.content.Context
import android.content.ServiceConnection
import android.icu.number.IntegerWidth
import android.os.Build
import android.os.Looper
import android.os.ParcelUuid
import android.os.RemoteException
import android.provider.Settings
import android.util.Log
import androidx.activity.result.contract.ActivityResultContracts
import androidx.compose.animation.defaultDecayAnimationSpec
import androidx.compose.ui.res.integerResource
import androidx.compose.ui.res.stringArrayResource
import androidx.core.app.NotificationCompat
import androidx.lifecycle.ViewModelProvider
import io.reactivex.rxjava3.disposables.Disposable
import io.reactivex.rxjava3.core.Completable
import io.reactivex.rxjava3.core.Observable
import io.reactivex.rxjava3.core.Single
import io.reactivex.rxjava3.core.SingleSource
import io.reactivex.rxjava3.disposables.CompositeDisposable
import io.reactivex.rxjava3.internal.operators.observable.ObservableIgnoreElementsCompletable
import io.reactivex.rxjava3.internal.operators.observable.ObservableJust
import io.reactivex.rxjava3.internal.operators.observable.ObservableScan
import io.reactivex.rxjava3.schedulers.Schedulers
import kotlinx.coroutines.newFixedThreadPoolContext
import java.lang.StringBuilder
import java.nio.ByteBuffer
[... 349 lines cut ...]
2024-07-08 07:03:49.365 12677-12953 BluetoothAdapter        com.gtdvm.echopoint                  D  isLeEnabled(): ON
2024-07-08 07:03:49.380 12677-12953 BluetoothAdapter        com.gtdvm.echopoint                  D  isLeEnabled(): ON
2024-07-08 07:03:49.390 12677-12917 BluetoothLeScanner      com.gtdvm.echopoint                  D  onScannerRegistered() - status=0 scannerId=2 mScannerId=0
2024-07-08 07:03:56.021 12677-12953 BluetoothAdapter        com.gtdvm.echopoint                  D  isLeEnabled(): ON
2024-07-08 07:03:56.037 12677-12953 BluetoothAdapter        com.gtdvm.echopoint                  D  isLeEnabled(): ON
2024-07-08 07:03:56.048 12677-12917 BluetoothLeScanner      com.gtdvm.echopoint                  D  onScannerRegistered() - status=0 scannerId=2 mScannerId=0
2024-07-08 07:04:02.660 12677-12953 BluetoothAdapter        com.gtdvm.echopoint                  D  isLeEnabled(): ON
2024-07-08 07:04:02.671 12677-12953 BluetoothAdapter        com.gtdvm.echopoint                  D  isLeEnabled(): ON
2024-07-08 07:04:02.683 12677-12917 BluetoothLeScanner      com.gtdvm.echopoint                  D  onScannerRegistered() - status=0 scannerId=2 mScannerId=0
2024-07-08 07:04:06.776 12677-12677 BeaconService           com.gtdvm.echopoint                  I  stop monitoring received
2024-07-08 07:04:06.779 12677-12677 CycledLeScanner         com.gtdvm.echopoint                  I  Adjusted scanStopTime to be 255195385
2024-07-08 07:04:06.779 12677-12677 BeaconService           com.gtdvm.echopoint                  I  stop ranging received
[... cut ...]
2024-07-08 07:04:07.401 12677-12677 VRI[ListDevices]        com.gtdvm.echopoint                  D  visibilityChanged oldVisibility=true newVisibility=false
2024-07-08 07:04:07.416 12677-12677 WindowOnBackDispatcher  com.gtdvm.echopoint                  W  sendCancelIfRunning: isInProgress=falsecallback=androidx.activity.OnBackPressedDispatcher$Api34Impl$createOnBackAnimationCallback$1@c65e3f8
2024-07-08 07:04:07.422 12677-12677 WindowOnBackDispatcher  com.gtdvm.echopoint                  W  sendCancelIfRunning: isInProgress=falsecallback=android.app.Activity$$ExternalSyntheticLambda0@7273f5b
 please help me with any idea.

G-Roman1988, 2024-07-08:
I solved it with the foreground service notification, when I start the scan, the notification also appears, but the beacon devices are not identified on android 12 and higher but on android 11, everything is fine. in the log, no changes, I see no errors or warnings.

VolodaUa, 2024-07-16:
Hi @G-Roman1988 
 
 I'll check once I have free time. Please attach the minimal reproducible project without the build folder. Just project folder.
 It's difficult to help with hundreds of lines...
 
 Before that, I would recommend you do the following now:
 
 1) Download this app:
 https://github.com/vhiribarren/beacon-simulator-android
 
 Emulate your beacon using this app. Only iBeacon part!!!
 2) Check if it works with this app.
 
3)
  beaconManager.foregroundScanPeriod = 1100L
  beaconManager.foregroundBetweenScanPeriod = 0L
 Additionally, I would recommend removing such lines and use default library settings.

GurgenHOVH, 2024-07-16:
I had a similar issue and removing neverForLocation from BLUETOOTH_SCAN permission solved my issue.

G-Roman1988, 2024-07-24:
[Echopoint.zip](https://github.com/user-attachment  Hello ! I have tried all your ideas with no success. I have attached code from the project zip file, maybe you can identify what the problem is and where. for me when i start scan the notification shows after a second and I don't know if that's good or not. I have this in my diary: 2024-07-24 12:34:07.085  6848-6848  BeaconManager           com.gtdvm.echopoint                  I  BeaconManager started up on pid 6848 named 'com.gtdvm.echopoint' for application package 'com.gtdvm.echopoint'.  isMainProcess=true
2024-07-24 12:34:07.088  6848-6848  BeaconParser            com.gtdvm.echopoint                  D  Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
2024-07-24 12:34:07.106  6848-6848  Compatibil...geReporter com.gtdvm.echopoint                  D  Compat change id reported: 263076149; UID 10316; state: ENABLED
2024-07-24 12:34:07.111  6848-6848  BeaconParser            com.gtdvm.echopoint                  D  Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
2024-07-24 12:34:07.111  6848-6848  NOTIFICATION_CHANEL     com.gtdvm.echopoint                  D  the create notification channel function was called
2024-07-24 12:34:07.112  6848-6848  BeaconManager           com.gtdvm.echopoint                  W  Disabling ScanJobs on Android 8+ may disable delivery of beacon callbacks in the background unless a foreground service is active.
2024-07-24 12:34:07.114  6848-6848  ScanJob                 com.gtdvm.echopoint                  I  Using immediateScanJobId from manifest: 208352939
2024-07-24 12:34:07.114  6848-6848  ScanJob                 com.gtdvm.echopoint                  I  Using periodicScanJobId from manifest: 208352940
2024-07-24 12:34:07.114  6848-6848  CREATE_NOTIFICATION     com.gtdvm.echopoint                  D  the notification has been created
2024-07-24 12:34:07.117  6848-6848  BeaconManager           com.gtdvm.echopoint                  W  Disabling ScanJobs on Android 8+ may disable delivery of beacon callbacks in the background unless a foreground service is active.
2024-07-24 12:34:07.117  6848-6848  ScanJob                 com.gtdvm.echopoint                  I  Using immediateScanJobId from manifest: 208352939
2024-07-24 12:34:07.118  6848-6848  ScanJob                 com.gtdvm.echopoint                  I  Using periodicScanJobId from manifest: 208352940
2024-07-24 12:34:07.135  6848-6848  LOCATION-MANAGER        com.gtdvm.echopoint                  D  locationManager is : true
2024-07-24 12:34:07.138  6848-6848  BackgroundPowerSaver    com.gtdvm.echopoint                  I  Background mode not set.  We assume we are in the foreground.
2024-07-24 12:34:07.139  6848-6848  BeaconManager           com.gtdvm.echopoint                  I  Attempting to starting foreground beacon scanning service.
2024-07-24 12:34:07.146  6848-6848  BeaconManager           com.gtdvm.echopoint                  I  successfully started foreground beacon scanning service.
2024-07-24 12:34:07.148  6848-6848  BeaconMonitoring        com.gtdvm.echopoint                  D  Outside region: myUniqueBeaconRegion
2024-07-24 12:34:07.152  6848-6848  ContentCaptureHelper    com.gtdvm.echopoint                  I  Setting logging level to OFF
2024-07-24 12:34:07.166  6848-6848  CycledLeScanner         com.gtdvm.echopoint                  I  Using Android O scanner
2024-07-24 12:34:07.167  6848-6848  ScanHelper              com.gtdvm.echopoint                  D  BeaconParsers set to  count: 0
2024-07-24 12:34:07.167  6848-6848  BeaconService           com.gtdvm.echopoint                  I  beaconService version 2.20.6 is starting up on the main process
2024-07-24 12:34:07.180  6848-6848  ModelSpeci...Calculator com.gtdvm.echopoint                  W  Cannot find match for this device.  Using default
2024-07-24 12:34:07.181  6848-6848  ModelSpeci...Calculator com.gtdvm.echopoint                  W  Cannot find match for this device.  Using default
2024-07-24 12:34:07.181  6848-6848  BeaconService           com.gtdvm.echopoint                  I  starting with intent Intent { cmp=com.gtdvm.echopoint/org.altbeacon.beacon.service.BeaconService }
2024-07-24 12:34:07.181  6848-6848  BeaconService           com.gtdvm.echopoint                  I  binding
2024-07-24 12:34:07.213  6848-6848  BeaconMonitoring        com.gtdvm.echopoint                  D  Outside region: myUniqueBeaconRegion
2024-07-24 12:34:07.220  6848-6848  BeaconService           com.gtdvm.echopoint                  I  start ranging received
2024-07-24 12:34:07.221  6848-6953  BluetoothAdapter        com.gtdvm.echopoint                  D  isLeEnabled(): ON
2024-07-24 12:34:07.221  6848-6848  Compatibil...geReporter com.gtdvm.echopoint                  D  Compat change id reported: 160794467; UID 10316; state: ENABLED
2024-07-24 12:34:07.222  6848-6848  BeaconService           com.gtdvm.echopoint                  I  start monitoring received
2024-07-24 12:34:07.226  6848-6865  BluetoothLeScanner      com.gtdvm.echopoint                  D  onScannerRegistered() - status=0 scannerId=2 mScannerId=0
2024-07-24 12:34:07.304  6848-6848  BeaconService           com.gtdvm.echopoint                  I  Received settings update
2024-07-24 12:34:07.710  6848-6848  VRI[MainActivity]       com.gtdvm.echopoint                  D  visibilityChanged oldVisibility=true newVisibility=false
[... cut ...]
2024-07-24 12:35:42.778  6848-6848  VRI[ListDevices]        com.gtdvm.echopoint                  D  visibilityChanged oldVisibility=true newVisibility=false
2024-07-24 12:35:42.790  6848-6848  WindowOnBackDispatcher  com.gtdvm.echopoint                  W  sendCancelIfRunning: isInProgress=falsecallback=androidx.activity.OnBackPressedDispatcher$Api34Impl$createOnBackAnimationCallback$1@6844145
2024-07-24 12:35:42.796  6848-6848  WindowOnBackDispatcher  com.gtdvm.echopoint                  W  sendCancelIfRunning: isInProgress=falsecallback=android.app.Activity$$ExternalSyntheticLambda0@76cbdbc

VolodaUa, 2024-07-24:
@G-Roman1988 This file is not available.

G-Roman1988, 2024-07-25:
I'm attaching the file again and if it doesn't work I'll leave this link for download  
[Echopoint.zip](https://github.com/user-attachments/files/16374431/Echopoint.zip)
https://cloud.gtdvm.com/index.php/s/PiT9Zxt9we56ktk

davidgyoung, 2025-03-15:
This request is out of scope for this forum because it asks for help debugging a custom app.  This forum is reserved for feature requests and reproducible bugs in the library using a the official reference app.   If you can reproduce with the reference app, please reopen a new issue.  For help with your custom app, please post a question to StackOverflow.com

closed 2025-03-15 by davidgyoung